coolsnowwolf/lede #6804: github action编译失败

IVANSOLIS0414, 2021-05-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：github action编译失败提示perf组件缺少依赖库？

### (2) 路由器型号和固件版本

A：X86 R4S等

### (3) 详细日志

A：
find /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.115/tools/perf-target-x86_64_musl/ipkg-x86_64/perf -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package perf is missing dependencies for the following libraries:
libzstd.so.1
Makefile:76: recipe for target '/workdir/openwrt/bin/targets/x86/64/packages/perf_5.4.115-3_x86_64.ipk' failed
make[3]: Leaving directory '/workdir/openwrt/package/devel/perf'
time: package/devel/perf/compile#0.17#0.07#0.23
make[3]: *** [/workdir/openwrt/bin/targets/x86/64/packages/perf_5.4.115-3_x86_64.ipk] Error 1
package/Makefile:111: recipe for target 'package/devel/perf/compile' failed
make[2]: Leaving directory '/workdir/openwrt'
package/Makefile:107: recipe for target '/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[2]: *** [package/devel/perf/compile] Error 2
make[1]: *** [/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
/workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2
Error: Process completed with exit code 2.

IVANSOLIS0414, 2021-05-08:
Actions地址 https://github.com/IVANSOLIS0414/OpenWrt-DIY/runs/2523879167?check_suite_focus=true

piaoyizy, 2021-05-09:
Github Actions编译失败 ,你来这里问啥问题。