coolsnowwolf/lede #6824: luci-app-ttyd无法与nginx-ssl用

Yofo-me, 2021-05-12:
当我通过网页点击 系统->ttyd终端,网页本应该出现ttyd的地方一片空白

[image]

F12tiao调试信息如下：
> Mixed Content: The page at 'https://192.168.xx.xx/cgi-bin/luci/admin/system/terminal' was loaded over HTTPS, but requested an insecure frame 'http://192.168.xx.xx:7681/'. This request has been blocked; the content must be served over HTTPS.

请问该如何配置，修改？

ZqinKing, 2021-05-13:
遇到同样问题

Dimitri2020007, 2021-05-13:
我遇到这问题已经好长时间了，而且像这样的问题不止是ttyd、还有netdata也是这样的问题。

调试信息显示是ttyd的请求被阻止了。

ttyd是http资源不是https资源

https的资源是不容许有http资源的

既然用的是nginx-ssl

可以尝试反代一下ttyd和netdata这样的http的资源

我不知道咋样设置nginx才能反代http的资源

PyXiaoxin, 2021-05-13:
需要相同的问题+1

jarod360, 2021-05-15:
貌似没有配置ttyd的地方，https不支持套http访问

closed 2021-06-17 by coolsnowwolf

carlsue, 2024-09-11:
可以启用ttyd的ssl解决内网https嵌套访问。但互联网访问就不行了，建议还是要通过反代实现