coolsnowwolf/lede #6827: Softether VPN 拆分通道问题

tester013, 2021-05-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：由于登录服务的限源需求，因此打算在有固定IP的软路由上部署带拆分通道的Softether，以减少路由器的负载压力，搜索的时候找到了这篇文章https://www.viper.pub/softether-vpn-split-tunnel/
其中解决方案是下载了源码后修改一个判断值为false以解锁区域限制，最后重新编译上传覆盖。但是尝试了之后发现服务无法启动，logread -f也没发现问题。
使用的版本是https://github.com/SoftEtherVPN/SoftEtherVPN/releases下的 softether-vpn-src-5.01.9671.tar.gz，与lean的一致。
请问有人成功修改或者有已编译好的ipk吗？谢谢

### (2) 路由器型号和固件版本

A：路由器是X86版本，3125U处理器，OpenWrt GCC 8.4.0 r0-3c3ea0d

### (3) 详细日志

A：无

galaxyskyknight, 2021-05-13:
何必舍近求远用什么softethervpn,配置复杂，用Openvpn完美解决你的问题。
[image]

tester013, 2021-05-13:
> 
> 
> 何必舍近求远用什么softethervpn,配置复杂，用Openvpn完美解决你的问题。
> [image]

因为有一定的审计需求，所以打算用softether的，openvpn没尝试过，但是看界面上审计似乎不怎么方便的样子……