coolsnowwolf/lede #6828: 最近这个版本，helloworld提示依赖丢失

galaxyskyknight, 2021-05-13:
install 的时候，提示依赖丢失，请问如何解决？

builder@BuildServer:~/lede$ ./scripts/feeds install -a
WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'shadowsocksr-libev-ssr-check', which does not exist
WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'shadowsocksr-libev-ssr-local', which does not exist
WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'shadowsocksr-libev-ssr-redir', which does not exist
WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'shadowsocksr-libev-ssr-server', which does not exist

Dimitri2020007, 2021-05-13:
相同问题在这里：

https://github.com/fw876/helloworld/issues/520

592732377, 2021-05-14:
[#6814 ](https://github.com/coolsnowwolf/lede/pull/6814)

galaxyskyknight, 2021-05-14:
貌似已修复，可关闭了

closed 2021-05-14 by galaxyskyknight