coolsnowwolf/lede #6830: ssr plus 服务器端口正常，在别的客户端使用相同端口可以翻墙，但在ssr plus里就不行

djhdj1, 2021-05-13:
1.ssr plus 服务器端口正常，在别的客户端使用相同端口可以翻墙，但在ssr plus里就不行

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

2. ssr plus 服务器端口正常，在别的客户端使用相同端口可以翻墙，但在ssr plus里就不行，参考以下截图
[image]
[image]
[image]
[image]

### (2) 使用的是x86软路由，建立在PVE虚拟机上

### (3) 详细日志
参考上面的图

galaxyskyknight, 2021-05-14:
没法帮到你，这种东西因素太多，各处设置太分散，自个儿掌握点调试诊断技术，自个慢慢弄吧

Juvenile-lzj, 2021-05-22:
我之前的Issues有说到这个问题，ssr plus订阅中转过后的xray节点，TLS host默认还是服务器地址，就导致节点无法使用，需手动将TLS host修改为正确的地址。

Pantheon-Paradoxes, 2021-05-26:
同样发现这个问题，xray单独会出现这个问题，下次不编xray还是用原来的v2ray trojan试试

djhdj1, 2021-05-27:
> 我之前的Issues有说到这个问题，ssr plus订阅中转过后的xray节点，TLS host默认还是服务器地址，就导致节点无法使用，需手动将TLS host修改为正确的地址。

请问一下，应该怎么修改啊

closed 2021-06-17 by coolsnowwolf