coolsnowwolf/lede #6833: mt7615d网卡无法工作/识别

Yofo-me, 2021-05-14:
编译固件勾选mt7615低保，结果发现mt7615d网卡未识别出来，内核模块也没有
以下是相关信息：

- `iw dev`
> root@LEDE-mini:~# iw dev
phy#0
        Interface wlan0
                ifindex 9
                wdev 0x1
                addr 90:4c:e5:07:7b:37
                type managed
                txpower 17.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       0       0       0       0       0       0               0

- `iw phy`
> root@LEDE-mini:~# iw phy
Wiphy phy0
        wiphy index: 0
        max # scan SSIDs: 4
        max scan IEs length: 2257 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports RSN-IBSS.
        Device supports AP-side u-APSD.
        Device supports T-DLS.
        Supported Ciphers:
                * WEP40 (00-0f-ac:1)
                * WEP104 (00-0f-ac:5)
                * TKIP (00-0f-ac:2)
                * CCMP-128 (00-0f-ac:4)
                * CCMP-256 (00-0f-ac:10)
                * GCMP-128 (00-0f-ac:8)
                * GCMP-256 (00-0f-ac:9)
                * CMAC (00-0f-ac:6)
                * CMAC-256 (00-0f-ac:13)
                * GMAC-128 (00-0f-ac:11)
                * GMAC-256 (00-0f-ac:12)
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x3 RX 0x3
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
                 * P2P-client
                 * P2P-GO
                 * outside context of a BSS
        Band 1:
                Capabilities: 0x11ce
                        HT20/HT40
                        SM Power Save disabled
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 3839 bytes
                        DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 8 usec (0x06)
                HT TX/RX MCS rate indexes supported: 0-15
                Bitrates (non-HT):
[... 560 lines cut ...]
xt_policy              16384  4
xt_quota               16384  0
xt_recent              20480  0
xt_set                 16384  0
xt_socket              16384  0
xt_state               16384  0
xt_statistic           16384  0
xt_string              16384  0
xt_tcpmss              16384  0
xt_tcpudp              16384 28
xt_time                16384  0
xxhash                 16384  2 zstd_decompress,zstd_compress
zstd_compress         147456  1 btrfs
zstd_decompress        65536  1 btrfs

- 执行`echo 1 > /sys/kernel/debug/ieee80211/phy0/mt76/dbdc`
> root@LEDE-mini:~# echo 1 > /sys/kernel/debug/ieee80211/phy0/mt76/dbdc
-ash: can't create /sys/kernel/debug/ieee80211/phy0/mt76/dbdc: nonexistent directory

请问应该怎么办？
我有相关的`mt7615d`的闭源驱动，但是其`makefile`文件规则不完整

##### 驱动源码链接：
- 百度云盘 `https://pan.baidu.com/s/1yEssKJyfQnSt_XPv_DaOHg` 提取码： `vtbm`
- cowtranfer `https://joky.cowtransfer.com/s/29942ecfe29f47 ` 取件码：`861749`

Yofo-me, 2021-05-14:
固件中没有mt7615d kmod ipk
[image]

但我尝试单独` make package/lean/mt/drivers/mt7615d/compile V=s`编译后，无论是`packages/x86_64/base/` 还是`packages/x86_64/base/`都没有发现类似于mt7615d这类的插件

Yofo-me, 2021-05-14:
```shell
root@LEDE-mini:~# ls  /lib/modules/$(uname -r) |grep mt
mt76-sdio.ko
mt76-usb.ko
mt76.ko
mt7601u.ko
mt7603e.ko
mt7615-common.ko
mt7615e.ko
mt7663-usb-sdio-common.ko
mt7663s.ko
mt7663u.ko
mt76x0-common.ko
mt76x02-lib.ko
mt76x02-usb.ko
mt76x0e.ko
mt76x0u.ko
mt76x2-common.ko
mt76x2e.ko
mt76x2u.ko
mt7915e.ko
```

Yofo-me, 2021-05-14:
root@LEDE-mini:~# dmesg
[    0.000000] Linux version 5.4.117 (Yofo@cloud.joky.io) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r3178-910af7e48)) #0 SMP Tue May 11 13:28:59 2021
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=81db709f-a59f-aaf4-f3b9-5a4334355602 rootwait console=tty0 console=ttyS0,115200n8 noinitrd
[    0.000000] x86/fpu: x87 FPU will use FXSAVE
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000003efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000003f000-0x000000000003ffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000000040000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000001effffff] usable
[    0.000000] BIOS-e820: [mem 0x000000001f000000-0x00000000201fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000020200000-0x0000000077281fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000077282000-0x00000000772fcfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000772fd000-0x0000000077320fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x0000000077321000-0x000000007788bfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007788c000-0x0000000077b57fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000077b58000-0x0000000077ba9fff] type 20
[    0.000000] BIOS-e820: [mem 0x0000000077baa000-0x0000000077ffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fea00000-0x00000000feafffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed03000-0x00000000fed03fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed06000-0x00000000fed06fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed08000-0x00000000fed09fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1cfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed80000-0x00000000fedbffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffb00000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000027fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.40 by American Megatrends
[    0.000000] efi:  ESRT=0x77b54318  ACPI=0x77313000  ACPI 2.0=0x77313000  SMBIOS=0x77a7bf98
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: Haier           Haier DT Computer/BSWD-CM, BIOS EIBSHE03 11/23/2016
[    0.000000] tsc: Detected 1600.000 MHz processor
[    0.000035] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000040] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000059] last_pfn = 0x280000 max_arch_pfn = 0x400000000
[    0.000068] MTRR default type: uncachable
[    0.000072] MTRR fixed ranges enabled:
[    0.000074]   00000-9FFFF write-back
[    0.000078]   A0000-FFFFF uncachable
[    0.000081] MTRR variable ranges enabled:
[    0.000085]   0 base 000000000 mask F80000000 write-back
[    0.000088]   1 base 07A800000 mask FFF800000 uncachable
[    0.000091]   2 base 07B000000 mask FFF000000 uncachable
[    0.000093]   3 base 07C000000 mask FFC000000 uncachable
[    0.000096]   4 base 100000000 mask F00000000 write-back
[    0.000099]   5 base 200000000 mask F80000000 write-back
[    0.000101]   6 disabled
[    0.000102]   7 disabled
[    0.000246] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT
[    0.000531] e820: update [mem 0x7a800000-0xffffffff] usable ==> reserved
[    0.000538] last_pfn = 0x78000 max_arch_pfn = 0x400000000
[    0.009828] esrt: Reserving ESRT space from 0x0000000077b54318 to 0x0000000077b54350.
[    0.012034] Secure boot could not be determined
[    0.012044] ACPI: Early table checksum verification disabled
[    0.012054] ACPI: RSDP 0x0000000077313000 000024 (v02 haier )
[    0.012067] ACPI: XSDT 0x0000000077313088 00008C (v01 haier  computer 01072009 AMI  00010013)
[    0.012082] ACPI: FACP 0x000000007731F128 00010C (v05 haier  computer 01072009 AMI  00010013)
[    0.012092] ACPI BIOS Warning (bug): 32/64X length mismatch in FADT/Gpe0Block: 128/32 (20190816/tbfadt-569)
[    0.012105] ACPI: DSDT 0x00000000773131A0 00BF85 (v02 haier  computer 01072009 INTL 20120913)
[    0.012114] ACPI: FACS 0x000000007788BE80 000040
[    0.012121] ACPI: APIC 0x000000007731F238 000084 (v03 haier  computer 01072009 AMI  00010013)
[    0.012130] ACPI: FPDT 0x000000007731F2C0 000044 (v01 haier  computer 01072009 AMI  00010013)
[    0.012138] ACPI: FIDT 0x000000007731F308 00009C (v01 haier  computer 01072009 AMI  00010013)
[    0.012146] ACPI: MSDM 0x000000007731F3A8 000055 (v03 haier  computer 01072009 AMI  00010013)
[    0.012155] ACPI: MCFG 0x000000007731F400 00003C (v01 haier  computer 01072009 MSFT 00000097)
[    0.012163] ACPI: SLIC 0x000000007731F440 000176 (v01 haier  computer 01072009 AMI  00010013)
[... 655 lines cut ...]
[   39.619663] F2FS-fs (sda128): Can't find valid F2FS filesystem in 2th superblock
[   39.628269] fuseblk: Unknown parameter 'discard'
[   39.633296] exFAT-fs (sda128): invalid boot record signature
[   39.639241] exFAT-fs (sda128): failed to read boot sector
[   39.644873] exFAT-fs (sda128): failed to recognize exfat type
[   41.437745] r8168: eth0: link up
[... cut ...]
[   90.328230] hid-generic 0003:04F3:0103.0001: input,hidraw0: USB HID v1.11 Keyboard [HID 04f3:0103] on usb-0000:00:14.0-4/input0
[   90.343937] input: HID 04f3:0103 Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.1/0003:04F3:0103.0002/input/input12
[   90.356803] input: HID 04f3:0103 System Control as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.1/0003:04F3:0103.0002/input/input13
[   90.369457] hid-generic 0003:04F3:0103.0002: input,hidraw1: USB HID v1.11 Device [HID 04f3:0103] on usb-0000:00:14.0-4/input1
root@LEDE-mini:~#

AmadeusGhost, 2021-05-17:
> 但我尝试单独 make package/lean/mt/drivers/mt7615d/compile V=s 编译后，无论是 packages/x86_64/base/ 还是packages/x86_64/base/ 都没有发现类似于 mt7615d 这类的插件

这些闭源驱动只有 ramips 才能用，x86 选择 kmod-mt7615e kmod-mt7615-firmware wpad 应该就可以了

closed 2021-06-17 by coolsnowwolf