coolsnowwolf/lede #6835: ./scripts/feeds update -a && ./scripts/feeds install -a之后报错了

zxjack, 2021-05-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：./scripts/feeds update -a && ./scripts/feeds install -a之后报错了

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志
WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'shadowsocksr-libev-ssr-check', which does not exist
WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'shadowsocksr-libev-ssr-local', which does not exist
WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'shadowsocksr-libev-ssr-redir', which does not exist
WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'shadowsocksr-libev-ssr-server', which does not exist
WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'v2ray-plugin', which does not exist

liuxichi, 2021-05-15:
清理一次即可

galaxyskyknight, 2021-05-15:
rm -rf tmp && ./scripts/feeds install -f -p helloworld shadowsocksr-libev

ddzzhen, 2021-05-22:
meet the  same  problem ,and  solved  it  ,Thanks .

closed 2021-06-17 by coolsnowwolf