coolsnowwolf/lede #6841: 微信延迟，电脑端正常，只是手机端微信延迟

Nicepaul, 2021-05-16:
(1) 具体问题
最近发现微信延迟，电脑端微信正常（电脑是网线连接），只是手机端微信延迟。如果不用家里wifi，用手机流量微信就正常了。
 (2) 路由器型号和固件版本
主机型号	Intel(R) Core(TM) i5-7300U CPU @ 2.60GHz : 1 Core 2 Thread (CpuMark : 53507.231837 Scores)
CPU 信息	2712.000 MHz
固件版本	OpenWrt R21.4.18 / LuCI Master (git-21.110.12495-2c780cc)
内核版本	5.4.113
(3) 详细日志
无报错

xiaoguoyi, 2021-05-16:
还以为是特例。我也是这样，是安卓，老毛病了

galaxyskyknight, 2021-05-17:
这跟固件有啥关系？？？

9l, 2021-05-20:
1, 可能是你设置有问题
2, 很可能是你dns的问题

你自己检查吧，没log，没配置详情的

yunxi993, 2021-05-20:
情况差不多，不过我是偶尔延迟，不是常有
passwall-DNS设置（自定义DNS）-双ADG内外分开，模式-四项GFW（UDP节点关闭）

pipihu, 2021-05-21:
我也遇到这个情况,路由是小米R3G.
编译lede如果使用开源驱动,或者直接用openwrt官方固件,或者用padavan都没有问题.
使用mtk闭源驱动手机微信就延迟.
固件刷好其它啥都不配置,就设置个wifi密码 pppoe拨号就会这样.也没发现日志有报错信息.
主要是头疼这个问题不知道从哪儿入手查.

Ivanriver, 2021-05-25:
同样的问题，MTK7621

cxgreat2014, 2021-06-11:
同样的问题，极路由B70

closed 2021-06-17 by coolsnowwolf