coolsnowwolf/lede #6843: acrh17 wifi驱动好像有问题，导致短暂性无网络

icsfy, 2021-05-16:
### (1) 具体问题

A：wifi有问题，详见下面的系统日志。看日志这是ath10k驱动的问题？20:42左右出现崩溃时也会导致有线短暂性没网

### (2) 路由器型号和固件版本

A：编译固件时用的默认的驱动，没有特别地选择其它的驱动
[image]

### (3) 详细日志

A：系统日志：
<details>
```
Sun May 16 16:41:00 2021 kern.warn kernel: [1614936.518548] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Sun May 16 16:41:00 2021 kern.warn kernel: [1614936.526525] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Sun May 16 16:41:00 2021 kern.warn kernel: [1614936.535314] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Sun May 16 16:41:00 2021 kern.warn kernel: [1614936.541320] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Sun May 16 16:41:00 2021 kern.err kernel: [1614936.579705] ath10k_pci 0000:01:00.0: Cannot communicate with firmware, previous wmi cmds: 36872:161466762 37004:161466738 36967:161466376 36967:161466358, jiffies: 161467171, attempting to fake crash and restart firmware, dev-flags: 0x42
Sun May 16 16:41:00 2021 kern.warn kernel: [1614936.580257] ath10k_pci 0000:01:00.0: failed to set beacon mode for vdev 0: -11
Sun May 16 16:41:00 2021 kern.warn kernel: [1614936.600572] ath10k_pci 0000:01:00.0: failed to set dtim period for vdev 0: -108
Sun May 16 16:41:00 2021 kern.warn kernel: [1614936.619950] ath10k_pci 0000:01:00.0: failed to set cts protection for vdev 0: -108
Sun May 16 16:41:00 2021 kern.warn kernel: [1614936.620436] ath10k_pci 0000:01:00.0: failed to recalculate rts/cts prot for vdev 0: -108
Sun May 16 16:41:00 2021 kern.warn kernel: [1614936.627196] ath10k_pci 0000:01:00.0: failed to set preamble for vdev 0: -108
Sun May 16 16:41:00 2021 kern.warn kernel: [1614936.635668] ath10k_pci 0000:01:00.0: failed to set mgmt tx rate -108
Sun May 16 16:41:00 2021 kern.warn kernel: [1614936.645388] ath10k_pci 0000:01:00.0: could not request stats (type 25 ret -108 specifier 0)
Sun May 16 16:41:00 2021 kern.err kernel: [1614937.026628] ath10k_pci 0000:01:00.0: firmware crashed! (guid d6e55842-5771-4b8c-a14d-979dbce7c679)
Sun May 16 16:41:00 2021 kern.info kernel: [1614937.027257] ath10k_pci 0000:01:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
Sun May 16 16:41:00 2021 kern.info kernel: [1614937.035788] ath10k_pci 0000:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
Sun May 16 16:41:00 2021 kern.info kernel: [1614937.096225] ath10k_pci 0000:01:00.0: firmware ver 10.4b-ct-9984-fW-13-5ae337bb1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 7ea63dc5
Sun May 16 16:41:00 2021 kern.info kernel: [1614937.116122] ath10k_pci 0000:01:00.0: board_file api 2 bmi_id 0:3 crc32 85498734
Sun May 16 16:41:00 2021 kern.info kernel: [1614937.124618] ath10k_pci 0000:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
Sun May 16 16:41:00 2021 kern.err kernel: [1614937.132322] ath10k_pci 0000:01:00.0: firmware register dump:
Sun May 16 16:41:00 2021 kern.err kernel: [1614937.142324] ath10k_pci 0000:01:00.0: [00]: 0x0096068C 0x00400C00 0x00980000 0x009C20A0
Sun May 16 16:41:00 2021 kern.err kernel: [1614937.147752] ath10k_pci 0000:01:00.0: [04]: 0x009C1FC8 0x00961C1C 0x00961CB0 0x00961C64
Sun May 16 16:41:00 2021 kern.err kernel: [1614937.156291] ath10k_pci 0000:01:00.0: [08]: 0x00961C44 0x00962E70 0x00962D64 0x00962D4C
Sun May 16 16:41:00 2021 kern.err kernel: [1614937.164353] ath10k_pci 0000:01:00.0: [12]: 0x00961E04 0x00962EC4 0x00962EA4 0x00962EAC
Sun May 16 16:41:00 2021 kern.err kernel: [1614937.172419] ath10k_pci 0000:01:00.0: [16]: 0x00962EB4 0x00962EBC 0x00963350 0x0096308C
Sun May 16 16:41:00 2021 kern.err kernel: [1614937.180499] ath10k_pci 0000:01:00.0: [20]: 0x00963068 0x00962EEC 0x009630A0 0x00963004
Sun May 16 16:41:00 2021 kern.err kernel: [1614937.188290] ath10k_pci 0000:01:00.0: [24]: 0x00962F20 0x00963078 0x0096303C 0x009632F0
Sun May 16 16:41:00 2021 kern.err kernel: [1614937.196757] ath10k_pci 0000:01:00.0: [28]: 0x00963338 0x00963348 0x00963360 0x009631B4
Sun May 16 16:41:00 2021 kern.err kernel: [1614937.204723] ath10k_pci 0000:01:00.0: [32]: 0x009630EC 0x00962F80 0x009C2C3C 0x00962FDC
Sun May 16 16:41:00 2021 kern.err kernel: [1614937.212799] ath10k_pci 0000:01:00.0: [36]: 0x009636AC 0x00963C54 0x00963C9C 0x00963DB4
Sun May 16 16:41:00 2021 kern.err kernel: [1614937.220855] ath10k_pci 0000:01:00.0: [40]: 0x00963E08 0x00963E1C 0x00963E80 0x009606B8
Sun May 16 16:41:00 2021 kern.err kernel: [1614937.228657] ath10k_pci 0000:01:00.0: [44]: 0x009606CC 0x009C1CD4 0x00963F94 0x00963BC8
Sun May 16 16:41:00 2021 kern.err kernel: [1614937.237010] ath10k_pci 0000:01:00.0: [48]: 0x009638D4 0x00963508 0x00963834 0x00963858
Sun May 16 16:41:00 2021 kern.err kernel: [1614937.245045] ath10k_pci 0000:01:00.0: [52]: 0x0096386C 0x00963890 0x00973864 0x00973874
Sun May 16 16:41:00 2021 kern.err kernel: [1614937.253150] ath10k_pci 0000:01:00.0: [56]: 0x0097386C 0x009606C4 0x00963FA0 0x00963B50
Sun May 16 16:41:00 2021 kern.err kernel: [1614937.261362] ath10k_pci 0000:01:00.0: Copy Engine register dump:
Sun May 16 16:41:00 2021 kern.err kernel: [1614937.269314] ath10k_pci 0000:01:00.0: [00]: 0x0004a000  11  11   3   3
Sun May 16 16:41:00 2021 kern.err kernel: [1614937.275024] ath10k_pci 0000:01:00.0: [01]: 0x0004a400   8   8 110 111
Sun May 16 16:41:00 2021 kern.err kernel: [1614937.281725] ath10k_pci 0000:01:00.0: [02]: 0x0004a800  19  19  79  83
Sun May 16 16:41:00 2021 kern.err kernel: [1614937.288036] ath10k_pci 0000:01:00.0: [03]: 0x0004ac00  13  13  17  13
Sun May 16 16:41:00 2021 kern.err kernel: [1614937.294921] ath10k_pci 0000:01:00.0: [04]: 0x0004b000 2982 2982 113  73
Sun May 16 16:41:00 2021 kern.err kernel: [1614937.301512] ath10k_pci 0000:01:00.0: [05]: 0x0004b400  15  15 138 143
Sun May 16 16:41:00 2021 kern.err kernel: [1614937.308175] ath10k_pci 0000:01:00.0: [06]: 0x0004b800  21  21  21  21
Sun May 16 16:41:00 2021 kern.err kernel: [1614937.314870] ath10k_pci 0000:01:00.0: [07]: 0x0004bc00   1   1   1   1
Sun May 16 16:41:00 2021 kern.err kernel: [1614937.321462] ath10k_pci 0000:01:00.0: [08]: 0x0004c000   0   0 127   0
Sun May 16 16:41:00 2021 kern.err kernel: [1614937.327791] ath10k_pci 0000:01:00.0: [09]: 0x0004c400   0   0   0   0
Sun May 16 16:41:00 2021 kern.err kernel: [1614937.334678] ath10k_pci 0000:01:00.0: [10]: 0x0004c800   0   0   0   0
Sun May 16 16:41:00 2021 kern.err kernel: [1614937.341247] ath10k_pci 0000:01:00.0: [11]: 0x0004cc00   0   0   0   0
Sun May 16 16:41:00 2021 kern.warn kernel: [1614937.347825] ath10k_pci 0000:01:00.0: debug log header, dbuf: 0x423848  dropped: 0
Sun May 16 16:41:00 2021 kern.warn kernel: [1614937.354661] ath10k_pci 0000:01:00.0: [0] next: 0x423830 buf: 0x419640 sz: 1500 len: 0 count: 0 free: 0
Sun May 16 16:41:00 2021 kern.warn kernel: [1614937.362305] ath10k_pci 0000:01:00.0: [1] next: 0x423848 buf: 0x419050 sz: 1500 len: 0 count: 0 free: 0
Sun May 16 16:41:03 2021 kern.warn kernel: [1614939.459275] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Sun May 16 16:41:03 2021 kern.warn kernel: [1614940.048690] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Sun May 16 16:41:03 2021 kern.warn kernel: [1614940.048748] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Sun May 16 16:41:03 2021 kern.warn kernel: [1614940.565868] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Sun May 16 16:41:04 2021 kern.warn kernel: [1614941.231764] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Sun May 16 16:41:04 2021 kern.warn kernel: [1614941.231819] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[... 497 lines cut ...]
Sun May 16 20:42:03 2021 kern.warn kernel: [1629400.501695] ath10k_pci 0000:01:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
Sun May 16 20:42:04 2021 kern.info kernel: [1629400.787511] ath10k_pci 0000:01:00.0: device successfully recovered
Sun May 16 20:42:04 2021 kern.warn kernel: [1629400.787621] ath10k_pci 0000:01:00.0: failed to transmit packet, dropping: -19
Sun May 16 20:42:04 2021 kern.warn kernel: [1629400.793110] ath10k_pci 0000:01:00.0: failed to submit frame: -19
Sun May 16 20:42:04 2021 kern.warn kernel: [1629400.800140] ath10k_pci 0000:01:00.0: failed to push frame: -19
[... cut ...]
Sun May 16 20:43:16 2021 daemon.notice hostapd: wlan0: AP-STA-CONNECTED f2:18:4f:4b:0d:26
Sun May 16 20:43:16 2021 daemon.info hostapd: wlan0: STA f2:18:4f:4b:0d:26 RADIUS: starting accounting session 906A8B1FB27D70AE
Sun May 16 20:43:16 2021 daemon.info hostapd: wlan0: STA f2:18:4f:4b:0d:26 WPA: pairwise key handshake completed (RSN)
Sun May 16 20:50:00 2021 cron.err crond[9143]: USER root pid 9749 cmd /usr/bin/wifi_schedule.sh recheck
Sun May 16 20:50:00 2021 cron.err crond[9806]: crond (busybox 1.31.1) started, log level 5

```
</details>

AndyLin2020, 2021-05-18:
我的正常，不过有个问题，如果设置了定时关WIFI后每次执行时就会重启，开WIFI没问题

icsfy, 2021-05-18:
系统日志补充

```
Tue May 18 18:19:17 2021 kern.warn kernel: [1793630.773531] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Tue May 18 18:19:17 2021 kern.warn kernel: [1793630.780953] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Tue May 18 18:19:18 2021 kern.warn kernel: [1793631.134450] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Tue May 18 18:19:18 2021 kern.warn kernel: [1793631.134526] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Tue May 18 18:19:19 2021 kern.warn kernel: [1793632.013142] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Tue May 18 18:19:19 2021 kern.warn kernel: [1793632.493382] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Tue May 18 18:19:20 2021 kern.warn kernel: [1793633.457546] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Tue May 18 18:19:20 2021 kern.err kernel: [1793633.841519] ath10k_pci 0000:01:00.0: Cannot communicate with firmware, previous wmi cmds: 40859:179336982 37004:179336949 36967:179336948 36965:179336384, jiffies: 179337288, attempting to fake crash and restart firmware, dev-flags: 0x42
Tue May 18 18:19:20 2021 kern.warn kernel: [1793633.841594] ath10k_pci 0000:01:00.0: failed to send wmi nop: -11
Tue May 18 18:19:20 2021 kern.warn kernel: [1793633.861640] ath10k_pci 0000:01:00.0: could not request stats (type -268435456 ret -108 specifier 1)
Tue May 18 18:19:20 2021 kern.err kernel: [1793633.868019] ath10k_pci 0000:01:00.0: firmware crashed! (guid aae248dd-66e9-4963-8588-89a73ce68d2c)
Tue May 18 18:19:20 2021 kern.info kernel: [1793633.877295] ath10k_pci 0000:01:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
Tue May 18 18:19:20 2021 kern.info kernel: [1793633.886321] ath10k_pci 0000:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
Tue May 18 18:19:20 2021 kern.info kernel: [1793633.899172] ath10k_pci 0000:01:00.0: firmware ver 10.4b-ct-9984-fW-13-5ae337bb1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 7ea63dc5
Tue May 18 18:19:20 2021 kern.info kernel: [1793633.911939] ath10k_pci 0000:01:00.0: board_file api 2 bmi_id 0:3 crc32 85498734
Tue May 18 18:19:20 2021 kern.warn kernel: [1793633.913275] ath10k_pci 0000:01:00.0: failed to set beacon mode for vdev 0: -108
Tue May 18 18:19:20 2021 kern.info kernel: [1793633.933298] ath10k_pci 0000:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
Tue May 18 18:19:20 2021 kern.err kernel: [1793633.933398] ath10k_pci 0000:01:00.0: firmware register dump:
Tue May 18 18:19:20 2021 kern.err kernel: [1793633.958538] ath10k_pci 0000:01:00.0: [00]: 0x0096068C 0x00400C00 0x00980000 0x009C20A0
Tue May 18 18:19:20 2021 kern.err kernel: [1793633.964272] ath10k_pci 0000:01:00.0: [04]: 0x009C1FC8 0x00961C1C 0x00961CB0 0x00961C64
Tue May 18 18:19:20 2021 kern.err kernel: [1793633.972503] ath10k_pci 0000:01:00.0: [08]: 0x00961C44 0x00962E70 0x00962D64 0x00962D4C
Tue May 18 18:19:20 2021 kern.err kernel: [1793633.980541] ath10k_pci 0000:01:00.0: [12]: 0x00961E04 0x00962EC4 0x00962EA4 0x00962EAC
Tue May 18 18:19:20 2021 kern.err kernel: [1793633.988644] ath10k_pci 0000:01:00.0: [16]: 0x00962EB4 0x00962EBC 0x00963350 0x0096308C
Tue May 18 18:19:21 2021 kern.err kernel: [1793633.996708] ath10k_pci 0000:01:00.0: [20]: 0x00963068 0x00962EEC 0x009630A0 0x00963004
Tue May 18 18:19:21 2021 kern.err kernel: [1793634.004790] ath10k_pci 0000:01:00.0: [24]: 0x00962F20 0x00963078 0x0096303C 0x009632F0
Tue May 18 18:19:21 2021 kern.err kernel: [1793634.012850] ath10k_pci 0000:01:00.0: [28]: 0x00963338 0x00963348 0x00963360 0x009631B4
Tue May 18 18:19:21 2021 kern.err kernel: [1793634.020905] ath10k_pci 0000:01:00.0: [32]: 0x009630EC 0x00962F80 0x009C2C3C 0x00962FDC
Tue May 18 18:19:21 2021 kern.err kernel: [1793634.028997] ath10k_pci 0000:01:00.0: [36]: 0x009636AC 0x00963C54 0x00963C9C 0x00963DB4
Tue May 18 18:19:21 2021 kern.err kernel: [1793634.037068] ath10k_pci 0000:01:00.0: [40]: 0x00963E08 0x00963E1C 0x00963E80 0x009606B8
Tue May 18 18:19:21 2021 kern.err kernel: [1793634.045140] ath10k_pci 0000:01:00.0: [44]: 0x009606CC 0x009C1CD4 0x00963F94 0x00963BC8
Tue May 18 18:19:21 2021 kern.err kernel: [1793634.053211] ath10k_pci 0000:01:00.0: [48]: 0x009638D4 0x00963508 0x00963834 0x00963858
Tue May 18 18:19:21 2021 kern.err kernel: [1793634.061283] ath10k_pci 0000:01:00.0: [52]: 0x0096386C 0x00963890 0x00973864 0x00973874
Tue May 18 18:19:21 2021 kern.err kernel: [1793634.069345] ath10k_pci 0000:01:00.0: [56]: 0x0097386C 0x009606C4 0x00963FA0 0x00963B50
Tue May 18 18:19:21 2021 kern.err kernel: [1793634.077431] ath10k_pci 0000:01:00.0: Copy Engine register dump:
Tue May 18 18:19:21 2021 kern.err kernel: [1793634.085512] ath10k_pci 0000:01:00.0: [00]: 0x0004a000  11  11   3   3
Tue May 18 18:19:21 2021 kern.err kernel: [1793634.091502] ath10k_pci 0000:01:00.0: [01]: 0x0004a400   1   1 455 456
Tue May 18 18:19:21 2021 kern.err kernel: [1793634.097903] ath10k_pci 0000:01:00.0: [02]: 0x0004a800  46  46  44  46
Tue May 18 18:19:21 2021 kern.err kernel: [1793634.104519] ath10k_pci 0000:01:00.0: [03]: 0x0004ac00  26  26  30  26
[... cut ...]
Tue May 18 18:19:32 2021 daemon.info hostapd: wlan0: STA d8:63:75:83:f0:2a WPA: pairwise key handshake completed (RSN)
Tue May 18 19:05:03 2021 kern.warn kernel: [1796376.062773] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Tue May 18 19:05:03 2021 kern.warn kernel: [1796376.063081] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Tue May 18 19:05:03 2021 kern.warn kernel: [1796376.069714] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
```

closed 2021-06-17 by coolsnowwolf