coolsnowwolf/lede #6844: 修改network 默认配置

lexin8, 2021-05-16:
想编译一个AP模式的固件，配置好了怎么编译出来就是配置好的状态？
network配置包括了网口跟vlan,

`config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fde5:fd52:4f6e::/48'

config device 'wan_eth0_1_dev'
	option name 'eth0.1'
	option macaddr '96:d1:25:ba:66:aa'

config interface 'lan'
	option type 'bridge'
	option _orig_ifname 'eth0.2 ra0 rai0'
	option _orig_bridge 'true'
	option proto 'dhcp'
	option ifname 'eth0.2 ra0 rai0 apcli0 apcli1 apclii0'

config device 'lan_eth0_2_dev'
	option name 'eth0.2'
	option macaddr '31:d1:43:ba:66:a9'

config device 'lan_ra0_dev'
	option name 'ra0'
	option macaddr '31:d1:43:ba:66:a9'

config device 'lan_rai0_dev'
	option name 'rai0'
	option macaddr '31:d1:43:ba:66:a9'

config switch
	option name 'switch0'
	option reset '1'
	option enable_vlan '1'

config switch_vlan
	option device 'switch0'
	option vlan '2'
	option vid '2'
	option ports '0 2 3 4 6t'
`

lphgor, 2021-05-16:
file大法，或者在 [zzz-default-settings](https://github.com/coolsnowwolf/lede/blob/master/package/lean/default-settings/files/zzz-default-settings) 里面用 uci 配置

lexin8, 2021-05-17:
> file大法，或者在 [zzz-default-settings](https://github.com/coolsnowwolf/lede/blob/master/package/lean/default-settings/files/zzz-default-settings) 里面用 uci 配置

file大法可以配置etc下的文件，其他路径貌似不生效

closed 2021-05-17 by lexin8