coolsnowwolf/lede #6858: wireguard编译错误

harleywong, 2021-05-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：wireguard编译错误

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：make[4]: Entering directory '/home/harley/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.120'
  CC [M]  /home/harley/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/wireguard-linux-compat-1.0.20210219/src/main.o
In file included from <command-line>:
/home/harley/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/wireguard-linux-compat-1.0.20210219/src/compat/compat.h:1059:22: error: static declaration of 'ip_tunnel_parse_protocol' follows non-static declaration
 static inline __be16 ip_tunnel_parse_protocol(const struct sk_buff *skb)
                      ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from ./include/net/gre.h:6,
                 from ./include/linux/netfilter/nf_conntrack_proto_gre.h:5,
                 from ./include/net/netfilter/nf_conntrack.h:23,
                 from /home/harley/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/wireguard-linux-compat-1.0.20210219/src/compat/compat.h:941,
                 from <command-line>:
./include/net/ip_tunnels.h:293:8: note: previous declaration of 'ip_tunnel_parse_protocol' was here
 __be16 ip_tunnel_parse_protocol(const struct sk_buff *skb);
        ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from <command-line>:
/home/harley/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/wireguard-linux-compat-1.0.20210219/src/compat/compat.h:1072:32: error: static declaration of 'ip_tunnel_header_ops' follows non-static declaration
 static const struct header_ops ip_tunnel_header_ops = { .parse_protocol = ip_tunnel_parse_protocol };
                                ^~~~~~~~~~~~~~~~~~~~
In file included from ./include/net/gre.h:6,
                 from ./include/linux/netfilter/nf_conntrack_proto_gre.h:5,
                 from ./include/net/netfilter/nf_conntrack.h:23,
                 from /home/harley/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/wireguard-linux-compat-1.0.20210219/src/compat/compat.h:941,
                 from <command-line>:
./include/net/ip_tunnels.h:292:32: note: previous declaration of 'ip_tunnel_header_ops' was here
 extern const struct header_ops ip_tunnel_header_ops;
                                ^~~~~~~~~~~~~~~~~~~~
make[5]: *** [scripts/Makefile.build:262: /home/harley/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/wireguard-linux-compat-1.0.20210219/src/main.o] Error 1
make[4]: *** [Makefile:1734: /home/harley/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/wireguard-linux-compat-1.0.20210219/src] Error 2
make[4]: Leaving directory '/home/harley/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.120'
make[3]: *** [Makefile:90: /home/harley/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/wireguard-linux-compat-1.0.20210219/.built] Error 2
make[3]: Leaving directory '/home/harley/openwrt/lede/package/network/services/wireguard'
time: package/network/services/wireguard/compile#2.15#0.89#2.99
make[2]: *** [package/Makefile:113: package/network/services/wireguard/compile] Error 2
make[2]: Leaving directory '/home/harley/openwrt/lede'
make[1]: *** [package/Makefile:107: /home/harley/openwrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/harley/openwrt/lede'
make: *** [/home/harley/openwrt/lede/include/toplevel.mk:224：world] 错误 2

elonj, 2021-05-20:
遇到同样的

xiangfeidexiaohuo, 2021-05-20:
同样问题

CuteTony, 2021-05-20:
同样问题

closed 2021-05-21 by harleywong