coolsnowwolf/lede #6869: 编译一直出错，这个问题持续了几个月了

yinyouzhu, 2021-05-22:
/home/xiong/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ar: array_type_info.o: No such file or directory
src/Makefile.in:44: recipe for target 'src/libuClibc++.a' failed
make[4]: *** [src/libuClibc++.a] Error 1
make[4]: Leaving directory '/home/xiong/lede/build_dir/target-x86_64_musl/uClibc++-0.2.5'
Makefile:97: recipe for target '/home/xiong/lede/build_dir/target-x86_64_musl/uClibc++-0.2.5/.built' failed
make[3]: *** [/home/xiong/lede/build_dir/target-x86_64_musl/uClibc++-0.2.5/.built] Error 2
make[3]: Leaving directory '/home/xiong/lede/package/libs/uclibc++'
time: package/libs/uclibc++/compile#0.05#0.01#0.17
package/Makefile:111: recipe for target 'package/libs/uclibc++/compile' failed
make[2]: *** [package/libs/uclibc++/compile] Error 2
make[2]: Leaving directory '/home/xiong/lede'
package/Makefile:107: recipe for target '/home/xiong/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/xiong/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/xiong/lede'
/home/xiong/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

Neverends4, 2021-05-22:
`make distclean`
全部删除重新pull一次，第一次编译用单线程，make menuconfig 只选机型不改任何默认配置，我就是这么解决的。第一次编译成功后再改配置。

closed 2021-05-22 by coolsnowwolf

miaozilong, 2023-04-11:
能把“x86_64-openwrt-linux-musl-gcc”文件分享一下吗？实在编译不下去了

nightan42643, 2023-08-10:
> 

这种方式和直接修改menuconfig到自己想要的状态然后编译，有什么区别吗？为什么这样就能解决问题呢...不是很理解