coolsnowwolf/lede #6873: Haproxy 无法开启

zfh18, 2021-05-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：5月23号编译版本，haproxy 无法开启

### (2) 路由器型号和固件版本

A：路由器 x86 ESXI，固件版本 OpenWrt R21.4.18 / LuCI Master (git-21.114.56541-974fb04)

### (3) 详细日志

A：Sun May 23 14:27:33 2021 kern.info kernel: [    0.320605] pci_bus 0000:02: resource 0 [io  0x2000-0x3fff]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.320707] pci_bus 0000:02: resource 1 [mem 0xfd500000-0xfdffffff]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.320819] pci_bus 0000:02: resource 2 [mem 0xe7b00000-0xe7ffffff 64bit pref]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.320950] pci_bus 0000:02: resource 4 [mem 0x000a0000-0x000bffff window]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.321072] pci_bus 0000:02: resource 5 [mem 0x000cc000-0x000cffff window]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.321195] pci_bus 0000:02: resource 6 [mem 0x000d0000-0x000d3fff window]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.321325] pci_bus 0000:02: resource 7 [mem 0x000d4000-0x000d7fff window]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.321448] pci_bus 0000:02: resource 8 [mem 0x000d8000-0x000dbfff window]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.321571] pci_bus 0000:02: resource 9 [mem 0xc0000000-0xfebfffff window]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.321698] pci_bus 0000:02: resource 10 [io  0x0000-0x0cf7 window]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.321811] pci_bus 0000:02: resource 11 [io  0x0d00-0xfeff window]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.321924] pci_bus 0000:03: resource 1 [mem 0xfd400000-0xfd4fffff]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.322038] pci_bus 0000:03: resource 2 [mem 0xc0000000-0xc01fffff 64bit pref]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.322170] pci_bus 0000:04: resource 0 [io  0x7000-0x7fff]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.322282] pci_bus 0000:04: resource 1 [mem 0xfd000000-0xfd0fffff]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.322395] pci_bus 0000:04: resource 2 [mem 0xe7700000-0xe77fffff 64bit pref]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.322527] pci_bus 0000:05: resource 0 [io  0xb000-0xbfff]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.322628] pci_bus 0000:05: resource 1 [mem 0xfcc00000-0xfccfffff]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.322741] pci_bus 0000:05: resource 2 [mem 0xe7300000-0xe73fffff 64bit pref]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.322873] pci_bus 0000:06: resource 1 [mem 0xfc800000-0xfc8fffff]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.322986] pci_bus 0000:06: resource 2 [mem 0xe6f00000-0xe6ffffff 64bit pref]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.323117] pci_bus 0000:07: resource 1 [mem 0xfc400000-0xfc4fffff]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.323230] pci_bus 0000:07: resource 2 [mem 0xe6b00000-0xe6bfffff 64bit pref]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.323361] pci_bus 0000:08: resource 1 [mem 0xfc000000-0xfc0fffff]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.323474] pci_bus 0000:08: resource 2 [mem 0xe6700000-0xe67fffff 64bit pref]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.323604] pci_bus 0000:09: resource 1 [mem 0xfbc00000-0xfbcfffff]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.323717] pci_bus 0000:09: resource 2 [mem 0xe6300000-0xe63fffff 64bit pref]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.323848] pci_bus 0000:0a: resource 1 [mem 0xfb800000-0xfb8fffff]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.323961] pci_bus 0000:0a: resource 2 [mem 0xe5f00000-0xe5ffffff 64bit pref]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.324093] pci_bus 0000:0b: resource 0 [io  0x4000-0x4fff]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.324194] pci_bus 0000:0b: resource 1 [mem 0xfd300000-0xfd3fffff]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.324307] pci_bus 0000:0b: resource 2 [mem 0xe7a00000-0xe7afffff 64bit pref]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.324438] pci_bus 0000:0c: resource 0 [io  0x8000-0x8fff]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.324540] pci_bus 0000:0c: resource 1 [mem 0xfcf00000-0xfcffffff]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.324653] pci_bus 0000:0c: resource 2 [mem 0xe7600000-0xe76fffff 64bit pref]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.324787] pci_bus 0000:0d: resource 0 [io  0xc000-0xcfff]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.324889] pci_bus 0000:0d: resource 1 [mem 0xfcb00000-0xfcbfffff]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.325002] pci_bus 0000:0d: resource 2 [mem 0xe7200000-0xe72fffff 64bit pref]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.325132] pci_bus 0000:0e: resource 1 [mem 0xfc700000-0xfc7fffff]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.325245] pci_bus 0000:0e: resource 2 [mem 0xe6e00000-0xe6efffff 64bit pref]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.325376] pci_bus 0000:0f: resource 1 [mem 0xfc300000-0xfc3fffff]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.325489] pci_bus 0000:0f: resource 2 [mem 0xe6a00000-0xe6afffff 64bit pref]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.325620] pci_bus 0000:10: resource 1 [mem 0xfbf00000-0xfbffffff]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.325735] pci_bus 0000:10: resource 2 [mem 0xe6600000-0xe66fffff 64bit pref]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.325866] pci_bus 0000:11: resource 1 [mem 0xfbb00000-0xfbbfffff]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.325979] pci_bus 0000:11: resource 2 [mem 0xe6200000-0xe62fffff 64bit pref]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.326110] pci_bus 0000:12: resource 1 [mem 0xfb700000-0xfb7fffff]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.326223] pci_bus 0000:12: resource 2 [mem 0xe5e00000-0xe5efffff 64bit pref]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.326367] pci_bus 0000:13: resource 0 [io  0x5000-0x5fff]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.326469] pci_bus 0000:13: resource 1 [mem 0xfd200000-0xfd2fffff]
Sun May 23 14:27:33 2021 kern.info kernel: [    0.326582] pci_bus 0000:13: resource 2 [mem 0xe7900000-0xe79fffff 64bit pref]
[... 557 lines cut ...]
Sun May 23 14:36:42 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun May 23 14:36:42 2021 daemon.info dnsmasq[28797]: Connected to system UBus
Sun May 23 14:36:42 2021 cron.err crond[28853]: crond (busybox 1.31.1) started, log level 5
Sun May 23 14:36:51 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun May 23 14:36:51 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun May 23 14:36:51 2021 daemon.info dnsmasq[29991]: Connected to system UBus
Sun May 23 14:40:00 2021 cron.err crond[28853]: USER root pid 34736 cmd /etc/init.d/frp restart
Sun May 23 14:40:00 2021 user.notice FRPC: Shutting down frp service
Sun May 23 14:40:00 2021 user.notice FRPC: Starting frp service
Sun May 23 14:46:28 2021 user.notice haproxy: is initializing enabled is 1
Sun May 23 14:46:29 2021 user.notice restarting: haproxy
Sun May 23 14:46:29 2021 daemon.notice procd: /etc/init.d/haproxy: Syntax: /etc/init.d/haproxy [command]
Sun May 23 14:46:29 2021 daemon.notice procd: /etc/init.d/haproxy:
Sun May 23 14:46:29 2021 daemon.notice procd: /etc/init.d/haproxy: Available commands:
Sun May 23 14:46:29 2021 daemon.notice procd: /etc/init.d/haproxy: 	start           Start the service
Sun May 23 14:46:29 2021 daemon.notice procd: /etc/init.d/haproxy: 	stop            Stop the service
Sun May 23 14:46:29 2021 daemon.notice procd: /etc/init.d/haproxy: 	restart         Restart the service
Sun May 23 14:46:29 2021 daemon.notice procd: /etc/init.d/haproxy: 	reload          Reload configuration files (or restart if service does not implement reload)
Sun May 23 14:46:29 2021 daemon.notice procd: /etc/init.d/haproxy: 	enable          Enable service autostart
Sun May 23 14:46:29 2021 daemon.notice procd: /etc/init.d/haproxy: 	disable         Disable service autostart
Sun May 23 14:46:29 2021 daemon.notice procd: /etc/init.d/haproxy: 	enabled         Check if service is started on boot
Sun May 23 14:46:29 2021 daemon.notice procd: /etc/init.d/haproxy:
Sun May 23 14:46:44 2021 user.notice haproxy: is initializing enabled is
Sun May 23 14:46:44 2021 user.notice alex: stopping haproxy
Sun May 23 14:46:54 2021 daemon.info procd: Instance haproxy::instance1 s in a crash loop 6 crashes, 0 seconds since last crash

redjumper, 2021-05-25:
也有这个问题
`Tue May 25 11:23:21 2021 daemon.info procd: Instance haproxy::instance1 s in a crash loop 6 crashes, 0 seconds since last crash`
同样的配置在之前版本没有问题

hcym, 2021-05-26:
好像是，什么情况发生🙄

hcym, 2021-05-27:
用21.02源码编译lean科学功能正常haproxy，蛮好，用ipv6的节点负载均衡也方便😂

zfh18, 2021-05-30:
R21.6.1 已解决

closed 2021-05-30 by zfh18