coolsnowwolf/lede #6877: 编译时貌似驱动出错

DINKHo, 2021-05-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：make menuconfig 只改了设备选项，其他全部都默认
Target System (MediaTek Ralink MIPS)  --->
Subtarget (MT7621 based boards)  --->
Target Profile (Newifi D2)  --->

### (3) 详细日志

A：貌似编译驱动时出错？

  CC      drivers/net/ethernet/mediatek/mt7530.o
drivers/net/ethernet/mediatek/mt7530.c: In function 'mt7530_r32':
drivers/net/ethernet/mediatek/mt7530.c:327:3: error: implicit declaration of function '__mdiobus_write'; did you mean 'mdiobus_write'? [-Werror=implicit-function-declaration]
   __mdiobus_write(priv->bus, 0x1f, 0x1f, (reg >> 6) & 0x3ff);
   ^~~~~~~~~~~~~~~
   mdiobus_write
drivers/net/ethernet/mediatek/mt7530.c:328:9: error: implicit declaration of function '__mdiobus_read'; did you mean 'mdiobus_read'? [-Werror=implicit-function-declaration]
   low = __mdiobus_read(priv->bus, 0x1f, (reg >> 2) & 0xf);
         ^~~~~~~~~~~~~~
         mdiobus_read
cc1: some warnings being treated as errors
scripts/Makefile.build:329: recipe for target 'drivers/net/ethernet/mediatek/mt7530.o' failed
make[9]: *** [drivers/net/ethernet/mediatek/mt7530.o] Error 1
scripts/Makefile.build:588: recipe for target 'drivers/net/ethernet/mediatek' failed
make[8]: *** [drivers/net/ethernet/mediatek] Error 2
scripts/Makefile.build:588: recipe for target 'drivers/net/ethernet' failed
make[7]: *** [drivers/net/ethernet] Error 2
scripts/Makefile.build:588: recipe for target 'drivers/net' failed
make[6]: *** [drivers/net] Error 2
Makefile:1054: recipe for target 'drivers' failed
make[5]: *** [drivers] Error 2
make[5]: Leaving directory '/home/dink/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.231'
Makefile:26: recipe for target '/home/dink/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.231/.image' failed
make[4]: *** [/home/dink/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.231/.image] Error 2
make[4]: Leaving directory '/home/dink/openwrt/target/linux/ramips'
Makefile:13: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[3]: Leaving directory '/home/dink/openwrt/target/linux'
time: target/linux/install#2.73#2.19#4.86
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: *** [target/linux/install] Error 2
make[2]: Leaving directory '/home/dink/openwrt'
target/Makefile:19: recipe for target '/home/dink/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.target_install' failed
make[1]: *** [/home/dink/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/dink/openwrt'
/home/dink/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

AmadeusGhost, 2021-05-27:
这里已经不支持编译 4.14 了。

DINKHo, 2021-05-27:
> 这里已经不支持编译 4.14 了。

这是编译 https://github.com/coolsnowwolf/openwrt 稳定版的错误讯息

closed 2021-06-17 by coolsnowwolf

zhangsiyu0, 2021-07-01:
> 反馈bug/问题模板，提建议请删除
> 
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
> 
> * [ x] 没有类似的issue
> 
> ## 2. 详细叙述
> ### (1) 具体问题
> A：这是编译 https://github.com/coolsnowwolf/openwrt 稳定版的错误讯息
> make menuconfig 只改了设备选项，其他全部都默认
> Target System (MediaTek Ralink MIPS) --->
> Subtarget (MT7621 based boards) --->
> Target Profile (Newifi D2) --->
> 
> ### (3) 详细日志
> A：貌似编译驱动时出错？
> 
> CC drivers/net/ethernet/mediatek/mt7530.o
> drivers/net/ethernet/mediatek/mt7530.c: In function 'mt7530_r32':
> drivers/net/ethernet/mediatek/mt7530.c:327:3: error: implicit declaration of function '__mdiobus_write'; did you mean 'mdiobus_write'? [-Werror=implicit-function-declaration]
> __mdiobus_write(priv->bus, 0x1f, 0x1f, (reg >> 6) & 0x3ff);
> ^~~~~~~~~~~~~~~
> mdiobus_write
> drivers/net/ethernet/mediatek/mt7530.c:328:9: error: implicit declaration of function '__mdiobus_read'; did you mean 'mdiobus_read'? [-Werror=implicit-function-declaration]
> low = __mdiobus_read(priv->bus, 0x1f, (reg >> 2) & 0xf);
> ^~~~~~~~~~~~~~
> mdiobus_read
> cc1: some warnings being treated as errors
> scripts/Makefile.build:329: recipe for target 'drivers/net/ethernet/mediatek/mt7530.o' failed
> make[9]: *** [drivers/net/ethernet/mediatek/mt7530.o] Error 1
> scripts/Makefile.build:588: recipe for target 'drivers/net/ethernet/mediatek' failed
> make[8]: *** [drivers/net/ethernet/mediatek] Error 2
> scripts/Makefile.build:588: recipe for target 'drivers/net/ethernet' failed
> make[7]: *** [drivers/net/ethernet] Error 2
> scripts/Makefile.build:588: recipe for target 'drivers/net' failed
> make[6]: *** [drivers/net] Error 2
> Makefile:1054: recipe for target 'drivers' failed
> make[5]: *** [drivers] Error 2
> make[5]: Leaving directory '/home/xxxx/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.231'
> Makefile:26: recipe for target '/home/xxxx/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.231/.image' failed
> make[4]: *** [/home/xxxx/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.231/.image] Error 2
> make[4]: Leaving directory '/home/xxxx/openwrt/target/linux/ramips'
> Makefile:13: recipe for target 'install' failed
> make[3]: *** [install] Error 2
> make[3]: Leaving directory '/home/xxxx/openwrt/target/linux'
> time: target/linux/install#2.73#2.19#4.86
> target/Makefile:23: recipe for target 'target/linux/install' failed
> make[2]: *** [target/linux/install] Error 2
> make[2]: Leaving directory '/home/xxxx/openwrt'
> target/Makefile:19: recipe for target '/home/xxxx/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.target_install' failed
> make[1]: *** [/home/xxxx/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
> make[1]: Leaving directory '/home/xxxx/openwrt'
> /home/xxxx/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
> make: *** [world] Error 2

你好，我编译稳定版时遇到同样的错误，请问如何解决。