coolsnowwolf/lede #6881: 竞斗云没有频率调节页面，请问是缺了哪个模块？

Cathgao, 2021-05-25:
以前刚有超频的时候，记得是系统分类还是服务分类有个CPU频率调节，能选调度器和最高最低频率的，现在不见了。请问是少了哪个模块没选上吗？还是说已经把这个功能去掉了？
`CONFIG_PACKAGE_luci-app-cpufreq` 这个在menuconfig里搜索是没有的，配置文件里姑且是把他写上了

配置文件如下：
```
#
# 机型选择
#
CONFIG_TARGET_ipq40xx=y
CONFIG_TARGET_ipq40xx_generic_DEVICE_p2w_r619ac-128m=y

#
# IPV6支持
#
CONFIG_IPV6=y
CONFIG_PACKAGE_dnsmasq_full_dhcpv6=y
CONFIG_PACKAGE_ipv6helper=y
CONFIG_PACKAGE_odhcp6c=y
CONFIG_PACKAGE_odhcpd-ipv6only=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_kmod-ipt-nat6=y
CONFIG_PACKAGE_ipv6helper=y
CONFIG_PACKAGE_kmod-ip6tables=y
CONFIG_PACKAGE_kmod-ip6tables-extra=y
CONFIG_PACKAGE_6in4=y

#
# samba扩展
#
CONFIG_PACKAGE_autosamba=n #与CIFS冲突，禁用
#CONFIG_PACKAGE_samba36-server is not set

#
# USB与存储相关支持
#
CONFIG_PACKAGE_automount=y
CONFIG_PACKAGE_kmod-fs-qntfs=y
CONFIG_PACKAGE_kmod-scsi-core=y
CONFIG_PACKAGE_kmod-fs-exfat=y
CONFIG_PACKAGE_kmod-fs-ext4=y
CONFIG_PACKAGE_ntfs-3g=n
CONFIG_PACKAGE_kmod-fs-vfat=y
CONFIG_PACKAGE_kmod-fuse=y
CONFIG_PACKAGE_kmod-mtd-rw=y
CONFIG_PACKAGE_kmod-mmc=y
CONFIG_PACKAGE_kmod-sdhci=y
CONFIG_PACKAGE_kmod-mmc-spi=y
CONFIG_PACKAGE_kmod-usb-core=y
CONFIG_PACKAGE_kmod-usb-dwc3=n
CONFIG_PACKAGE_kmod-usb-dwc3-of-simple=y
CONFIG_PACKAGE_kmod-usb-ehci=y
CONFIG_PACKAGE_kmod-usb-ohci=y
CONFIG_PACKAGE_kmod-usb-printer=y
CONFIG_PACKAGE_kmod-usb-storage=y
CONFIG_PACKAGE_kmod-usb-storage-extras=y
CONFIG_PACKAGE_kmod-usb-uhci=y
CONFIG_PACKAGE_kmod-usb2=y
CONFIG_PACKAGE_kmod-usb3=y

#
# 无线驱动
#
CONFIG_PACKAGE_ipq-wifi-p2w_r619ac=y
CONFIG_PACKAGE_ath10k-firmware-qca4019-ct-htt=y
CONFIG_PACKAGE_ath10k-firmware-qca4019-ct=n
CONFIG_PACKAGE_ath10k-firmware-qca4019=n
CONFIG_PACKAGE_wireless-regdb=y
CONFIG_PACKAGE_kmod-ath=y
CONFIG_ATH_USER_REGD=y
CONFIG_PACKAGE_ATH_DFS=y
CONFIG_PACKAGE_kmod-ath10k-ct=y
CONFIG_PACKAGE_kmod-ath10k=n
#CONFIG_PACKAGE_kmod-ath10k-ct-smallbuffers=n
CONFIG_ATH10K_LEDS=y
CONFIG_ATH10K_THERMAL=y
CONFIG_PACKAGE_kmod-cfg80211=y
CONFIG_PACKAGE_kmod-mac80211=y
CONFIG_PACKAGE_MAC80211_DEBUGFS=y
CONFIG_PACKAGE_MAC80211_MESH=y

#
# 插件集成
#
CONFIG_PACKAGE_xray=y
CONFIG_XRAY_EXCLUDE_ASSETS=y
CONFIG_XRAY_COMPRESS_UPX=y
CONFIG_v2ray-plugin_INCLUDE_GOPROXY=y
CONFIG_PACKAGE_kcptun-client=y
CONFIG_PACKAGE_naiveproxy=y
CONFIG_PACKAGE_luci-app-uugamebooster=n #默认不启用uu加速器
CONFIG_PACKAGE_uugamebooster=n
CONFIG_PACKAGE_luci-app-jd-dailybonus=y
CONFIG_PACKAGE_luci-app-netdata=y
CONFIG_PACKAGE_luci-i18n-netdata-zh-cn=y
CONFIG_PACKAGE_luci-app-accesscontrol=n
CONFIG_PACKAGE_luci-app-adbyby-plus=n #被AdGuard Home取代
CONFIG_PACKAGE_luci-app-adguardhome=y #取代adbyby+
CONFIG_PACKAGE_luci-app-aria2=n
CONFIG_PACKAGE_luci-app-arpbind=y
CONFIG_PACKAGE_luci-app-autoreboot=y
CONFIG_PACKAGE_luci-app-cifsd=y
CONFIG_PACKAGE_luci-app-cifs-mount=y
CONFIG_PACKAGE_luci-app-cpufreq=y
CONFIG_PACKAGE_luci-app-ddns=y
CONFIG_PACKAGE_luci-app-dnspod=y
CONFIG_PACKAGE_luci-app-dnsforwarder=n #禁用
CONFIG_PACKAGE_luci-app-filetransfer=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-flowoffload=n
CONFIG_PACKAGE_luci-app-frpc=y
CONFIG_PACKAGE_luci-app-ipsec-vpnd=n
CONFIG_PACKAGE_luci-app-mwan3=n
CONFIG_PACKAGE_luci-app-mwan3helper=n
CONFIG_PACKAGE_luci-app-nlbwmon=n
CONFIG_PACKAGE_luci-app-pptp-server=n
CONFIG_PACKAGE_luci-app-ramfree=y
CONFIG_PACKAGE_luci-app-samba=n #与CIFS冲突，禁用
CONFIG_PACKAGE_luci-app-serverchan=y
CONFIG_PACKAGE_luci-app-sqm=y
CONFIG_PACKAGE_luci-app-ssr-plus=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Xray=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_NaiveProxy=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_ShadowsocksR_Server=y
CONFIG_PACKAGE_luci-app-syncdial=n
CONFIG_PACKAGE_luci-app-transmission=n
CONFIG_PACKAGE_luci-app-ttyd=n
CONFIG_PACKAGE_luci-app-uhttpd=y
CONFIG_PACKAGE_luci-app-unblockmusic=y
CONFIG_PACKAGE_luci-app-upnp=y
CONFIG_PACKAGE_luci-app-usb-printer=y
CONFIG_PACKAGE_luci-app-v2ray-server=y
CONFIG_PACKAGE_luci-app-vlmcsd=y
CONFIG_PACKAGE_luci-app-vsftpd=y
CONFIG_PACKAGE_luci-app-wifischedule=y
CONFIG_PACKAGE_luci-app-wol=y
CONFIG_PACKAGE_luci-app-xlnetacc=n
CONFIG_PACKAGE_luci-app-zerotier=y
CONFIG_PACKAGE_luci-app-diskman=y
CONFIG_PACKAGE_luci-app-music-remote-center=y
CONFIG_PACKAGE_ddns-scripts_cloudflare.com-v4=y
CONFIG_PACKAGE_luci-app-rclone=y
CONFIG_PACKAGE_luci-app-rclone_INCLUDE_rclone-webui=y
CONFIG_PACKAGE_luci-app-rclone_INCLUDE_rclone-ng=y
CONFIG_PACKAGE_luci-app-rclone_INCLUDE_fuse-utils=y
# CONFIG_PACKAGE_qBittorrent is not set
# CONFIG_DEFAULT_luci-app-qbittorrent is not set

#
# 主题
#
CONFIG_PACKAGE_luci-theme-argon=y
CONFIG_PACKAGE_luci-app-argon-config=y
CONFIG_PACKAGE_luci-theme-bootstrap=y

#
# 基础命令及其他
#
CONFIG_PACKAGE_openssh-sftp-server=y
CONFIG_PACKAGE_nano=y
CONFIG_PACKAGE_fdisk=y
CONFIG_PACKAGE_bash=y
CONFIG_PACKAGE_lscpu=y
CONFIG_PACKAGE_mount-utils=y
```

Boos4721, 2021-05-26:
https://github.com/coolsnowwolf/lede/tree/master/package/lean/luci-app-cpufreq

qpzr, 2021-06-02:
> > https://github.com/coolsnowwolf/lede/tree/master/package/lean/luci-app-cpufreq
> 
> CONFIG_PACKAGE_luci-app-cpufreq=y

https://github.com/coolsnowwolf/lede/blob/master/target/linux/ipq40xx/Makefile
[image]

Cathgao, 2021-06-03:
> > > https://github.com/coolsnowwolf/lede/tree/master/package/lean/luci-app-cpufreq
> > 
> > 
> > CONFIG_PACKAGE_luci-app-cpufreq=y
> 
> https://github.com/coolsnowwolf/lede/blob/master/target/linux/ipq40xx/Makefile
> [image]

[image]

closed 2021-06-03 by Cathgao