coolsnowwolf/lede #6888: 首次编译 执行 ./scripts/feeds install -a之后 出现错误

sayakaextra, 2021-05-27:
Q：是否搜索了issue (使用 "x" 选择)

* [ √] 没有类似的issue

## 2. 详细叙述
最新版源码

### (1) 具体问题

A：
首次编译
执行 ./scripts/feeds install -a之后
出现错误

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

A：WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libpng', which does not exist
WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libupnp', which does not exist
WARNING: Makefile 'package/lean/amule/Makefile' has a build dependency on 'libgd', which does not exist
WARNING: Makefile 'package/lean/antileech/Makefile' has a build dependency on 'libgd', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'lm-sensors', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'bc', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'lm-sensors', which does not exist
WARNING: Makefile 'package/lean/automount/Makefile' has a dependency on 'antfs-mount', which does not exist
WARNING: Makefile 'package/lean/autosamba/Makefile' has a dependency on 'luci-app-samba', which does not exist
WARNING: Makefile 'package/lean/autosamba/Makefile' has a dependency on 'wsdd2', which does not exist
WARNING: Makefile 'package/lean/baidupcs-web/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libgnutls', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libzstd', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libopenldap', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libidn2', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libssh2', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_aliyun/Makefile' has a dependency on 'ddns-scripts', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_dnspod/Makefile' has a dependency on 'ddns-scripts', which does not exist
WARNING: Makefile 'package/lean/default-settings/Makefile' has a dependency on 'luci-base', which does not exist
WARNING: Makefile 'package/lean/default-settings/Makefile' has a dependency on 'luci', which does not exist
WARNING: Makefile 'package/lean/default-settings/Makefile' has a dependency on 'luci-compat', which does not exist
WARNING: Makefile 'package/lean/dnsforwarder/Makefile' has a dependency on 'coreutils-base64', which does not exist
WARNING: Makefile 'package/lean/frp/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/gmediarender/Makefile' has a dependency on 'gstreamer1', which does not exist
WARNING: Makefile 'package/lean/gmediarender/Makefile' has a dependency on 'libgstreamer1', which does not exist
WARNING: Makefile 'package/lean/gmediarender/Makefile' has a dependency on 'gstreamer1-libs', which does not exist
WARNING: Makefile 'package/lean/gmediarender/Makefile' has a dependency on 'glib2', which does not exist
WARNING: Makefile 'package/lean/gmediarender/Makefile' has a dependency on 'libupnp', which does not exist
WARNING: Makefile 'package/lean/ipv6-helper/Makefile' has a dependency on 'luci-proto-ipv6', which does not exist
WARNING: Makefile 'package/lean/kcptun/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/boot/kexec-tools/Makefile' has a dependency on 'liblzma', which does not exist
WARNING: Makefile 'package/libs/libselinux/Makefile' has a dependency on 'musl-fts', which does not exist
WARNING: Makefile 'package/network/services/lldpd/Makefile' has a dependency on 'libnetsnmp', which does not exist
WARNING: Makefile 'package/lean/luci-app-accesscontrol/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-adbyby-plus/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-airplay2/Makefile' has a dependency on 'alsa-utils', which does not exist
WARNING: Makefile 'package/lean/luci-app-airplay2/Makefile' has a dependency on 'shairport-sync-openssl', which does not exist
WARNING: Makefile 'package/lean/luci-app-airplay2/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-amule/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-arpbind/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-autoreboot/Makefile' has a dependency on 'luci', which does not exist
WARNING: Makefile 'package/lean/luci-app-autoreboot/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-baidupcs-web/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-cifs-mount/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-cifsd/Makefile' has a dependency on 'ksmbd-server', which does not exist
WARNING: Makefile 'package/lean/luci-app-cifsd/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-cpufreq/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-diskman/Makefile' has a dependency on 'smartmontools', which does not exist
[... 113 lines cut ...]
WARNING: Makefile 'package/lean/samba4/Makefile' has a dependency on 'attr', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a dependency on 'libaio', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a dependency on 'libavahi-client', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a dependency on 'python3-cryptodome', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a dependency on 'libopenldap', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a dependency on 'libarchive', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a dependency on 'acl', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a dependency on 'attr', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a build dependency on 'libtasn1/host', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a build dependency on 'rpcsvc-proto/host', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a build dependency on 'python3/host', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a build dependency on 'perl/host', which does not exist
WARNING: Makefile 'package/lean/simple-obfs/Makefile' has a dependency on 'libev', which does not exist
WARNING: Makefile 'package/lean/simple-obfs/Makefile' has a dependency on 'libev', which does not exist
WARNING: Makefile 'package/lean/simple-obfs/Makefile' has a build dependency on 'libev', which does not exist
WARNING: Makefile 'package/lean/tcpping/Makefile' has a dependency on 'libnet-1.2.x', which does not exist
WARNING: Makefile 'package/lean/trojan/Makefile' has a dependency on 'boost', which does not exist
WARNING: Makefile 'package/lean/trojan/Makefile' has a dependency on 'boost-system', which does not exist
WARNING: Makefile 'package/lean/trojan/Makefile' has a dependency on 'boost-program_options', which does not exist
WARNING: Makefile 'package/lean/trojan/Makefile' has a dependency on 'boost-date_time', which does not exist
WARNING: Makefile 'package/lean/UnblockNeteaseMusic/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/lean/UnblockNeteaseMusicGo/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/wxbase/Makefile' has a dependency on 'libexpat', which does not exist

帮忙请求

EnnawYang, 2021-05-27:
无需理会，下面会进行安装。另：善用搜索issue功能

sayakaextra, 2021-05-27:
> 无需理会，下面会进行安装。另：善用搜索issue功能

好的谢谢

Cathgao, 2021-05-27:
warning：
n. 警告；预兆；预告
adj. 警告的；引以为戒的
v. 警告（warn的ing形式）
n. (Warning)人名；(德)瓦宁

luii2020, 2021-06-04:
升级下依赖吧，命令行输入 sudo apt-get update ，然后输入 sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3 python2.7 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib antlr3 gperf wget curl swig rsync

closed 2021-06-17 by coolsnowwolf