coolsnowwolf/lede #6907: 使用SDK编译openssl出现devcrypto.so No such file or directory

shenghuang147, 2021-05-29:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
使用潘多拉SDK 编译OpenSSL 
因为潘多拉是我目前能找到的16M 能支持USB的固件，且无线非常稳定
但是插件版本低，所以我选择使用使用SDK编译

我在编译 **`OpenSSL`** 的时候提示 没有这个文件

**`build_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/openssl-1.1.1k/ipkg-install/usr/lib/engines-1.1/devcrypto.so'`**

在网上多方搜寻无果后只能来此询问

### (1) 具体问题

A：使用SDK编译 `openssl` 时出现 `install: cannot stat '/home/psheng/Downloads/PandoraBox-SDK/build_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/openssl-1.1.1k/ipkg-install/usr/lib/engines-1.1/devcrypto.so': No such file or directory`

### (2) 路由器型号和固件版本

A：路由器型号：**K2P**   固件版本： **PandoraBox R8.1.12**

### (3) 详细日志

A：

````bash
touch /home/psheng/Downloads/PandoraBox-SDK/staging_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/root-ralink/stamp/.libopenssl-conf_installed
if [ -f /home/psheng/Downloads/PandoraBox-SDK/staging_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/pkginfo/openssl.default.install.clean ]; then rm -f /home/psheng/Downloads/PandoraBox-SDK/staging_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/pkginfo/openssl.default.install /home/psheng/Downloads/PandoraBox-SDK/staging_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/pkginfo/openssl.default.install.clean; fi
echo "libopenssl-conf" >> /home/psheng/Downloads/PandoraBox-SDK/staging_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/pkginfo/openssl.default.install
if [ -f /home/psheng/Downloads/PandoraBox-SDK/staging_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/pkginfo/openssl.default.install.clean ]; then rm -f /home/psheng/Downloads/PandoraBox-SDK/staging_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/pkginfo/openssl.default.install /home/psheng/Downloads/PandoraBox-SDK/staging_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/pkginfo/openssl.default.install.clean; fi
mkdir -p /home/psheng/Downloads/PandoraBox-SDK/bin/targets/ralink/mt7621/packages /home/psheng/Downloads/PandoraBox-SDK/build_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/openssl-1.1.1k/ipkg-mipsel_1004kc_dsp/libopenssl-devcrypto/CONTROL /home/psheng/Downloads/PandoraBox-SDK/staging_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/pkginfo
install -d -m0755 /home/psheng/Downloads/PandoraBox-SDK/build_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/openssl-1.1.1k/ipkg-mipsel_1004kc_dsp/libopenssl-devcrypto/usr/lib/engines-1.1
install -m0755 /home/psheng/Downloads/PandoraBox-SDK/build_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/openssl-1.1.1k/ipkg-install/usr/lib/engines-1.1/devcrypto.so /home/psheng/Downloads/PandoraBox-SDK/build_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/openssl-1.1.1k/ipkg-mipsel_1004kc_dsp/libopenssl-devcrypto/usr/lib/engines-1.1
install: cannot stat '/home/psheng/Downloads/PandoraBox-SDK/build_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/openssl-1.1.1k/ipkg-install/usr/lib/engines-1.1/devcrypto.so': No such file or directory
Makefile:413: recipe for target '/home/psheng/Downloads/PandoraBox-SDK/bin/packages/mipsel_1004kc_dsp/base/libopenssl-devcrypto_1.1.1k-1_mipsel_1004kc_dsp.ipk' failed
make[2]: *** [/home/psheng/Downloads/PandoraBox-SDK/bin/packages/mipsel_1004kc_dsp/base/libopenssl-devcrypto_1.1.1k-1_mipsel_1004kc_dsp.ipk] Error 1
make[2]: Leaving directory '/home/psheng/Downloads/PandoraBox-SDK/package/libs/openssl'
package/Makefile:105: recipe for target 'package/libs/openssl/compile' failed
make[1]: *** [package/libs/openssl/compile] Error 2
make[1]: Leaving directory '/home/psheng/Downloads/PandoraBox-SDK'
/home/psheng/Downloads/PandoraBox-SDK/include/toplevel.mk:192: recipe for target 'package/libs/openssl/compile' failed
make: *** [package/libs/openssl/compile] Error 2
````

Cathgao, 2021-05-30:
PandoraBox报的错跟我lede有什么关系呢？

shenghuang147, 2021-05-30:
> PandoraBox报的错跟我lede有什么关系呢？

>  首先Pandorabox，与openwrt同源，所以处理异常的方法理论上也是互通的，其次你觉得不关你的事，你可以选择不回答。谢谢。

EnnawYang, 2021-05-30:
所以PandoraBox和这边有啥关系？

Cathgao, 2021-05-30:
> > PandoraBox报的错跟我lede有什么关系呢？
> 
> > 首先Pandorabox，与openwrt同源，所以处理异常的方法理论上也是互通的，其次你觉得不关你的事，你可以选择不回答。谢谢。

笑死，那你买个大众去找宝马给你售后呗，还得是免费。反正都是德国牌子，符合你说的同源，是这个道理吧？
告诉你来错地方了还理直气壮回怼，看有没有人回答你就完事了

shenghuang147, 2021-05-30:
问题已经解决。

解决方式：

````bash
ln -s /home/psheng/Downloads/test_pandorabox/PandoraBox-SDK/build_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/openssl-1.1.1k/ipkg-mipsel_1004kc_dsp/libopenssl/usr/lib/libcrypto.so.1.1 /home/psheng/Downloads/test_pandorabox/PandoraBox-SDK/build_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/openssl-1.1.1k/ipkg-install/usr/lib/engines-1.1/devcrypto.so
````

解决思路：

+ 将`build_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/openssl-1.1.1k/ipkg-mipsel_1004kc_dsp/libopenssl/usr/lib` 下的 `libcrypto.so.1.1` 做个软连接到`build_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/openssl-1.1.1k/ipkg-install/usr/lib/engines-1.1` 下的 `devcrypto.so`

不同cpu架构和版本，文件夹的名称不一样，仅供参考。

closed 2021-05-30 by shenghuang147

ashesofdream, 2021-11-18:
> 问题已经解决。
> 
> 解决方式：
> 
> ```shell
> ln -s /home/psheng/Downloads/test_pandorabox/PandoraBox-SDK/build_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/openssl-1.1.1k/ipkg-mipsel_1004kc_dsp/libopenssl/usr/lib/libcrypto.so.1.1 /home/psheng/Downloads/test_pandorabox/PandoraBox-SDK/build_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/openssl-1.1.1k/ipkg-install/usr/lib/engines-1.1/devcrypto.so
> ```
> 
> 解决思路：
> 
> * 将`build_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/openssl-1.1.1k/ipkg-mipsel_1004kc_dsp/libopenssl/usr/lib` 下的 `libcrypto.so.1.1` 做个软连接到`build_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/openssl-1.1.1k/ipkg-install/usr/lib/engines-1.1` 下的 `devcrypto.so`
> 
> 不同cpu架构和版本，文件夹的名称不一样，仅供参考。

软连接实测不行 要硬链接