coolsnowwolf/lede #6908: 如何修改/tmp的大小

galaxyskyknight, 2021-05-30:
R2S 因为image默认/tmp为大小物理内存的一半，只有500M左右，问题有两个，1.我的squashfs版本卡刷进去R2S死都起不来，只能用ext4的（R4s也是如此，不知为何）。2.用ext4版本，在make menuconfigure里面只能指定kernel partition和root filesystem partition,找不到哪里指定tmpfs大小。当root fs指定太大时（比如960M) 编译出来的.img就有1.1G，就会超出/tmp的大小，sysupgrade会把.img先拷贝到/tmp再刷，这个时候拷贝就会失败。所以我想吧/tmp改大把root fs改小，保证image大小不超过/tmp的size，但menuconfigure里没有/tmp的partition scheme改动地方貌似，求大佬指教。

galaxyskyknight, 2021-05-30:
> R2S 因为image默认/tmp为大小物理内存的一半(应该是默认指定50%映射了一半的内存作为内存文件系统），只有500M左右，问题有两个，1.我的squashfs版本卡刷进去R2S死都起不来，只能用ext4的（R4s也是如此，不知为何）。2.用ext4版本，在make menuconfigure里面只能指定kernel partition和root filesystem partition,找不到哪里指定tmpfs大小。当root fs指定太大时（比如960M) 编译出来的.img就有1.1G，就会超出/tmp的大小，sysupgrade会把.img先拷贝到/tmp再刷，这个时候拷贝就会失败。所以我想吧/tmp改大把root fs改小，保证image大小不超过/tmp的size，但menuconfigure里没有/tmp的partition scheme改动地方貌似，求大佬指教。

ly20202020, 2021-05-30:
呵呵，那么复杂吗？还是X86的舒服啊，虚拟机安装固件的时候分配1G运行内存，TMP空间就差不多500MB大小，分配2G内存就990几MB大小

closed 2021-06-17 by coolsnowwolf