coolsnowwolf/lede #6913: 新版本之后ipsec可以连接但是没有网络

blackshot-zard, 2021-05-31:
有搜索到类似的issues，但是情况不一样
最新源码没有做任何更改，只加入了hello world
之前的版本是可以用的，应该是在5.4.101之前的版本是一切正常的
团购的J4125 2.5G4口软路由 

可以一直保持连接，但是没有网络，本地网关也无法访问
详细日志如下

Tue Jun  1 07:23:59 2021 daemon.info : 05[NET] received packet: from 112.96.*.*[24929] to 119.131.*.*[500] (848 bytes)
Tue Jun  1 07:23:59 2021 daemon.info : 05[ENC] parsed ID_PROT request 0 [ SA V V V V V V V V V V V V V V ]
Tue Jun  1 07:23:59 2021 daemon.info : 05[IKE] received NAT-T (RFC 3947) vendor ID
Tue Jun  1 07:23:59 2021 daemon.info : 05[IKE] received draft-ietf-ipsec-nat-t-ike vendor ID
Tue Jun  1 07:23:59 2021 daemon.info : 05[IKE] received draft-ietf-ipsec-nat-t-ike-08 vendor ID
Tue Jun  1 07:23:59 2021 daemon.info : 05[IKE] received draft-ietf-ipsec-nat-t-ike-07 vendor ID
Tue Jun  1 07:23:59 2021 daemon.info : 05[IKE] received draft-ietf-ipsec-nat-t-ike-06 vendor ID
Tue Jun  1 07:23:59 2021 daemon.info : 05[IKE] received draft-ietf-ipsec-nat-t-ike-05 vendor ID
Tue Jun  1 07:23:59 2021 daemon.info : 05[IKE] received draft-ietf-ipsec-nat-t-ike-04 vendor ID
Tue Jun  1 07:23:59 2021 daemon.info : 05[IKE] received draft-ietf-ipsec-nat-t-ike-03 vendor ID
Tue Jun  1 07:23:59 2021 daemon.info : 05[IKE] received draft-ietf-ipsec-nat-t-ike-02 vendor ID
Tue Jun  1 07:23:59 2021 daemon.info : 05[IKE] received draft-ietf-ipsec-nat-t-ike-02\n vendor ID
Tue Jun  1 07:23:59 2021 daemon.info : 05[IKE] received XAuth vendor ID
Tue Jun  1 07:23:59 2021 daemon.info : 05[IKE] received Cisco Unity vendor ID
Tue Jun  1 07:23:59 2021 daemon.info : 05[IKE] received FRAGMENTATION vendor ID
Tue Jun  1 07:23:59 2021 daemon.info : 05[IKE] received DPD vendor ID
Tue Jun  1 07:23:59 2021 daemon.info : 05[IKE] 112.96.*.* is initiating a Main Mode IKE_SA
Tue Jun  1 07:23:59 2021 authpriv.info : 05[IKE] 112.96.*.* is initiating a Main Mode IKE_SA
Tue Jun  1 07:23:59 2021 daemon.info : 05[CFG] selected proposal: IKE:AES_CBC_128/HMAC_SHA1_96/PRF_HMAC_SHA1/MODP_1024
Tue Jun  1 07:23:59 2021 daemon.info : 05[ENC] generating ID_PROT response 0 [ SA V V V V ]
Tue Jun  1 07:23:59 2021 daemon.info : 05[NET] sending packet: from 119.131.*.*[500] to 112.96.*.*[24929] (160 bytes)
Tue Jun  1 07:23:59 2021 daemon.info : 08[NET] received packet: from 112.96.*.*[24929] to 119.131.*.*[500] (228 bytes)
Tue Jun  1 07:23:59 2021 daemon.info : 08[ENC] parsed ID_PROT request 0 [ KE No NAT-D NAT-D ]
Tue Jun  1 07:23:59 2021 daemon.info : 08[IKE] remote host is behind NAT
Tue Jun  1 07:23:59 2021 daemon.info : 08[ENC] generating ID_PROT response 0 [ KE No NAT-D NAT-D ]
Tue Jun  1 07:23:59 2021 daemon.info : 08[NET] sending packet: from 119.131.*.*[500] to 112.96.*.*[24929] (244 bytes)
Tue Jun  1 07:23:59 2021 daemon.info : 07[NET] received packet: from 112.96.*.*[34868] to 119.131.*.*[4500] (108 bytes)
Tue Jun  1 07:23:59 2021 daemon.info : 07[ENC] parsed ID_PROT request 0 [ ID HASH N(INITIAL_CONTACT) ]
Tue Jun  1 07:23:59 2021 daemon.info : 07[CFG] looking for XAuthInitPSK peer configs matching 119.131.*.*...112.96.*.*[10.32.12.149]
Tue Jun  1 07:23:59 2021 daemon.info : 07[CFG] selected peer config "xauth_psk"
Tue Jun  1 07:23:59 2021 daemon.info : 07[ENC] generating ID_PROT response 0 [ ID HASH ]
Tue Jun  1 07:23:59 2021 daemon.info : 07[NET] sending packet: from 119.131.*.*[4500] to 112.96.*.*[34868] (76 bytes)
Tue Jun  1 07:23:59 2021 daemon.info : 07[ENC] generating TRANSACTION request 3631897718 [ HASH CPRQ(X_USER X_PWD) ]
Tue Jun  1 07:23:59 2021 daemon.info : 07[NET] sending packet: from 119.131.*.*[4500] to 112.96.*.*[34868] (76 bytes)
Tue Jun  1 07:23:59 2021 daemon.info : 11[NET] received packet: from 112.96.*.*[34868] to 119.131.*.*[4500] (92 bytes)
Tue Jun  1 07:23:59 2021 daemon.info : 11[ENC] parsed TRANSACTION response 3631897718 [ HASH CPRP(X_USER X_PWD) ]
Tue Jun  1 07:23:59 2021 daemon.info : 11[IKE] XAuth authentication of 'blackshot' successful
Tue Jun  1 07:23:59 2021 daemon.info : 11[ENC] generating TRANSACTION request 3829291189 [ HASH CPS(X_STATUS) ]
Tue Jun  1 07:23:59 2021 daemon.info : 11[NET] sending packet: from 119.131.*.*[4500] to 112.96.*.*[34868] (76 bytes)
Tue Jun  1 07:23:59 2021 daemon.info : 12[NET] received packet: from 112.96.*.*[34868] to 119.131.*.*[4500] (76 bytes)
Tue Jun  1 07:23:59 2021 daemon.info : 12[ENC] parsed TRANSACTION response 3829291189 [ HASH CPA(X_STATUS) ]
Tue Jun  1 07:23:59 2021 daemon.info : 12[IKE] IKE_SA xauth_psk[1] established between 119.131.*.*[119.131.*.*]...112.96.*.*[10.32.12.149]
Tue Jun  1 07:23:59 2021 authpriv.info : 12[IKE] IKE_SA xauth_psk[1] established between 119.131.*.*[119.131.*.*]...112.96.*.*[10.32.12.149]
Tue Jun  1 07:23:59 2021 daemon.info : 12[IKE] scheduling reauthentication in 9836s
Tue Jun  1 07:23:59 2021 daemon.info : 12[IKE] maximum IKE_SA lifetime 10376s
Tue Jun  1 07:23:59 2021 daemon.info : 10[NET] received packet: from 112.96.*.*[34868] to 119.131.*.*[4500] (172 bytes)
Tue Jun  1 07:23:59 2021 daemon.info : 10[ENC] unknown attribute type (28683)
Tue Jun  1 07:23:59 2021 daemon.info : 10[ENC] parsed TRANSACTION request 3286780138 [ HASH CPRQ(ADDR MASK DNS NBNS EXP VER U_BANNER U_DEFDOM U_SPLITDNS U_SPLITINC U_LOCALLAN U_PFS U_SAVEPWD U_FWTYPE U_BKPSRV (28683)) ]
Tue Jun  1 07:23:59 2021 daemon.info : 10[IKE] peer requested virtual IP %any
Tue Jun  1 07:23:59 2021 daemon.info : 10[CFG] assigning new lease to 'blackshot'
Tue Jun  1 07:23:59 2021 daemon.info : 10[IKE] assigning virtual IP 192.168.0.101 to peer 'blackshot'
Tue Jun  1 07:23:59 2021 daemon.info : 10[ENC] generating TRANSACTION response 3286780138 [ HASH CPRP(ADDR DNS) ]
Tue Jun  1 07:23:59 2021 daemon.info : 10[NET] sending packet: from 119.131.*.*[4500] to 112.96.*.*[34868] (92 bytes)
Tue Jun  1 07:23:59 2021 daemon.info : 04[NET] received packet: from 112.96.*.*[34868] to 119.131.*.*[4500] (380 bytes)
Tue Jun  1 07:23:59 2021 daemon.info : 04[ENC] parsed QUICK_MODE request 2603352716 [ HASH SA No ID ID ]
Tue Jun  1 07:23:59 2021 daemon.info : 04[CFG] selected proposal: ESP:AES_CBC_128/HMAC_SHA1_96/NO_EXT_SEQ
Tue Jun  1 07:23:59 2021 daemon.info : 04[ENC] generating QUICK_MODE response 2603352716 [ HASH SA No ID ID ]
Tue Jun  1 07:23:59 2021 daemon.info : 04[NET] sending packet: from 119.131.*.*[4500] to 112.96.*.*[34868] (172 bytes)
Tue Jun  1 07:23:59 2021 daemon.info : 13[NET] received packet: from 112.96.*.*[34868] to 119.131.*.*[4500] (60 bytes)
Tue Jun  1 07:23:59 2021 daemon.info : 13[ENC] parsed QUICK_MODE request 2603352716 [ HASH ]
Tue Jun  1 07:24:00 2021 daemon.info : 13[KNL] received netlink error: No such file or directory (2)
Tue Jun  1 07:24:00 2021 daemon.info : 13[KNL] unable to add SAD entry with SPI c1e1a294 (FAILED)
Tue Jun  1 07:24:00 2021 daemon.info : 13[KNL] received netlink error: No such file or directory (2)
Tue Jun  1 07:24:00 2021 daemon.info : 13[KNL] unable to add SAD entry with SPI 0871b986 (FAILED)
Tue Jun  1 07:24:00 2021 daemon.info : 13[IKE] unable to install inbound and outbound IPsec SA (SAD) in kernel
Tue Jun  1 07:24:00 2021 daemon.info : 13[KNL] deleting policy 192.168.0.101/32 === 0.0.0.0/0 in failed, not found
Tue Jun  1 07:24:00 2021 daemon.info : 13[KNL] deleting policy 192.168.0.101/32 === 0.0.0.0/0 fwd failed, not found
Tue Jun  1 07:24:00 2021 daemon.info : 13[IKE] sending DELETE for ESP CHILD_SA with SPI 0871b986
Tue Jun  1 07:24:00 2021 daemon.info : 13[ENC] generating INFORMATIONAL_V1 request 213133679 [ HASH D ]
Tue Jun  1 07:24:00 2021 daemon.info : 13[NET] sending packet: from 119.131.*.*[4500] to 112.96.*.*[34868] (76 bytes)
Tue Jun  1 07:24:10 2021 daemon.err /usr/bin/ss-local[4735]: remote_recv_cb_recv: Connection reset by peer

svover, 2021-06-01:
一样 openvpn 也是

ragouhuo, 2021-06-08:
同楼主，log也一样

eearphon, 2021-06-08:
> 有搜索到类似的issues，但是情况不一样
> 最新源码没有做任何更改，只加入了hello world
> 之前的版本是可以用的，应该是在5.4.101之前的版本是一切正常的
> 团购的J4125 2.5G4口软路由
> 
> 可以一直保持连接，但是没有网络，本地网关也无法访问
> 详细日志如下
> 
> Tue Jun 1 07:23:59 2021 daemon.info : 05[NET] received packet: from 112.96._._[24929] to 119.131._._[500] (848 bytes)

请问能确认下哪个历史版本可用吗？刚测试了5.4.100，好像也不行。

blackshot-zard, 2021-06-09:
> > 有搜索到类似的issues，但是情况不一样
> > 最新源码没有做任何更改，只加入了hello world
> > 之前的版本是可以用的，应该是在5.4.101之前的版本是一切正常的
> > 团购的J4125 2.5G4口软路由
> > 可以一直保持连接，但是没有网络，本地网关也无法访问
> > 详细日志如下
> > Tue Jun 1 07:23:59 2021 daemon.info : 05[NET] received packet: from 112.96._._[24929] to 119.131._._[500] (848 bytes)
> 
> 请问能确认下哪个历史版本可用吗？刚测试了5.4.100，好像也不行。

不太记得清了 我也不会版本回退😂

eearphon, 2021-06-10:
重新默认编译了x86_64版本，最新 d871216，能连上，但啥网也访问不了，加上[#5262](https://github.com/coolsnowwolf/lede/issues/5262)的那些命令也不行

blackshot-zard, 2021-06-10:
> 重新默认编译了x86_64版本，最新 [d871216](https://github.com/coolsnowwolf/lede/commit/d871216a7ce1f1a6b8efae99870361dfd74b7ff8)，能连上，但啥网也访问不了，加上[#5262](https://github.com/coolsnowwolf/lede/issues/5262)的那些命令也不行

是的 最新版本也一样不行，一开始我以为是运营商封udp了，结果老家的老版本可以，好像是86？ 然后我作死给老家的版本也更新了，然后老家的也GG了

closed 2021-06-17 by coolsnowwolf

eearphon, 2021-09-14:
用了隔壁L大的版本一段时间，昨晚编译了 fe5786bfa4f0f1c383fc448778c8c2c327c7f40a 版，又一切正常了，x86-64