coolsnowwolf/lede #6914: 为什么Dnsmasq DHCP的剩余租期显示不出来？

gztingting, 2021-06-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

以前编译的版本，DHCP显示的剩余租期都是正确的倒计时，最近版本一律显示为“过期时间”，不知道在哪里调整正确。

[image]

### (2) 路由器型号和固件版本

A：x86-64软路由Lede OpenWrt R21.2.1

### (3) 详细日志

Sun May 30 01:00:00 2021 user.notice FRPC: Shutting down frp service
Sun May 30 01:00:00 2021 user.notice FRPC: Starting frp service
Sun May 30 01:40:00 2021 cron.err crond[12586]: USER root pid 26194 cmd /etc/init.d/frp restart
Sun May 30 01:40:00 2021 user.notice FRPC: Shutting down frp service
Sun May 30 01:40:00 2021 user.notice FRPC: Starting frp service
Sun May 30 02:00:00 2021 cron.err crond[12586]: USER root pid 27121 cmd /usr/share/shadowsocksr/ssrplusupdate.sh >/var/log/ssrplus.log
Sun May 30 02:00:00 2021 cron.err crond[12586]: USER root pid 27122 cmd /etc/init.d/frp restart
Sun May 30 02:00:00 2021 user.notice FRPC: Shutting down frp service
Sun May 30 02:00:00 2021 user.notice FRPC: Starting frp service
Sun May 30 02:40:00 2021 cron.err crond[12586]: USER root pid 29112 cmd /etc/init.d/frp restart
Sun May 30 02:40:00 2021 user.notice FRPC: Shutting down frp service
Sun May 30 02:40:00 2021 user.notice FRPC: Starting frp service
Sun May 30 03:00:00 2021 cron.err crond[12586]: USER root pid 30044 cmd /etc/init.d/frp restart
Sun May 30 03:00:00 2021 user.notice FRPC: Shutting down frp service
Sun May 30 03:00:00 2021 user.notice FRPC: Starting frp service
Sun May 30 03:40:00 2021 cron.err crond[12586]: USER root pid 31871 cmd /etc/init.d/frp restart
Sun May 30 03:40:00 2021 user.notice FRPC: Shutting down frp service
Sun May 30 03:40:00 2021 user.notice FRPC: Starting frp service
Sun May 30 04:00:00 2021 cron.err crond[12586]: USER root pid 347 cmd /etc/init.d/frp restart
Sun May 30 04:00:00 2021 user.notice FRPC: Shutting down frp service
Sun May 30 04:00:00 2021 user.notice FRPC: Starting frp service
Sun May 30 04:40:00 2021 cron.err crond[12586]: USER root pid 2257 cmd /etc/init.d/frp restart
Sun May 30 04:40:00 2021 user.notice FRPC: Shutting down frp service
Sun May 30 04:40:00 2021 user.notice FRPC: Starting frp service
Sun May 30 05:00:00 2021 cron.err crond[12586]: USER root pid 3213 cmd /etc/init.d/frp restart
Sun May 30 05:00:00 2021 user.notice FRPC: Shutting down frp service
Sun May 30 05:00:00 2021 user.notice FRPC: Starting frp service
Sun May 30 05:40:00 2021 cron.err crond[12586]: USER root pid 5036 cmd /etc/init.d/frp restart
Sun May 30 05:40:00 2021 user.notice FRPC: Shutting down frp service
Sun May 30 05:40:00 2021 user.notice FRPC: Starting frp service
Sun May 30 06:00:00 2021 cron.err crond[12586]: USER root pid 5994 cmd /etc/init.d/frp restart
Sun May 30 06:00:00 2021 user.notice FRPC: Shutting down frp service
Sun May 30 06:00:00 2021 user.notice FRPC: Starting frp service
Sun May 30 06:40:00 2021 cron.err crond[12586]: USER root pid 7808 cmd /etc/init.d/frp restart
Sun May 30 06:40:00 2021 user.notice FRPC: Shutting down frp service
Sun May 30 06:40:00 2021 user.notice FRPC: Starting frp service
Sun May 30 07:00:00 2021 cron.err crond[12586]: USER root pid 8746 cmd /etc/init.d/frp restart
Sun May 30 07:00:00 2021 user.notice FRPC: Shutting down frp service
Sun May 30 07:00:00 2021 user.notice FRPC: Starting frp service
Sun May 30 07:40:00 2021 cron.err crond[12586]: USER root pid 10574 cmd /etc/init.d/frp restart
Sun May 30 07:40:00 2021 user.notice FRPC: Shutting down frp service
Sun May 30 07:40:00 2021 user.notice FRPC: Starting frp service
Sun May 30 07:48:49 2021 daemon.err nlbwmon[2855]: Netlink receive failure: Out of memory
Sun May 30 07:48:49 2021 daemon.err nlbwmon[2855]: Unable to dump conntrack: No buffer space available
Sun May 30 08:00:00 2021 cron.err crond[12586]: USER root pid 11886 cmd /etc/init.d/frp restart
Sun May 30 08:00:00 2021 user.notice FRPC: Shutting down frp service
Sun May 30 08:00:00 2021 user.notice FRPC: Starting frp service
Sun May 30 08:17:54 2021 daemon.err nlbwmon[2855]: Netlink receive failure: Object busy
Sun May 30 08:17:54 2021 daemon.err nlbwmon[2855]: Unable to dump conntrack: I/O error
[... 1747 lines cut ...]
[6016370.938270] igb 0000:01:00.0 eth0: igb: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[6016371.127622] pppoe-wan: renamed from ppp0
[6016374.513600] 8021q: adding VLAN 0 to HW filter on device eth0
[6016374.609623] 8021q: adding VLAN 0 to HW filter on device eth0
[6016377.782329] igb 0000:01:00.0 eth0: igb: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[6016377.978628] pppoe-wan: renamed from ppp0
[6019009.414538] igb 0000:02:00.0 eth1: igb: eth1 NIC Link is Down
[6019009.421847] br-lan: port 1(eth1) entered disabled state
[6019011.155265] igb 0000:02:00.0 eth1: igb: eth1 NIC Link is Up 10 Mbps Full Duplex, Flow Control: RX
[6019011.165408] br-lan: port 1(eth1) entered blocking state
[6019011.170931] br-lan: port 1(eth1) entered forwarding state
[6019025.983053] igb 0000:02:00.0 eth1: igb: eth1 NIC Link is Down
[6019025.990435] br-lan: port 1(eth1) entered disabled state
[6019028.967575] igb 0000:02:00.0 eth1: igb: eth1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[6019029.087132] br-lan: port 1(eth1) entered blocking state
[6019029.092946] br-lan: port 1(eth1) entered forwarding state
[6019550.667644] igb 0000:02:00.0 eth1: igb: eth1 NIC Link is Down
[6019550.674508] br-lan: port 1(eth1) entered disabled state
[6019553.431858] igb 0000:02:00.0 eth1: igb: eth1 NIC Link is Up 10 Mbps Full Duplex, Flow Control: RX
[6019553.441989] br-lan: port 1(eth1) entered blocking state
[6019553.447483] br-lan: port 1(eth1) entered forwarding state
[6189488.029770] 8021q: adding VLAN 0 to HW filter on device eth0
[6189488.125562] 8021q: adding VLAN 0 to HW filter on device eth0
[6189491.258311] igb 0000:01:00.0 eth0: igb: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[6189491.448762] pppoe-wan: renamed from ppp0

hank9999, 2021-06-08:
遇到了+1

gztingting, 2021-06-14:
因为选了更多的dnsmasq组件导致的，不确定是哪个插件冲突。建议只选v4 dhcp和ipset即可

closed 2021-06-14 by gztingting