coolsnowwolf/lede #6917: rock pi 4b 编译成功后刷入sd卡无法启动

DH-FU, 2021-06-02:
rock pi 4b 4G版(RK3399) LEDE编译成功后，刷入sd卡无法启动，openwrt编译刷入启动没问题，尝试了好几次，但不知道是什么原因？

DH-FU, 2021-06-02:
CONFIG_TARGET_rockchip=y
CONFIG_TARGET_rockchip_armv8=y
CONFIG_TARGET_rockchip_armv8_DEVICE_radxa_rock-pi-4=y
# CONFIG_PACKAGE_UnblockNeteaseMusicGo is not set
# CONFIG_PACKAGE_coreutils is not set
# CONFIG_PACKAGE_etherwake is not set
# CONFIG_PACKAGE_ip-full is not set
# CONFIG_PACKAGE_ipset is not set
CONFIG_PACKAGE_iptables-mod-conntrack-extra=y
CONFIG_PACKAGE_iptables-mod-filter=y
CONFIG_PACKAGE_iptables-mod-ipopt=y
CONFIG_PACKAGE_iptables-mod-nfqueue=y
# CONFIG_PACKAGE_iptables-mod-tproxy is not set
CONFIG_PACKAGE_iptables-mod-u32=y
# CONFIG_PACKAGE_kmod-ip6tables is not set
CONFIG_PACKAGE_kmod-ipt-conntrack-extra=y
CONFIG_PACKAGE_kmod-ipt-filter=y
CONFIG_PACKAGE_kmod-ipt-ipopt=y
CONFIG_PACKAGE_kmod-ipt-nfqueue=y
# CONFIG_PACKAGE_kmod-ipt-tproxy is not set
CONFIG_PACKAGE_kmod-ipt-u32=y
# CONFIG_PACKAGE_kmod-nf-ipt6 is not set
# CONFIG_PACKAGE_kmod-nf-reject6 is not set
CONFIG_PACKAGE_kmod-nfnetlink-queue=y
# CONFIG_PACKAGE_kmod-tun is not set
# CONFIG_PACKAGE_libelf is not set
# CONFIG_PACKAGE_libminiupnpc is not set
# CONFIG_PACKAGE_libnatpmp is not set
CONFIG_PACKAGE_libnetfilter-queue=y
CONFIG_PACKAGE_libnfnetlink=y
# CONFIG_PACKAGE_luci-app-accesscontrol is not set
# CONFIG_PACKAGE_luci-app-arpbind is not set
# CONFIG_PACKAGE_luci-app-autoreboot is not set
# CONFIG_PACKAGE_luci-app-ssr-plus is not set
# CONFIG_PACKAGE_luci-app-unblockmusic is not set
# CONFIG_PACKAGE_luci-app-vlmcsd is not set
# CONFIG_PACKAGE_luci-app-vsftpd is not set
# CONFIG_PACKAGE_luci-app-webadmin is not set
# CONFIG_PACKAGE_luci-app-wol is not set
# CONFIG_PACKAGE_luci-app-zerotier is not set
# CONFIG_PACKAGE_microsocks is not set
# CONFIG_PACKAGE_resolveip is not set
# CONFIG_PACKAGE_shadowsocks-rust-sslocal is not set
# CONFIG_PACKAGE_shadowsocks-rust-ssserver is not set
# CONFIG_PACKAGE_shadowsocksr-libev-ssr-check is not set
# CONFIG_PACKAGE_tcping is not set
CONFIG_PACKAGE_ua2f=y
# CONFIG_PACKAGE_vlmcsd is not set
# CONFIG_PACKAGE_vsftpd-alt is not set
# CONFIG_PACKAGE_wol is not set
# CONFIG_PACKAGE_zerotier is not set
CONFIG_TARGET_ROOTFS_PARTSIZE=2048

coolsnowwolf, 2021-06-02:
TF 卡的兼容性问题

DH-FU, 2021-06-02:
> TF 卡的兼容性问题

什么sd卡好点推荐一下吧

closed 2021-06-17 by coolsnowwolf