coolsnowwolf/lede #6922: X86 多核心负载非常不均

ostrichii, 2021-06-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
大多数情况下，PT下载或者其它大量小包场景下时，查看网卡中断主要集中在一个cpu核心上。
但在使用过程中偶尔也出现能够均匀地把负载分配到每个核心上，这时候关闭Fullcone nat 能够跑到30W ppsUDP小包的样子。
不知道是不是还有哪里没配置好。求大佬们支招

### (1) 具体问题

A：
大多数情况下，PT下载或者其它大量小包场景下时，查看网卡中断主要集中在一个cpu核心上。
但在使用过程中偶尔也出现能够均匀地把负载分配到每个核心上，这时候关闭Fullcone nat 能够跑到30W ppsUDP小包的样子。
不知道是不是还有哪里没配置好。求大佬们支招
以下是4wpps小包cpu的表现
[image][image]
[image]

中断分布
[image]

### (2) 路由器型号和固件版本

Intel(R) Core(TM) i5-8250U CPU @ 1.60GHz : 4 Core 8 Thread (CpuMark : 70768.278119 Scores)
固件是最新版
--

<br class="Apple-interchange-newline">
A：

### (3) 详细日志

A：

galaxyskyknight, 2021-06-03:
你有没有多拨？有多拨把Flow offloading关掉, 然后试试自己配置一下网卡中断和CPU affinity

ostrichii, 2021-06-03:
> 你有没有多拨？有多拨把Flow offloading关掉, 然后试试自己配置一下网卡中断和CPU affinity

无多拨，就一拨千兆。网卡和CPU affinity试过，但是网卡不是多队列，一个LAN一个WAF，分RX/TX，最多用到4个核心。而且看还是不均匀。

ostrichii, 2021-06-03:
记得大雕曾经有一个打散软中断的conmmit @coolsnowwolf ，求教育

ostrichii, 2021-06-03:
通过开启和关闭sqm QoS功能，成功把中断分配到每个core上了，应该是哪里有冲突

neavo, 2021-06-04:
这个信息面板是哪个组件提供的呀

ostrichii, 2021-06-07:
> 这个信息面板是哪个组件提供的呀

netdata

closed 2021-06-07 by ostrichii