coolsnowwolf/lede #6940: Udpxy malfunction

XiaoliChan, 2021-06-06:
Last week, my udpxy can work fine. But now, it did not work.

Here is the logs:

`2021-06-06 16:57:33.817089 CST	S(5316)	udpxy 1.0-24.1 (prod) standard [Linux 5.4.123 x86_64]: /usr/bin/udpxy -T -S -a 192.168.1.1 -p 8888 -m eth1.48 -c 5000 -l /tmp/udpxy.log -B 2097152 -R -1 -M 55 
2021-06-06 16:57:33.817133 CST	S(5316)	Server is starting up, max clients = [5000]
2021-06-06 16:57:33.817750 CST	S(5316)	Setting up listener for [192.168.1.1:8888]
2021-06-06 16:57:33.817786 CST	S(5316)	Setting low watermark for server socket [8] to [10]
2021-06-06 16:57:33.817823 CST	S(5316)	Created server socket=[8], backlog=[16]
2021-06-06 16:57:33.817870 CST	S(5316)	Entering server loop [pselect(2)]
2021-06-06 16:57:33.817919 CST	S(5316)	Waiting for input from [2] fd's, NO timeout
2021-06-06 17:05:39.036755 CST	S(5316)	No children exited since last check
2021-06-06 17:05:39.036801 CST	S(5316)	Got 1 requests
2021-06-06 17:05:39.036826 CST	S(5316)	Accepting new connection
2021-06-06 17:05:39.036858 CST	S(5316)	Accepted socket=[9] from 192.168.1.8:34422 n=1/nmax=16
2021-06-06 17:05:39.036881 CST	S(5316)	Accepting new connection
2021-06-06 17:05:39.036908 CST	S(5316)	Nothing more to accept
2021-06-06 17:05:39.036929 CST	S(5316)	accept_requests: Sockets accepted: [1]
2021-06-06 17:05:39.036951 CST	S(5316)	Waiting for input from [3] fd's, with timeout
2021-06-06 17:05:39.037231 CST	S(5316)	No children exited since last check
2021-06-06 17:05:39.037284 CST	S(5316)	Got 1 requests
pre-process sockets [1]: 9
2021-06-06 17:05:39.037309 CST	S(5316)	acting on accepted socket [9] (1/1)
2021-06-06 17:05:39.037332 CST	S(5316)	Reading command from socket [9]
2021-06-06 17:05:39.037360 CST	S(5316)	HTTP buffer [306 bytes] received
HEAD /rtp/239.20.0.163:2124 HTTP/1.1

Host: 192.168.1.1:8888

User-Agent: Kodi/19.1 (Linux; Android 8.0; VIDAA_TV Build/OPR5.170623.014) Android/8.0.0 Sys_CPU/armv8l App_Bitness/32 Version/19.1-(19.1.0)-Git:20210508-85e05228b4

Accept: */*

Accept-Encoding: deflate, gzip

Accept-Charset: UTF-8,*;q=0.8

2021-06-06 17:05:39.037384 CST	S(5316)	error [1] processing client socket [9]
2021-06-06 17:05:39.037405 CST	S(5316)	process_requests: closing accepted socket [9]
2021-06-06 17:05:39.037440 CST	S(5316)	Processed [1/1] accepted sockets
newly-accepted sockets [1]: -1
2021-06-06 17:05:39.037466 CST	S(5316)	All accepted sockets processed
2021-06-06 17:05:39.037486 CST	S(5316)	Waiting for input from [2] fd's, NO timeout
2021-06-06 17:05:39.048938 CST	S(5316)	No children exited since last check
2021-06-06 17:05:39.048974 CST	S(5316)	Got 1 requests
2021-06-06 17:05:39.048998 CST	S(5316)	Accepting new connection
2021-06-06 17:05:39.049026 CST	S(5316)	Accepted socket=[9] from 192.168.1.8:34424 n=1/nmax=16
2021-06-06 17:05:39.049050 CST	S(5316)	Accepting new connection
2021-06-06 17:05:39.049074 CST	S(5316)	Nothing more to accept
2021-06-06 17:05:39.049096 CST	S(5316)	accept_requests: Sockets accepted: [1]
2021-06-06 17:05:39.049118 CST	S(5316)	Waiting for input from [3] fd's, with timeout
2021-06-06 17:05:39.049247 CST	S(5316)	No children exited since last check
2021-06-06 17:05:39.049287 CST	S(5316)	Got 1 requests
pre-process sockets [1]: 9
2021-06-06 17:05:39.049313 CST	S(5316)	acting on accepted socket [9] (1/1)
2021-06-06 17:05:39.049336 CST	S(5316)	Reading command from socket [9]
2021-06-06 17:05:39.049361 CST	S(5316)	HTTP buffer [306 bytes] received
HEAD /rtp/239.20.0.163:2124 HTTP/1.1

Host: 192.168.1.1:8888

User-Agent: Kodi/19.1 (Linux; Android 8.0; VIDAA_TV Build/OPR5.170623.014) Android/8.0.0 Sys_CPU/armv8l App_Bitness/32 Version/19.1-(19.1.0)-Git:20210508-85e05228b4

Accept: */*

Accept-Encoding: deflate, gzip

Accept-Charset: UTF-8,*;q=0.8

2021-06-06 17:05:39.049384 CST	S(5316)	error [1] processing client socket [9]
[... 127 lines cut ...]
2021-06-06 17:05:44.360699 CST	c(23392)	socket 6: SEND timeout set to 5 sec, 0 usec
2021-06-06 17:05:44.360722 CST	c(23392)	current send buffer size is [130560] bytes for socket [9]
2021-06-06 17:05:44.360745 CST	c(23392)	current receive buffer size is [425984] bytes for socket [6]
2021-06-06 17:05:44.360768 CST	c(23392)	send buffer size set to [425984] bytes for socket [9]
2021-06-06 17:05:44.360808 CST	c(23392)	Sent HTTP response code=[200], reason=[OK] to socket=[9]
HTTP/1.1 200 OK

Server: udpxy 1.0-24.1 (prod) standard [Linux 5.4.123 x86_64]

Content-Type:application/octet-stream

2021-06-06 17:05:44.360842 CST	c(23392)	Relaying traffic from socket[6] to socket[9], buffer size=[2097152], Rmsgs=[-1], pauses=[0]
2021-06-06 17:05:49.465906 CST	c(23392)	read_buf: socket time-out on read2021-06-06 17:05:49.465981 CST	c(23392)	read_data - EOF
2021-06-06 17:05:49.466023 CST	c(23392)	Exited relay loop: received=[-1], sent=[0], quit=[0]
2021-06-06 17:05:49.466197 CST	c(23392)	multicast-group [DROP]
2021-06-06 17:05:49.466229 CST	c(23392)	Mcast listener socket=[6] closed
2021-06-06 17:05:49.466298 CST	c(23392)	Child process=[23392] exits with rc=[0]
2021-06-06 17:05:49.466439 CST	S(5316)	*** Caught SIGCHLD (17) ***
2021-06-06 17:05:49.466484 CST	S(5316)	Waiting on exited children
2021-06-06 17:05:49.466514 CST	S(5316)	Client [23392] has exited.
2021-06-06 17:05:49.466537 CST	S(5316)	Deleted client: pid=[23392]
2021-06-06 17:05:49.466559 CST	S(5316)	Cleaned up 1 children, 0 still running
2021-06-06 17:05:49.466581 CST	S(5316)	INTERRUPTED, yet will continue.
2021-06-06 17:05:49.466601 CST	S(5316)	Waiting for input from [2] fd's, NO timeout
`

XiaoliChan, 2021-06-06:
Here is my udpxy config file:
`config udpxy
        option respawn '1'
        option verbose '0'
        option status '1'
        option port '8888'
        option buffer_size '2097152'
        option bind '192.168.1.1'
        option source 'eth1.48'
        option disabled '0'
        option log_file '/tmp/udpxy.log'
        option buffer_messages '-1'
        option mcsub_renew '55'
        option max_clients '5000'`

XiaoliChan, 2021-06-06:
Solved: Guangzhou CMCC change ZTE IPTV source to Huawei IPTV source.

closed 2021-06-06 by XiaoliChan

fqx, 2022-11-25:
> Solved: Guangzhou CMCC changes ZTE IPTV source to Huawei IPTV source.

How do you solve that?
Is the rtp url changed?

I can only watch 4 mins or so, and I have add "-M 55" to udpxy.