coolsnowwolf/lede #6941: 升级固件MWAN3至最新版本（2.10.9-1）

tangyisheng2, 2021-06-06:
目前lean大的版本仍然是使用mwan3 2.8的版本，建议更新到最新的mwan3 2.10（截止发布issue最新版本是2.10.9-1）。新版本解决了UDP流量分流失败的问题。

我自己使用opkg upgrade升级mwan3到新版本，发现在目前系统中会偶尔出错，出错提示见下面issue。

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：偶尔提示interface wan1 is error (15)或interface wan1 is error (16)。出现错误提示时流量不走wan1，但其实wan1的网络是正常可ping通的状态。问题在在ifup wan1后问题解决，此外尝试通过mwan3 restart重启mwan3并不能解决问题。

### (2) 路由器型号和固件版本

A：x86软路由，使用eSir大大编译的OpenWrt R21.4.18 GDQ v5.1[2021]

### (3) 详细日志

A：因问题随机出现目前并没有抓到出错的log，后续有log我将进行补充。

感谢Lean大

Beginner-Go, 2021-06-06:
luci需要升级吗？

Beginner-Go, 2021-06-07:
https://github.com/coolsnowwolf/luci/pull/63
https://github.com/coolsnowwolf/packages/pull/158

closed 2021-06-17 by coolsnowwolf