coolsnowwolf/lede #6942: What is this repository about?

charlemagnelasse, 2021-06-06:
It seems like people think that this repository is the upstream LEDE/OpenWrt version (most likely because it isn't marked as fork of https://github.com/openwrt/openwrt ). As result, they are reporting bogus build errors against the official OpenWrt feeds.

Please clarify in your project what it is about in your README and that it is not OpenWrt nor LEDE.

Btw. based on your commit history, it is clear that your development strategy is a complete mess. Please behave like a good open source project and contribute your changes back to the official repositories. This makes sure that everyone can easily use your changes and at the same time makes it trivial for you to get changes from upstream into your repository - just merge the relevant branches into your repository instead of cherry picking random commits.

charlemagnelasse, 2021-06-06:
Here is an example for such an bogus bug: https://github.com/openwrt-routing/packages/issues/665

JohnSpirit, 2021-06-07:
@charlemagnelasse  This "lede" branch is a version of openwrt maintained by Chinese developers. The leader is an excellent developer called "Lean". Due to various reasons, China's network environment is subject to various restrictions, and many users need the Openwrt system to improve their network experience. The system maintained by Lean is easy to use for Chinese users, such as adding some network proxy plugins. This branch is for users in mainland China, so there is no need to fully synchronize with the original version.

adore595, 2021-06-07:
You don't understand china.

bili-guigui, 2021-06-07:
China GFW...

fhbyljj, 2021-06-07:
You don't understand china.

charlemagnelasse, 2021-06-07:
> You don't understand china.

Maybe. But I understand that the way this project advertises itself resulted in confusions with the actual upstream. And thus the own users were not able to understand that this repository is not upstream LEDE/OpenWrt and nobody should expect that packages from the master branch of the official OpenWrt feeds build against this fork.

And its development methodology is a big waste of time and energy. It would be a lot better to use a release branch of OpenWrt, move the new packages to a feed, bring shared code back to upstream and merge the upstream code changes instead of cherry picking a couple of patches.

realLoser1984, 2021-06-07:
> But I understand that the way this project advertises itself resulted in confusions with the actual upstream.

So, could you point out, how this project "advertises" itself resulted in confusions with the actual upstream?
In README, it states that this project is "**Lean's** git source of OpenWrt and packages", instead of "**official**" or "**upstream**".

> It would be a lot better to use a release branch of OpenWrt, move the new packages to a feed, bring shared code back to upstream and merge the upstream code changes instead of cherry picking a couple of patches.

1. This project is based on OpenWrt 18.06, with latest kernel support.
2. Not all the patches commited by upstream are needed by this project, it just needs to cherry-pick what it actually needed.
3. There're many stuffs which cannot be accpected by upstream, like the patches which could never go to upstream kernel, source code from vendor, some hacks for performance, and overclocking etc.
4. This project never stops anyone submitting any code to anywhere, as long as they follows the license.

> And its development methodology is a big waste of time and energy.

How did you come to this conclusion?
What did you do for this project?

Now I see a guy who made no contribution to this project, opened an issue, superciliously said "your project is a waste of time".

BTW, these questions may be more suitable for Router vendors.

Regards.

ZqinKing, 2021-06-08:
Eat radish and worry about it

closed 2021-06-10 by coolsnowwolf