coolsnowwolf/lede #6950: 21.02编译好后点击网络-接口报错,望大神帮忙看看是啥问题

trombonist852, 2021-06-09:
x86软路由,21.02,点继续之后还是会跳,这个应该咋解决?望大神们不吝赐教,感谢!
[image]

EnnawYang, 2021-06-09:
为什么不问问官方呢？

trombonist852, 2021-06-09:
> 为什么不问问官方呢？
在这里问比较亲切 哈哈

EnnawYang, 2021-06-09:
> > 为什么不问问官方呢？
> > 在这里问比较亲切 哈哈

可这里是18.06啊

hcym, 2021-06-09:
多了个设备设置，多点几次，或者参考正常工作的配置自己写了，应该没问题

trombonist852, 2021-06-09:
> > > 为什么不问问官方呢？
> > > 在这里问比较亲切 哈哈
> 
> 可这里是18.06啊

期待有好心的大佬能帮忙看看呢 哈哈  官方issue我去看过了 也有好多反应这个问题的,好像是都在等修复

hcym, 2021-06-09:
和1907比较多了设备管理，master分支最多跳三四次吧，21.02最多二次
我是记下来直接写成单臂路由，即插即用
x86和armv8都OK，2102肯定绝对完全可以，master分支可能是拼人品

trombonist852, 2021-06-09:
> 我是用的lienol的21.02的分支，目前测试n1的有这个问题，合成的内核是5.4.93，不知道跟内核有没有关系，我看op官方的issue里面有说把内核升级到5.10这个问题就消失了

Neverends4, 2021-06-10:
21.02对network语法有修改，然后正式使用DSA做某些设备的网络架构，烦死了，VLAN之前用swconfig好好的现在整不出来了
另外21.02rc2出来了，Luci对从swconfig到DSA脚本的升级有改动，你可以试试
另另外…………确实不应该在这里报官方op的issue吧……ʕ⊙ᴥ⊙ʔ

trombonist852, 2021-06-10:
> —官方已经修复了，我在官方也已经发issue问过了，就在我发了之后的几个小时，就修复了…🤣🤣此贴关闭

closed 2021-06-10 by trombonist852