coolsnowwolf/lede #6957: 谁能教我如何替换n2n_v2到n2n_v3

galaxyskyknight, 2021-06-10:
如题，现在代码里是n2n_v2 2.8 的branch, 想换成最新的3.0的分支支持supernode federation,但不知道怎么弄，不知道该改哪里，单独能编一个ipk出来也行啊。建议大佬升级一下到3.0吧？

galaxyskyknight, 2021-06-10:
尝试在package/lean/n2n_v2/Makefile里改git 的commit id到最新branch,但编译失败，报一堆错，不知道何故....

galaxyskyknight, 2021-06-11:
n2n_v3(2.9.1) 在交叉编译平台下(x86/R4s/R2s都如此）configure失败，在目前n2n_v2(2.8)下就没问题，是啥情况，下面是config.log大佬帮我看看如何解决？

Thread model: posix
gcc version 7.5.0 (Ubuntu 7.5.0-6ubuntu2)
configure:2129: $? = 0
configure:2118: gcc -V >&5
gcc: error: unrecognized command line option '-V'
gcc: fatal error: no input files
compilation terminated.
configure:2129: $? = 1
configure:2118: gcc -qversion >&5
gcc: error: unrecognized command line option '-qversion'; did you mean '--version'?
gcc: fatal error: no input files
compilation terminated.
configure:2129: $? = 1
configure:2149: checking whether the C compiler works
configure:2171: gcc -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/luke/lede_arm/build_dir/target-aarch64_generic_musl/n2n-2.9.1_git-701befe6df8eaa43e94384a234fa962fd74ff70e=n2n-2.9.1_git-701befe6df8eaa43e94384a234fa962fd74ff70e -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/luke/lede_arm/staging_dir/target-aarch64_generic_musl/usr/include -I/home/luke/lede_arm/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/usr/include -I/home/luke/lede_arm/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/include/fortify -I/home/luke/lede_arm/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/include  -L/home/luke/lede_arm/staging_dir/target-aarch64_generic_musl/usr/lib -L/home/luke/lede_arm/staging_dir/target-aarch64_generic_musl/lib -L/home/luke/lede_arm/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/usr/lib -L/home/luke/lede_arm/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/lib -znow -zrelro  conftest.c  >&5
**gcc: warning: '-mcpu=' is deprecated; use '-mtune=' or '-march=' instead
gcc: error: unrecognized command line option '-fhonour-copts'; did you mean '-fhonor-std'?
gcc: error: unrecognized command line option '-ffile-prefix-map=/home/luke/lede_arm/build_dir/target-**aarch64_generic_musl/n2n-2.9.1_git-701befe6df8eaa43e94384a234fa962fd74ff70e=n2n-2.9.1_git-701befe6df8eaa43e94384a234fa962fd74ff70e'
configure:2175: $? = 1
configure:2213: result: no
configure: failed program was:
| /* confdefs.h */
| #define PACKAGE_NAME "edge"
| #define PACKAGE_TARNAME "edge"
| #define PACKAGE_VERSION "2.9.0"
| #define PACKAGE_STRING "edge 2.9.0"
| #define PACKAGE_BUGREPORT ""
| #define PACKAGE_URL ""
| /* end confdefs.h.  */
|
| int
| main ()
| {
|
|   ;
|   return 0;
| }
configure:2218: error: in `/home/luke/lede_arm/build_dir/target-aarch64_generic_musl/n2n-2.9.1_git-701befe6df8eaa43e94384a234fa962fd74ff70e':
configure:2220: error: C compiler cannot create executables
See `config.log' for more details

galaxyskyknight, 2021-06-11:
大佬们指导一下呀。。。。。好像这里都没人？还是我发错地方了。。。。

AmadeusGhost, 2021-06-13:
试试这样改可不可以吧，我试了下在 ipq40xx 上编译正常：https://github.com/AmadeusGhost/lede/commit/2319fe81d23daf98b6a6fafd26702f4e48daadfa

galaxyskyknight, 2021-06-13:
> 试试这样改可不可以吧，我试了下在 ipq40xx 上编译正常：[AmadeusGhost@2319fe8](https://github.com/AmadeusGhost/lede/commit/2319fe81d23daf98b6a6fafd26702f4e48daadfa)

兄弟，可以了，感谢！你可帮了我的大忙啊，原理是啥，我看就改了一个默认的configure的路径？

closed 2021-06-17 by coolsnowwolf