coolsnowwolf/lede #6967: 源码里的keepalived使用问题

freesilent, 2021-06-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
 默认编译通过init.d启动的keepalived无法正常使用。
### (1) 具体问题

[A：默认启动时会在建立/tmp/keepalived.conf文件，这个文件为空，没有配置，曾经上网搜过解决方法，也去op论坛提出问题，无法解决](https://forum.openwrt.org/t/how-to-set-keepalived-config/97404)。只能通过添加手动添加启动命令行使用。
后来通过删除原有所有配置文件/etc/config/keepalived、/etc/keepalived/keepalived.conf，然后opkg升级keepalived解决。
源码中的keepalived为2.0.20-4，opkg最新的keepalived为2.2.2

### (2) 路由器型号和固件版本

A：x86、小娱C5。

Beginner-Go, 2021-06-13:
意思是要更新源码内的keepalived？

Beginner-Go, 2021-06-13:
Can you test?

freesilent, 2021-06-14:
ok,i wil test

freesilent, 2021-06-15:
抓取新代码后，编译的keepalived为2.2.2，没有问题了。

closed 2021-06-17 by coolsnowwolf

six7ths, 2022-05-26:
我用最新的版本编译的，还是不行，只能用关闭做成系统服务，采取手动启动，加入启动项，默认配置文件才好用。

six7ths, 2022-05-26:
找到解决办法了，需要两步配置，按这个思路来的
https://yanke.info/?id=161

补充一个 今天发现的，可以以服务启动的方法：

1、修改 vim /etc/config/keepalived 为这个样子。

config globals 'globals'
option alt_config_file "/etc/keepalived/keepalived.conf"

2、再修改默认配置文件是vim /etc/keepalived/keepalived.conf，跟上面一样。

3、service keepalived restart, 就可以vip正常获取了。   查看日志， 用的还是tmp下的配置文件，但此时已经不是空的了，已经把配置复制过去了。