coolsnowwolf/lede #6968: 可以更新一下strongswan版本吗？

zhoulizhengithub, 2021-06-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
现在的版本无法使用IPsec 连接无法装载内核
A：

### (2) 路由器型号和固件版本

A：AC2100

### (3) 详细日志

A：

Beginner-Go, 2021-06-13:
What's this for？

Beginner-Go, 2021-06-13:
Can you test? 
https://github.com/coolsnowwolf/packages/pull/161

zhoulizhengithub, 2021-06-14:
我现在去测试一下

zhoulizhengithub, 2021-06-14:
> 你能测试吗？
> [coolsnowwolf/包#161](https://github.com/coolsnowwolf/packages/pull/161)
我从新下载为什么编出来还是。strongswan5.8.4 ,没一更新，还是我需要什么操作才可以更新到您这个结点，我不太懂请教谢谢！！

zhoulizhengithub, 2021-06-14:
更新Makefile 后编译出错！

/usr/bin/env bash ./run_tic.sh
** Building terminfo database, please wait...
Running tic to install /home/zhoulizhen/lede/staging_dir/hostpkg/share/terminfo ...

	You may see messages regarding extended capabilities, e.g., AX.
	These are extended terminal capabilities which are compiled
	using
		tic -x
	If you have ncurses 4.2 applications, you should read the INSTALL
	document, and install the terminfo without the -x option.

"terminfo.tmp", line 1115, col 36, terminal 'fbterm': limiting value of `pairs' from 0x10000 to 0x7fff
"terminfo.tmp", line 4801, col 36, terminal 'xterm+256color': limiting value of `pairs' from 0x10000 to 0x7fff
"terminfo.tmp", line 4820, col 36, terminal 'xterm+256setaf': limiting value of `pairs' from 0x10000 to 0x7fff
"terminfo.tmp", line 4864, col 25, terminal 'xterm+direct2': limiting value of `colors' from 0x1000000 to 0x7fff
"terminfo.tmp", line 4864, col 40, terminal 'xterm+direct2': limiting value of `pairs' from 0x10000 to 0x7fff
"terminfo.tmp", line 4879, col 25, terminal 'xterm+direct': limiting value of `colors' from 0x1000000 to 0x7fff
"terminfo.tmp", line 4879, col 40, terminal 'xterm+direct': limiting value of `pairs' from 0x10000 to 0x7fff
"terminfo.tmp", line 4901, col 25, terminal 'xterm+indirect': limiting value of `colors' from 0x1000000 to 0x7fff
"terminfo.tmp", line 4901, col 40, terminal 'xterm+indirect': limiting value of `pairs' from 0x10000 to 0x7fff
"terminfo.tmp", line 7810, col 36, terminal 'dvtm-256color': limiting value of `pairs' from 0x10000 to 0x7fff
1750 entries written to /home/zhoulizhen/lede/staging_dir/hostpkg/share/terminfo
** built new /home/zhoulizhen/lede/staging_dir/hostpkg/share/terminfo
** sym-linked /home/zhoulizhen/lede/staging_dir/hostpkg/lib/terminfo for compatibility
installing std
installing stdcrt
installing vt100
installing vt300
finished install.data
/usr/bin/install -c ncurses-config /home/zhoulizhen/lede/staging_dir/hostpkg/bin/ncurses6-config
make[4]: Leaving directory '/home/zhoulizhen/lede/build_dir/hostpkg/ncurses-6.2/misc'
make[3]: Leaving directory '/home/zhoulizhen/lede/build_dir/hostpkg/ncurses-6.2'
mkdir -p /home/zhoulizhen/lede/staging_dir/hostpkg/stamp
touch /home/zhoulizhen/lede/build_dir/hostpkg/ncurses-6.2/.built
touch /home/zhoulizhen/lede/staging_dir/hostpkg/stamp/.ncurses_installed
make[2]: Leaving directory '/home/zhoulizhen/lede/package/libs/ncurses'
time: package/libs/ncurses/host-compile#22.27#5.42#34.80
make[2]: Entering directory '/home/zhoulizhen/lede/package/libs/toolchain'
touch /home/zhoulizhen/lede/build_dir/target-mipsel_24kc_musl/toolchain/.prepared_39f701e73c609898e872f212c6da595e_6664517399ebbbc92a37c5bb081b5c53_check
mkdir -p /home/zhoulizhen/lede/build_dir/target-mipsel_24kc_musl/toolchain
touch /home/zhoulizhen/lede/build_dir/target-mipsel_24kc_musl/toolchain/.prepared_39f701e73c609898e872f212c6da595e_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/zhoulizhen/lede/build_dir/target-mipsel_24kc_musl/toolchain/.configured_*
rm -f /home/zhoulizhen/lede/staging_dir/target-mipsel_24kc_musl/stamp/.toolchain_installed
(cd /home/zhoulizhen/lede/build_dir/target-mipsel_24kc_musl/toolchain/./; if [ -x ./configure ]; then find /home/zhoulizhen/lede/build_dir/target-mipsel_24kc_musl/toolchain/ -name config.guess | xargs -r chmod u+w; find /home/zhoulizhen/lede/build_dir/target-mipsel_24kc_musl/toolchain/ -name config.guess | xargs -r -n1 cp --remove-destination /home/zhoulizhen/lede/scripts/config.guess; find /home/zhoulizhen/lede/build_dir/target-mipsel_24kc_musl/toolchain/ -name config.sub | xargs -r chmod u+w; find /home/zhoulizhen/lede/build_dir/target-mipsel_24kc_musl/toolchain/ -name config.sub | xargs -r -n1 cp --remove-destination /home/zhoulizhen/lede/scripts/config.sub; AR="mipsel-openwrt-linux-musl-gcc-ar" AS="mipsel-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -ffile-prefix-map=/home/zhoulizhen/lede/build_dir/target-mipsel_24kc_musl/toolchain=toolchain -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=mipsel-openwrt-linux-musl-ld NM="mipsel-openwrt-linux-musl-gcc-nm" CC="mipsel-openwrt-linux-musl-gcc" GCC="mipsel-openwrt-linux-musl-gcc" CXX="mipsel-openwrt-linux-musl-g++" RANLIB="mipsel-openwrt-linux-musl-gcc-ranlib" STRIP=mipsel-openwrt-linux-musl-strip OBJCOPY=mipsel-openwrt-linux-musl-objcopy OBJDUMP=mipsel-openwrt-linux-musl-objdump SIZE=mipsel-openwrt-linux-musl-size CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -ffile-prefix-map=/home/zhoulizhen/lede/build_dir/target-mipsel_24kc_musl/toolchain=toolchain -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -ffile-prefix-map=/home/zhoulizhen/lede/build_dir/target-mipsel_24kc_musl/toolchain=toolchain -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/zhoulizhen/lede/staging_dir/target-mipsel_24kc_musl/usr/include -I/home/zhoulizhen/lede/staging_dir/toolchain-mipsel [... cut ...]
make[2]: *** [/home/zhoulizhen/lede/build_dir/target-mipsel_24kc_musl/toolchain/.pkgdir/libc.installed] Error 1
make[2]: Leaving directory '/home/zhoulizhen/lede/package/libs/toolchain'
time: package/libs/toolchain/compile#0.13#0.04#0.15
package/Makefile:111: recipe for target 'package/libs/toolchain/compile' failed
make[1]: *** [package/libs/toolchain/compile] Error 2
make[1]: Leaving directory '/home/zhoulizhen/lede'
/home/zhoulizhen/lede/include/toplevel.mk:222: recipe for target 'package/strongswan/compile' failed

zhoulizhengithub, 2021-06-14:
谁帮我看一下要怎么更新呢，之前的老版本4。14，的源码就没问题但我想用新的

zhoulizhengithub, 2021-06-14:
> 你能测试吗？
> [coolsnowwolf/包#161](https://github.com/coolsnowwolf/packages/pull/161)

能加一下你的QQ或WX吗？我的386515350，我不想用回老版本了，请求更新一下，加我微信我加鸡腿200元谢谢

EnnawYang, 2021-06-14:
哦吼

Beginner-Go, 2021-06-14:
只更新了Makefile？

zhoulizhengithub, 2021-06-14:
> 只更新了Makefile？
您可以加一下我微信或QQ吗？我不太会帮下忙谢谢，付你辛苦费谢谢，我太小白了不会更新这个的节点
git checkout 7d3671f7fe73b1ac52b1b3c22eb0aa676673a46b
fatal: 引用不是一个树：7d3671f7fe73b1ac52b1b3c22eb0aa676673a46b

Beginner-Go, 2021-06-14:
我更喜欢用Telegram

zhoulizhengithub, 2021-06-14:
> Telegram

您可以用TODESK 或向日葵远程一下吗

zhoulizhengithub, 2021-06-14:
> 只更新了Makefile？

/home/zhoulizhen/lede/build_dir/target-mipsel_24kc_musl/toolchain/.pkgdir/libc/usr/bin
cp -fpR /home/zhoulizhen/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib/ld-musl-.so /home/zhoulizhen/lede/build_dir/target-mipsel_24kc_musl/toolchain/.pkgdir/libc/lib/
cp: cannot stat '/home/zhoulizhen/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib/ld-musl-.so': No such file or directory
Makefile:750: recipe for target '/home/zhoulizhen/lede/build_dir/target-mipsel_24kc_musl/toolchain/.pkgdir/libc.installed' failed
make[2]: *** [/home/zhoulizhen/lede/build_dir/target-mipsel_24kc_musl/toolchain/.pkgdir/libc.installed] Error 1

这个是5.9.2 ,没有对应的库文件，但我不会怎么处理

Beginner-Go, 2021-06-14:
k5.4内核？

zhoulizhengithub, 2021-06-14:
> k5.4内核？

是的，就是lede的这个源码，红米 AC2100

zhoulizhengithub, 2021-06-14:
> k5.4内核？

您可以测试一下吗，

Beginner-Go, 2021-06-14:
用todesk  我看看吧

zhoulizhengithub, 2021-06-14:
> 用todesk 我看看吧

ToDesk:
设备代码:799 211 971
临时密码:518518

Beginner-Go, 2021-06-14:
快删了吧

closed 2021-06-17 by coolsnowwolf