coolsnowwolf/lede #6971: R4S编译失败

IvanSolis1989, 2021-06-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：R4S编译失败

### (2) 路由器型号和固件版本

A：R4S 4G编译失败

### (3) 详细日志

A：make[4]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.124'
  CC [M]  /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/batman-adv-2021.1/net/batman-adv/bridge_loop_avoidance.o
/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/batman-adv-2021.1/net/batman-adv/bridge_loop_avoidance.c: In function 'batadv_bla_send_claim':
/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/batman-adv-2021.1/net/batman-adv/bridge_loop_avoidance.c:440:2: error: implicit declaration of function 'netif_rx_any_context'; did you mean 'get_task_io_context'? [-Werror=implicit-function-declaration]
  netif_rx_any_context(skb);
  ^~~~~~~~~~~~~~~~~~~~
  get_task_io_context
cc1: some warnings being treated as errors
make[5]: *** [/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/batman-adv-2021.1/net/batman-adv/bridge_loop_avoidance.o] Error 1
scripts/Makefile.build:261: recipe for target '/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/batman-adv-2021.1/net/batman-adv/bridge_loop_avoidance.o' failed
Makefile:1734: recipe for target '/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/batman-adv-2021.1/net/batman-adv' failed
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.124'
Makefile:89: recipe for target '/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/batman-adv-2021.1/.built' failed
make[3]: Leaving directory '/workdir/openwrt/feeds/routing/batman-adv'
make[4]: *** [/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/batman-adv-2021.1/net/batman-adv] Error 2
make[3]: *** [/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/batman-adv-2021.1/.built] Error 2
time: package/feeds/routing/batman-adv/compile#1.63#0.37#1.99
package/Makefile:111: recipe for target 'package/feeds/routing/batman-adv/compile' failed
make[2]: *** [package/feeds/routing/batman-adv/compile] Error 2
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [/workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
package/Makefile:107: recipe for target '/workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile' failed
make[1]: Leaving directory '/workdir/openwrt'
/workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2
Error: Process completed with exit code 2.

adamgonglj, 2021-06-16:
@coolsnowwolf 默认配置编译R4S失败。

adamgonglj, 2021-06-16:
autoreconf: running: /home/adam/lede/staging_dir/host/bin/aclocal -I /home/adam/lede/staging_dir/target-aarch64_generic_musl/host/share/aclocal -I /home/adam/lede/staging_dir/hostpkg/share/aclocal -I /home/adam/lede/staging_dir/target-aarch64_generic_musl/usr/share/aclocal -I m4 -I . --force -I m4
autoreconf: configure.ac: tracing
autoreconf: configure.ac: adding subdirectory libsodium to autoreconf
autoreconf: Entering directory `libsodium'
autoreconf: running: /home/adam/lede/staging_dir/host/bin/libtoolize --install --force
autoreconf: running: /home/adam/lede/staging_dir/host/bin/autoconf --include=/home/adam/lede/staging_dir/target-aarch64_generic_musl/host/share/aclocal --include=/home/adam/lede/staging_dir/hostpkg/share/aclocal --include=/home/adam/lede/staging_dir/target-aarch64_generic_musl/usr/share/aclocal --include=m4 --include=. --prepend-include=/home/adam/lede/staging_dir/host/share/aclocal --force
autoreconf: configure.ac: not using Autoheader
autoreconf: running: /home/adam/lede/staging_dir/host/bin/automake --add-missing --force-missing
configure.ac:150: installing 'build-aux/compile'
autoreconf: Leaving directory `libsodium'
autoreconf: running: /home/adam/lede/staging_dir/host/bin/autoheader --include=/home/adam/lede/staging_dir/target-aarch64_generic_musl/host/share/aclocal --include=/home/adam/lede/staging_dir/hostpkg/share/aclocal --include=/home/adam/lede/staging_dir/target-aarch64_generic_musl/usr/share/aclocal --include=m4 --include=. --prepend-include=/home/adam/lede/staging_dir/host/share/aclocal --force
autoreconf: Leaving directory `.'
configure: WARNING: using cross tools not prefixed with host triplet
configure: WARNING: result yes guessed because of cross compilation
/home/adam/lede/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ar: `u' modifier ignored since `D' is the default (see `U')
/home/adam/lede/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ar: `u' modifier ignored since `D' is the default (see `U')
/home/adam/lede/staging_dir/toolchain-aarch64_generic_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ar: `u' modifier ignored since `D' is the default (see `U')
OpenWrt-libtool: install: warning: remember to run `libtool --finish /usr/lib'
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.cdn.openwrt.org/tcping-0.3.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.openwrt.org/tcping-0.3.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/tcping-0.3.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
No more mirrors to try - giving up.
strip: Unable to recognise the format of the input file `tcping'
make[4]: *** [Makefile:12: tcping] Error 1
make[3]: *** [Makefile:48: /home/adam/lede/build_dir/target-aarch64_generic_musl/tcping-0.3/.built] Error 2
make[2]: *** [package/Makefile:116: package/helloworld/tcping/compile] Error 1
make[1]: *** [package/Makefile:110: /home/adam/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make: *** [/home/adam/lede/include/toplevel.mk:230: world] Error 2
lede.sh: line 38: i: command not found

closed 2021-06-17 by coolsnowwolf