coolsnowwolf/lede #6972: 下载错误

qqsir-dev, 2021-06-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
SHELL= flock /home/qqsir/lede/tmp/.vlmcsd-svn1113.tar.gz.flock -c '     /home/qqsir/lede/scripts/download.pl "/home/qqsir/lede/dl" "vlmcsd-svn1113.tar.g                           z" "62f55c48f5de1249c2348ab6b96dabbe7e38899230954b0c8774efb01d9c42cc" "" "https://codeload.github.com/Wind4/vlmcsd/tar.gz/svn1113?"    '
Hash of the local file vlmcsd-svn1113.tar.gz does not match (file: 9510cf4e48e809b09fea60f29e41d3a0fb0f58257644dc60929ad503e8a156af, requested: 62f55c48                           f5de1249c2348ab6b96dabbe7e38899230954b0c8774efb01d9c42cc) - deleting download.
make[2]: *** [Makefile:45: /home/qqsir/lede/dl/vlmcsd-svn1113.tar.gz] Error 2
make[2]: Leaving directory '/home/qqsir/lede/package/lean/vlmcsd'
time: package/lean/vlmcsd/download#0.10#0.02#0.11
    ERROR: package/lean/vlmcsd failed to build.
make[1]: *** [package/Makefile:116: package/lean/vlmcsd/download] Error 1

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

closed 2021-06-16 by qqsir-dev

m-aslo, 2021-07-20:
请问你最后解决这个问题了吗？

AdorableJiang, 2021-08-28:
我也遇到了这个问题