coolsnowwolf/lede #6975: luci-app-passwall和luci-app-vssr缺少依赖包shadowsocksr-libev-alt和shadowsocksr-libev-server

timesnewmen, 2021-06-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
第一次自己编译. 碰到这个问题. 还请大佬们帮忙看看.
缺少依赖、编译时报错. 看上去是因为luci-app-passwall和luci-app-vssr缺少依赖导致的.

`feeds.conf.default` 已经包含了small仓库.
src-git packages https://github.com/coolsnowwolf/packages
src-git luci https://github.com/coolsnowwolf/luci
src-git routing https://git.openwrt.org/feed/routing.git
src-git telephony https://git.openwrt.org/feed/telephony.git
#src-git video https://github.com/openwrt/video.git
#src-git targets https://github.com/openwrt/targets.git
#src-git oldpackages http://git.openwrt.org/packages.git
#src-link custom /usr/src/openwrt/custom-feed
src-git helloworld https://github.com/fw876/helloworld
src-git kenzo https://github.com/V2RaySSR/openwrt-packages
src-git small https://github.com/kenzok8/small
A：

### (2) 路由器型号和固件版本
Redmi2100
A：

### (3) 详细日志
Collected errors:
```
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-passwall:
 *      shadowsocksr-libev-alt
 *      shadowsocksr-libev-server
 * opkg_install_cmd: Cannot install package luci-app-passwall.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-vssr:
 *      shadowsocksr-libev-alt
 *      shadowsocksr-libev-server
 * opkg_install_cmd: Cannot install package luci-app-vssr.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/openwrt/lede'
package/Makefile:108: recipe for target '/home/openwrt/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_install' failed
make[1]: *** [/home/openwrt/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/openwrt/lede'
/home/openwrt/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2
A：
```

Beginner-Go, 2021-06-14:
这边建议用原作者源
[vssr](https://github.com/jerrykuku/openwrt-package)
[passwall](https://github.com/xiaorouji/openwrt-passwall)

timesnewmen, 2021-06-14:
> 这边建议用原作者源
> [vssr](https://github.com/jerrykuku/openwrt-package)
> [passwall](https://github.com/xiaorouji/openwrt-passwall)

非常感谢. 确实可以. 而且我发现把包直接clone下来成功率要高. 我再编译一次试试.

closed 2021-06-14 by timesnewmen