coolsnowwolf/lede #6977: 大版本更新后X86编译成功，保留配置升级后无法启动，虚拟环境

chenxuan5858, 2021-06-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
大版本更新后X86编译成功，保留配置升级后无法启动，虚拟环境
Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：大版本更新后X86编译成功，保留配置升级后无法启动，虚拟环境下卡在如下日志，无法启动完成

### (2) 路由器型号和固件版本

A：X86 VMWARE 6.7U2

### (3) 详细日志
[image]
A：

chenxuan5858, 2021-06-14:
之前一直是使用EFI模式启动，磁盘控制器为SCSI，正常启动使用，但是这次更新后就出现了问题

chenxuan5858, 2021-06-14:
[image]
这是之前的磁盘配置，正常启动

deplives, 2021-06-15:
同无法启动，j4125软路由，之前用的是uefi启动，现在不管combined还是efi都启动不起来了，我的是一直卡在了GRUB loading

chenxuan5858, 2021-06-15:
> 同无法启动，j4125软路由，之前用的是uefi启动，现在不管combined还是efi都启动不起来了，我的是一直卡在了GRUB loading

是的，两种方式都无法启动

hcym, 2021-06-15:
大雕的x86启动确实麻烦，我是用21.02加入插件编译，随便用U盘还是nvme都正常，其他架构还是用大雕源码

shu0802, 2021-06-15:
> 反馈bug/问题模板，提建议请删除
> 
> ## 1.关于你要提交的问题
> 大版本更新后X86编译成功，保留配置升级后无法启动，虚拟环境
> Q：是否搜索了issue (使用 "x" 选择)
> 
> * [x]  没有类似的issue
> 
> ## 2. 详细叙述
> ### (1) 具体问题
> A：大版本更新后X86编译成功，保留配置升级后无法启动，虚拟环境下卡在如下日志，无法启动完成
> 
> ### (2) 路由器型号和固件版本
> A：X86 VMWARE 6.7U2
> 
> ### (3) 详细日志
> [image]
> A：

用写盘工具重新刷入都不行吗

closed 2021-06-16 by coolsnowwolf