coolsnowwolf/lede #6979: X86 新版本传统无法引导而uefi就可以引导

brianzc0306, 2021-06-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
新版本传统无法引导而uefi就可以引导
### (1) 具体问题

A：新版本传统无法引导而uefi就可以引导

### (2) 路由器型号和固件版本

A：4415u PVE6.4.* 虚拟机

### (3) 详细日志
[image]

A：

chenxuan5858, 2021-06-15:
> 反馈bug/问题模板，提建议请删除
> 
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
> 
> * [ x] 没有类似的issue
> 
> ## 2. 详细叙述
> 新版本传统无法引导而uefi就可以引导
> 
> ### (1) 具体问题
> A：新版本传统无法引导而uefi就可以引导
> 
> ### (2) 路由器型号和固件版本
> A：4415u PVE6.4.* 虚拟机
> OpenWrt R21.6.14 / LuCI Master (git-21.165.36547-60d6dfc)
> 
> ### (3) 详细日志
> [image]
> 
> A：

UEFI启动成功了吗？麻烦问下您的磁盘是怎么配置的

brianzc0306, 2021-06-15:
> > 反馈bug/问题模板，提建议请删除
> > ## 1.关于你要提交的问题
> > Q：是否搜索了issue (使用 "x" 选择)
> > 
> > * [ x] 没有类似的issue
> > 
> > ## 2. 详细叙述
> > 新版本传统无法引导而uefi就可以引导
> > ### (1) 具体问题
> > A：新版本传统无法引导而uefi就可以引导
> > ### (2) 路由器型号和固件版本
> > A：4415u PVE6.4.* 虚拟机
> > OpenWrt R21.6.14 / LuCI Master (git-21.165.36547-60d6dfc)
> > ### (3) 详细日志
> > [image]
> > A：
> 
> UEFI启动成功了吗？麻烦问下您的磁盘是怎么配置的

UEFI 成功起启动
磁盘不就是 pve虚拟机 qm importdisk 101 openwrt-x86-64-uefi-gpt-squashfs.vmdk local-lvm

demo0102, 2021-06-15:
同样的问题，本人环境为群晖VMM虚拟机

pangfeng161, 2021-06-15:
感觉新内核配置还不完善,刷入u盘出错分区表出错

13810169, 2021-06-15:
我也是pve，使用传统方法引导卡在grub这里，没有编译uefi的镜像，无法测试。

closed 2021-06-15 by brianzc0306