coolsnowwolf/lede #6981: ERROR: package/feeds/packages/dockerd failed to build

DreamWalkerXZ, 2021-06-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：`ERROR: package/feeds/packages/dockerd failed to build`

### (2) 路由器型号和固件版本

A：型号：x86-64(J1900); 固件版本：R21.6.14

### (3) 详细日志

A：
```
make[3]: Entering directory '/home/user/lede-x86/feeds/packages/utils/dockerd'
mkdir -p /home/user/lede-x86/dl
SHELL= flock /home/user/lede-x86/tmp/.dockerd-20.10.7.tar.gz.flock -c '  	/home/user/lede-x86/scripts/download.pl "/home/user/lede-x86/dl" "dockerd-20.10.7.tar.gz" "b80142035de46904605fb7b8f18075cd94154f8c3d67ff346ea554d1e9d579b9" "" "https://codeload.github.com/moby/moby/tar.gz/v20.10.7?"    '
touch /home/user/lede-x86/build_dir/target-x86_64_musl/dockerd-20.10.7/.prepared_ee4f7796d2dc04a4963303d3eed3054f_6664517399ebbbc92a37c5bb081b5c53_check
. /home/user/lede-x86/include/shell.sh; gzip -dc /home/user/lede-x86/dl/dockerd-20.10.7.tar.gz | tar -C /home/user/lede-x86/build_dir/target-x86_64_musl/dockerd-20.10.7 --strip-components 1 -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/user/lede-x86/build_dir/target-x86_64_musl/dockerd-20.10.7

Applying ./patches/001-libdevmapper_name_fix.patch using plaintext:
patching file hack/make.sh
# Verify dependencies are the vendored version
( DEP_VER=$( grep --only-matching --perl-regexp '(?<=PKG_SOURCE_VERSION:=)(.*)' "../containerd/Makefile" ); VEN_VER=$( grep --only-matching --perl-regexp '(?<=_COMMIT:=)(.*)(?=})' "/home/user/lede-x86/build_dir/target-x86_64_musl/dockerd-20.10.7/hack/dockerfile/install/containerd.installer" ); if [ "${VEN_VER}" != "${DEP_VER}" ]; then echo "ERROR: Expected 'PKG_SOURCE_VERSION:=${VEN_VER}' in '../containerd/Makefile', found 'PKG_SOURCE_VERSION:=${DEP_VER}'"; exit 1; fi )
( DEP_VER=$( grep --only-matching --perl-regexp '(?<=PKG_SOURCE_VERSION:=)(.*)' "../libnetwork/Makefile" ); VEN_VER=$( grep --only-matching --perl-regexp '(?<=_COMMIT:=)(.*)(?=})' "/home/user/lede-x86/build_dir/target-x86_64_musl/dockerd-20.10.7/hack/dockerfile/install/proxy.installer" ); if [ "${VEN_VER}" != "${DEP_VER}" ]; then echo "ERROR: Expected 'PKG_SOURCE_VERSION:=${VEN_VER}' in '../libnetwork/Makefile', found 'PKG_SOURCE_VERSION:=${DEP_VER}'"; exit 1; fi )
( DEP_VER=$( grep --only-matching --perl-regexp '(?<=PKG_SOURCE_VERSION:=)(.*)' "../runc/Makefile" ); VEN_VER=$( grep --only-matching --perl-regexp '(?<=_COMMIT:=)(.*)(?=})' "/home/user/lede-x86/build_dir/target-x86_64_musl/dockerd-20.10.7/hack/dockerfile/install/runc.installer" ); if [ "${VEN_VER}" != "${DEP_VER}" ]; then echo "ERROR: Expected 'PKG_SOURCE_VERSION:=${VEN_VER}' in '../runc/Makefile', found 'PKG_SOURCE_VERSION:=${DEP_VER}'"; exit 1; fi )
( DEP_VER=$( grep --only-matching --perl-regexp '(?<=PKG_SOURCE_VERSION:=)(.*)' "../tini/Makefile" ); VEN_VER=$( grep --only-matching --perl-regexp '(?<=_COMMIT:=)(.*)(?=})' "/home/user/lede-x86/build_dir/target-x86_64_musl/dockerd-20.10.7/hack/dockerfile/install/tini.installer" ); if [ "${VEN_VER}" != "${DEP_VER}" ]; then echo "ERROR: Expected 'PKG_SOURCE_VERSION:=${VEN_VER}' in '../tini/Makefile', found 'PKG_SOURCE_VERSION:=${DEP_VER}'"; exit 1; fi )
ERROR: Expected 'PKG_SOURCE_VERSION:=de40ad007797e0dcd8b7126f27bb87401d224240' in '../tini/Makefile', found 'PKG_SOURCE_VERSION:=fec3683b971d9c3ef73f284f176672c44b448662'
make[3]: *** [Makefile:167: /home/user/lede-x86/build_dir/target-x86_64_musl/dockerd-20.10.7/.prepared_ee4f7796d2dc04a4963303d3eed3054f_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/user/lede-x86/feeds/packages/utils/dockerd'
time: package/feeds/packages/dockerd/compile#0.44#0.40#0.65
    ERROR: package/feeds/packages/dockerd failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/dockerd/compile] Error 1
make[2]: Leaving directory '/home/user/lede-x86'
make[1]: *** [package/Makefile:110: /home/user/lede-x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/user/lede-x86'
make: *** [/home/user/lede-x86/include/toplevel.mk:230: world] Error 2
```

c939984606, 2021-06-15:
同样  ERROR: package/feeds/packages/dockerd failed to build

slippersheepig, 2021-06-16:
同J1900，一样的报错

closed 2021-06-17 by coolsnowwolf