coolsnowwolf/lede #6987: 云编译x86-64编译报错，请大神们看看

281677160, 2021-06-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

2021-06-15T06:24:50.4822464Z  /usr/bin/mkdir -p '/workdir/openwrt/build_dir/target-x86_64_musl/sqlite-autoconf-3320100/ipkg-install/usr/include'
2021-06-15T06:24:50.4852289Z  /workdir/openwrt/staging_dir/host/bin/install -c -m 644 sqlite3.h sqlite3ext.h '/workdir/openwrt/build_dir/target-x86_64_musl/sqlite-autoconf-3320100/ipkg-install/usr/include'
2021-06-15T06:24:50.4889701Z  /usr/bin/mkdir -p '/workdir/openwrt/build_dir/target-x86_64_musl/sqlite-autoconf-3320100/ipkg-install/usr/man/man1'
2021-06-15T06:24:50.4944411Z  /workdir/openwrt/staging_dir/host/bin/install -c -m 644 sqlite3.1 '/workdir/openwrt/build_dir/target-x86_64_musl/sqlite-autoconf-3320100/ipkg-install/usr/man/man1'
2021-06-15T06:24:50.4971679Z  /usr/bin/mkdir -p '/workdir/openwrt/build_dir/target-x86_64_musl/sqlite-autoconf-3320100/ipkg-install/usr/lib/pkgconfig'
2021-06-15T06:24:50.5000426Z  /workdir/openwrt/staging_dir/host/bin/install -c -m 644 sqlite3.pc '/workdir/openwrt/build_dir/target-x86_64_musl/sqlite-autoconf-3320100/ipkg-install/usr/lib/pkgconfig'
2021-06-15T06:24:50.5015753Z make[5]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/sqlite-autoconf-3320100'
2021-06-15T06:24:50.5017348Z make[4]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/sqlite-autoconf-3320100'
2021-06-15T06:24:50.5019066Z touch /workdir/openwrt/build_dir/target-x86_64_musl/sqlite-autoconf-3320100/.built
2021-06-15T06:24:50.5044488Z rm -rf /workdir/openwrt/build_dir/target-x86_64_musl/sqlite-autoconf-3320100/.pkgdir/libsqlite3.installed /workdir/openwrt/build_dir/target-x86_64_musl/sqlite-autoconf-3320100/.pkgdir/libsqlite3
2021-06-15T06:24:50.5069108Z mkdir -p /workdir/openwrt/build_dir/target-x86_64_musl/sqlite-autoconf-3320100/.pkgdir/libsqlite3
2021-06-15T06:24:50.5091913Z install -d -m0755 /workdir/openwrt/build_dir/target-x86_64_musl/sqlite-autoconf-3320100/.pkgdir/libsqlite3/usr/lib
2021-06-15T06:24:50.5114155Z cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/sqlite-autoconf-3320100/ipkg-install/usr/lib/libsqlite3.so.0* /workdir/openwrt/build_dir/target-x86_64_musl/sqlite-autoconf-3320100/.pkgdir/libsqlite3/usr/lib
2021-06-15T06:24:50.5150411Z touch /workdir/openwrt/build_dir/target-x86_64_musl/sqlite-autoconf-3320100/.pkgdir/libsqlite3.installed
2021-06-15T06:24:50.5168913Z mkdir -p /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
2021-06-15T06:24:50.5191150Z echo '0' | cmp -s - /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/libsqlite3.version || { echo '0' > /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/libsqlite3.version;  } 
2021-06-15T06:24:50.5215634Z SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/sqlite-autoconf-3320100/.pkgdir/libsqlite3/. /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/'
2021-06-15T06:24:50.5265043Z touch /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.libsqlite3_installed
2021-06-15T06:24:50.5400788Z removed '/workdir/openwrt/bin/packages/x86_64/packages/libsqlite30_3320100-1_x86_64.ipk'
2021-06-15T06:24:50.5408571Z mkdir -p /workdir/openwrt/bin/targets/x86/64/packages /workdir/openwrt/build_dir/target-x86_64_musl/sqlite-autoconf-3320100/ipkg-x86_64/libsqlite3/CONTROL /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo
2021-06-15T06:24:50.5429700Z install -d -m0755 /workdir/openwrt/build_dir/target-x86_64_musl/sqlite-autoconf-3320100/ipkg-x86_64/libsqlite3/usr/lib
2021-06-15T06:24:50.5455449Z cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/sqlite-autoconf-3320100/ipkg-install/usr/lib/libsqlite3.so.0* /workdir/openwrt/build_dir/target-x86_64_musl/sqlite-autoconf-3320100/ipkg-x86_64/libsqlite3/usr/lib
2021-06-15T06:24:50.5485318Z find /workdir/openwrt/build_dir/target-x86_64_musl/sqlite-autoconf-3320100/ipkg-x86_64/libsqlite3 -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
2021-06-15T06:24:50.5699037Z export CROSS="x86_64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="x86_64-openwrt-linux-musl-nm" STRIP="/workdir/openwrt/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/workdir/openwrt/scripts/strip-kmod.sh" PATCHELF="/workdir/openwrt/staging_dir/host/bin/patchelf" /workdir/openwrt/scripts/rstrip.sh /workdir/openwrt/build_dir/target-x86_64_musl/sqlite-autoconf-3320100/ipkg-x86_64/libsqlite3
2021-06-15T06:24:50.5771531Z rstrip.sh: /workdir/openwrt/build_dir/target-x86_64_musl/sqlite-autoconf-3320100/ipkg-x86_64/libsqlite3/usr/lib/libsqlite3.so.0.8.6: shared object
2021-06-15T06:24:50.5831171Z (cd /workdir/openwrt/build_dir/target-x86_64_musl/sqlite-autoconf-3320100/ipkg-x86_64/libsqlite3/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
2021-06-15T06:24:50.5886607Z install -d -m0755 /workdir/openwrt/bin/packages/x86_64/packages
2021-06-15T06:24:50.5907473Z /workdir/openwrt/staging_dir/host/bin/fakeroot /workdir/openwrt/scripts/ipkg-build -m "" /workdir/openwrt/build_dir/target-x86_64_musl/sqlite-autoconf-3320100/ipkg-x86_64/libsqlite3 /workdir/openwrt/bin/packages/x86_64/packages
2021-06-15T06:24:50.7390942Z Packaged contents of /workdir/openwrt/build_dir/target-x86_64_musl/sqlite-autoconf-3320100/ipkg-x86_64/libsqlite3 into /workdir/openwrt/bin/packages/x86_64/packages/libsqlite30_3320100-1_x86_64.ipk
2021-06-15T06:24:50.7472493Z echo "libsqlite3" >> /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/sqlite3.default.install
2021-06-15T06:24:50.7489479Z rm -rf /workdir/openwrt/tmp/stage-sqlite3
2021-06-15T06:24:50.7512374Z mkdir -p /workdir/openwrt/tmp/stage-sqlite3/host /workdir/openwrt/staging_dir/target-x86_64_musl/packages /workdir/openwrt/staging_dir/host/packages
2021-06-15T06:24:50.7534437Z install -d -m0755 /workdir/openwrt/tmp/stage-sqlite3/usr/include
2021-06-15T06:24:50.7559355Z install -d -m0755 /workdir/openwrt/tmp/stage-sqlite3/usr/lib/pkgconfig
2021-06-15T06:24:50.7582996Z cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/sqlite-autoconf-3320100/ipkg-install/usr/lib/libsqlite3.{a,so*} /workdir/openwrt/tmp/stage-sqlite3/usr/lib
2021-06-15T06:24:50.7649724Z install -m0644 /workdir/openwrt/build_dir/target-x86_64_musl/sqlite-autoconf-3320100/ipkg-install/usr/include/sqlite3{,ext}.h /workdir/openwrt/tmp/stage-sqlite3/usr/include
2021-06-15T06:24:50.7682431Z install -m0644 /workdir/openwrt/build_dir/target-x86_64_musl/sqlite-autoconf-3320100/ipkg-install/usr/lib/pkgconfig/sqlite3.pc /workdir/openwrt/tmp/stage-sqlite3/usr/lib/pkgconfig
2021-06-15T06:24:50.7709687Z find /workdir/openwrt/tmp/stage-sqlite3 -name '*.la' | xargs -r rm -f; 
2021-06-15T06:24:50.7739110Z if [ -f /workdir/openwrt/staging_dir/target-x86_64_musl/packages/sqlite3.list ]; then /workdir/openwrt/scripts/clean-package.sh "/workdir/openwrt/staging_dir/target-x86_64_musl/packages/sqlite3.list" "/workdir/openwrt/staging_dir/target-x86_64_musl"; fi
2021-06-15T06:24:50.7871564Z if [ -d /workdir/openwrt/tmp/stage-sqlite3 ]; then (cd /workdir/openwrt/tmp/stage-sqlite3; find ./ > /workdir/openwrt/tmp/stage-sqlite3.files); 	SHELL= flock /workdir/openwrt/tmp/.staging-dir.flock -c ' mv /workdir/openwrt/tmp/stage-sqlite3.files /workdir/openwrt/staging_dir/target-x86_64_musl/packages/sqlite3.list && cp -fpR /workdir/openwrt/tmp/stage-sqlite3/* /workdir/openwrt/staging_dir/target-x86_64_musl/; '; fi
2021-06-15T06:24:50.7972009Z rm -rf /workdir/openwrt/tmp/stage-sqlite3
2021-06-15T06:24:50.7998379Z touch /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.sqlite3_installed
2021-06-15T06:24:50.8028451Z make[3]: Leaving directory '/workdir/openwrt/feeds/packages/libs/sqlite3'
2021-06-15T06:24:50.8046865Z time: package/feeds/packages/sqlite3/compile#7.76#1.13#9.47
2021-06-15T06:24:50.8225397Z make[3]: Entering directory '/workdir/openwrt/feeds/packages/libs/nss'
2021-06-15T06:24:50.9597583Z mkdir -p /workdir/openwrt/dl
[... cut ...]
2021-06-15T06:24:51.8711482Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'core\.sshCommand' && git config --local --unset-all 'core.sshCommand' || :
2021-06-15T06:24:51.9017109Z [command]/usr/bin/git config --local --name-only --get-regexp http\.https\:\/\/github\.com\/\.extraheader
2021-06-15T06:24:51.9048058Z http.https://github.com/.extraheader
2021-06-15T06:24:51.9052416Z [command]/usr/bin/git config --local --unset-all http.https://github.com/.extraheader
2021-06-15T06:24:51.9087002Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp

shoucandanghehe, 2021-06-15:
2021-06-15T06:24:51.5693135Z make[4]: Entering directory '/workdir/openwrt/build_dir/target-x86_64_musl/nss-3.53/nss'
2021-06-15T06:24:51.5693856Z make[4]: pic: No such file or directory

281677160, 2021-06-15:
> 2021-06-15T06:24:51.5693135Z make[4]: Entering directory '/workdir/openwrt/build_dir/target-x86_64_musl/nss-3.53/nss'
> 2021-06-15T06:24:51.5693856Z make[4]: pic: No such file or directory

感谢，不知道什么解决啊，呵呵

shoucandanghehe, 2021-06-15:
问就删库

flyhong666, 2021-06-15:
本地编译也出现你那个nss 啥的出错。。。不知道如何解决。。。

281677160, 2021-06-15:
> 本地编译也出现你那个nss 啥的出错。。。不知道如何解决。。。

是啊，我也本地测试了一次，一样报错

[image]

281677160, 2021-06-15:
本地编译自带一份logs显示

Makefile:16: ../../../feeds/packages/meson/meson.mk: No such file or directory
Makefile:17: ../../../feeds/packages/ninja/ninja.mk: No such file or directory
Build-Depends: meson/host ninja/host
Build-Types: host

Package: dpdk
Version: 21.05
Depends: +libc +USE_GLIBC:librt +USE_GLIBC:libpthread @(arm) +librt +libpthread +libbpf +libpcap
Conflicts: 
Menu-Depends: 
Provides: 
Section: net
Category: Network
Repository: base
Title: Data Plane Development Kit (DPDK) 
Maintainer: 
Source: dpdk-21.05.tar.xz
Type: ipkg
Description:   DPDK consists of libraries to accelerate packet processing 
  workloads running on a wide variety of CPU architectures. 

@@

make[2]: *** No rule to make target '../../../feeds/packages/ninja/ninja.mk'.  Stop.

shoucandanghehe, 2021-06-15:
我倒是没这个报错
我的是

> /home/scdhh/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/config/i386/sync.md:2474:1: internal compiler error: Segmentation fault
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-9/README.Bugs> for instructions.
make[6]: *** [Makefile:1110: insn-emit.o] Error 1
make[6]: Leaving directory '/home/scdhh/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final/gcc'
make[5]: *** [Makefile:4270: all-gcc] Error 2
make[5]: Leaving directory '/home/scdhh/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final'
make[4]: *** [Makefile:906: all] Error 2
make[4]: Leaving directory '/home/scdhh/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final'
make[3]: *** [Makefile:91: /home/scdhh/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final/.built] Error 2
make[3]: Leaving directory '/home/scdhh/lede/toolchain/gcc/final'
time: toolchain/gcc/final/compile#27.34#0.50#27.81
    ERROR: toolchain/gcc/final failed to build.
make[2]: *** [toolchain/Makefile:97: toolchain/gcc/final/compile] Error 1
make[2]: Leaving directory '/home/scdhh/lede'
make[1]: *** [toolchain/Makefile:93: /home/scdhh/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/scdhh/lede'
make: *** [/home/scdhh/lede/include/toplevel.mk:230: world] Error 2

closed 2021-06-15 by 281677160

zhangguanzhang, 2021-08-13:
> 我倒是没这个报错
> 我的是

怎么解决的

281677160, 2021-08-13:
看不懂你的，你是拉取的最新源码编译吗?我昨晚上很晚才编译了一次，没问题啊

Boos4721, 2021-08-25:
我不李姐为什么你X86要编译nss

zhangguanzhang, 2021-08-25:
我换了config后编译成功了