coolsnowwolf/lede #6994: ./scripts/feeds install -a出现错误提示 求问怎么解决

wanhzin, 2021-06-15:
## 1.关于你要提交的问题
./scripts/feeds install -a 出现错误提示
Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
./scripts/feeds install -a出现错误提示

### (1) 具体问题
xxx@debian:~/openwrt/lede$ ./scripts/feeds install -a
Collecting package info: done
Installing all packages from feed packages.
WARNING: Not overriding core package 'frp'; use -f to force
WARNING: Not overriding core package 'kcptun'; use -f to force
WARNING: Not overriding core package 'libcap'; use -f to force
WARNING: Not overriding core package 'musl-fts'; use -f to force
WARNING: Not overriding core package 'nft-qos'; use -f to force
WARNING: Not overriding core package 'proxychains-ng'; use -f to force
WARNING: Not overriding core package 'samba4'; use -f to force
WARNING: Not overriding core package 'softethervpn5'; use -f to force
WARNING: Not overriding core package 'wget'; use -f to force
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.

没有修改feed

A：

### (2) 路由器型号和固件版本
menuconfig全部默认 x86
A：

### (3) 详细日志
xxx@debian:~/openwrt/lede$ ./scripts/feeds install -a
Collecting package info: done
Installing all packages from feed packages.
WARNING: Not overriding core package 'frp'; use -f to force
WARNING: Not overriding core package 'kcptun'; use -f to force
WARNING: Not overriding core package 'libcap'; use -f to force
WARNING: Not overriding core package 'musl-fts'; use -f to force
WARNING: Not overriding core package 'nft-qos'; use -f to force
WARNING: Not overriding core package 'proxychains-ng'; use -f to force
WARNING: Not overriding core package 'samba4'; use -f to force
WARNING: Not overriding core package 'softethervpn5'; use -f to force
WARNING: Not overriding core package 'wget'; use -f to force
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.
A：

rongcenter, 2021-06-15:
同

keeno1982, 2021-06-15:
同

asushugo, 2021-06-15:
rm -rf tmp/

XiaoliChan, 2021-06-15:
Try ```./scripts/feeds install -a -f```

lxwq, 2021-06-16:
> rm -rf tmp/

find -name wget试试,
里面有重复的包,所以才会报错,和tmp没什么关系,
我重新clone了一便,报一样的提示,

a920025608, 2021-06-16:
无视即可，为了5.10用户才有这“问题”

XiaoliChan, 2021-06-16:
> 无视即可，为了5.10用户才有这“问题”

But x86 has already switched to 5.10 kernel branch.

wanhzin, 2021-06-16:
编译成功过了 无视即可

closed 2021-06-16 by coolsnowwolf