coolsnowwolf/lede #6996: 最新的代码下，不能在“make menuconfig”中选择“helloworld”

GWONGJYULEI, 2021-06-15:
最新的代码下，不能在“make menuconfig”中选择酸酸乳Plus

* [x] 没有类似的issue

### (1) 具体问题
1.feeds.conf.default 已移除 helloworld 的源
> src-git packages https://github.com/coolsnowwolf/packages
> src-git luci https://github.com/coolsnowwolf/luci
> src-git routing https://git.openwrt.org/feed/routing.git
> src-git telephony https://git.openwrt.org/feed/telephony.git
> #src-git video https://github.com/openwrt/video.git
> #src-git targets https://github.com/openwrt/targets.git
> #src-git oldpackages http://git.openwrt.org/packages.git
> #src-link custom /usr/src/openwrt/custom-feed

2.取消所有对源备注后，编译时依然没有 helloworld 的选项

### (2) 路由器型号和固件版本

x86_64

### (3) 详细日志

无

EnnawYang, 2021-06-15:
自己加下不就完了 这算啥issue

demo0102, 2021-06-15:
关闭issue吧