coolsnowwolf/lede #7000: 编译出错，请大神们看看什么问题

281677160, 2021-06-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：
mkdir -p /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/libmaxminddb-1.6.0/.pkgdir/libmaxminddb/. /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.libmaxminddb_installed
echo "libmaxminddb" >> /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/libmaxminddb.default.install
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/libs/libmaxminddb'
time: package/feeds/packages/libmaxminddb/compile#0.17#0.03#0.19
make[3]: Entering directory '/workdir/openwrt/feeds/packages/utils/lua-maxminddb'
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.lua-maxminddb-0.2.tar.gz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "lua-maxminddb-0.2.tar.gz" "f9b870968f75c8020a311cca2482139ac8b2b55d4f1b5b0ce00f7844b083c866" "" || (  /workdir/openwrt/scripts/dl_github_archive.py --dl-dir="/workdir/openwrt/dl" --url="https://github.com/fabled/lua-maxminddb.git" --version="93da9f4e6c814c3a23044dd2cdd22d4a6b4f665b" --subdir="lua-maxminddb-0.2" --source="lua-maxminddb-0.2.tar.gz" --hash="f9b870968f75c8020a311cca2482139ac8b2b55d4f1b5b0ce00f7844b083c866" || ( 	echo "Checking out files from the git repository..."; mkdir -p /workdir/openwrt/tmp/dl && cd /workdir/openwrt/tmp/dl && rm -rf lua-maxminddb-0.2 && [ \! -d lua-maxminddb-0.2 ] && git clone  https://github.com/fabled/lua-maxminddb.git lua-maxminddb-0.2 && (cd lua-maxminddb-0.2 && git checkout 93da9f4e6c814c3a23044dd2cdd22d4a6b4f665b && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd lua-maxminddb-0.2 && git log -1 --format='\''@%ct'\''` && rm -rf lua-maxminddb-0.2/.git && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c lua-maxminddb-0.2 | 	gzip -nc > /workdir/openwrt/tmp/dl/lua-maxminddb-0.2.tar.gz && mv /workdir/openwrt/tmp/dl/lua-maxminddb-0.2.tar.gz /workdir/openwrt/dl/ && rm -rf lua-maxminddb-0.2; );  )    '
Hash of the local file lua-maxminddb-0.2.tar.gz does not match (file: 8aff0958546e4082d48854fdeb32f4fd1ef6a9f78ef066a1886f4d9569898d0e, requested: f9b870968f75c8020a311cca2482139ac8b2b55d4f1b5b0ce00f7844b083c866) - deleting download.
gzip: warning: GZIP environment variable is deprecated; use an alias or script
lua-maxminddb-0.2.tar.gz: Download from https://github.com/fabled/lua-maxminddb.git failed
lua-maxminddb-0.2.tar.gz: Wrong hash (probably caused by .gitattributes), expecting f9b870968f75c8020a311cca2482139ac8b2b55d4f1b5b0ce00f7844b083c866, got 8aff0958546e4082d48854fdeb32f4fd1ef6a9f78ef066a1886f4d9569898d0e
Checking out files from the git repository...
Cloning into 'lua-maxminddb-0.2'...
Note: switching to '93da9f4e6c814c3a23044dd2cdd22d4a6b4f665b'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 93da9f4 Fixed implicit declaration of gai_strerror() compiler warning, that lead to occational segfualts, i.e. if the address we attempt to look up is invalid
Packing checkout...
touch /workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2/.prepared_222fd5e95371af760b15044047372987_6664517399ebbbc92a37c5bb081b5c53_check
. /workdir/openwrt/include/shell.sh; gzip -dc /workdir/openwrt/dl/lua-maxminddb-0.2.tar.gz | tar -C /workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2
touch /workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2/.prepared_222fd5e95371af760b15044047372987_6664517399ebbbc92a37c5bb081b5c53
rm -f /workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2/.configured_*
rm -f /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.lua-maxminddb_installed
touch /workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2/.built
touch /workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2/.built_check
MAKEFLAGS="" /workdir/openwrt/staging_dir/host/bin/ninja  -j1 -C /workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2 
ninja: Entering directory `/workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2'
ninja: error: loading 'build.ninja': No such file or directory
make[3]: *** [Makefile:57: /workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2/.built] Error 1
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/utils/lua-maxminddb'
time: package/feeds/packages/lua-maxminddb/compile#0.33#0.09#0.73
    ERROR: package/feeds/packages/lua-maxminddb failed to build.
make[2]: Leaving directory '/workdir/openwrt'
make[2]: *** [package/Makefile:116: package/feeds/packages/lua-maxminddb/compile] Error 1
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2

请问这个是V/S/S/R的lua-maxminddb依赖有问题还是什么？这里有一句ninja: error: loading 'build.ninja': No such file or directory

lyrielau, 2021-06-16:
我的也是这个情况

281677160, 2021-06-16:
> 我的也是这个情况

那就是要等大雕修复了？呵呵，编译luci-app-oaf也出错
2021-06-15T20:37:26.6121002Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf-5.0/app_filter.c:772:6: warning: unused variable 'i' [-Wunused-variable]
2021-06-15T20:37:26.6121855Z   int i;
2021-06-15T20:37:26.6122200Z       ^
2021-06-15T20:37:26.6123153Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf-5.0/app_filter.c: In function 'app_filter_hook':
2021-06-15T20:37:26.6124786Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf-5.0/app_filter.c:804:21: warning: unused variable 'total_packets' [-Wunused-variable]
2021-06-15T20:37:26.6125806Z   unsigned long long total_packets = 0;
2021-06-15T20:37:26.6126290Z                      ^~~~~~~~~~~~~
2021-06-15T20:37:26.6127455Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf-5.0/app_filter.c:803:13: warning: unused variable 'bytes1' [-Wunused-variable]
2021-06-15T20:37:26.6128374Z   static int bytes1 = 0;
2021-06-15T20:37:26.6129013Z              ^~~~~~
2021-06-15T20:37:26.6130039Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf-5.0/app_filter.c: In function 'af_send_msg_to_user':
2021-06-15T20:37:26.6131823Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf-5.0/app_filter.c:997:2: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
2021-06-15T20:37:26.6133171Z   char msg_buf[MAX_OAF_NL_MSG_LEN] = {0};
2021-06-15T20:37:26.6133590Z   ^~~~
2021-06-15T20:37:26.6135015Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf-5.0/app_filter.c:1001:2: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
2021-06-15T20:37:26.6136258Z   char *p_data = msg_buf + sizeof(struct af_msg_hdr);
2021-06-15T20:37:26.6136739Z   ^~~~
2021-06-15T20:37:26.6137664Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf-5.0/app_filter.c: In function 'oaf_msg_rcv':
2021-06-15T20:37:26.6139551Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf-5.0/app_filter.c:1031:3: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
2021-06-15T20:37:26.6140811Z    struct af_msg_hdr *af_hdr = (struct af_msg_hdr *)umsg;
2021-06-15T20:37:26.6141293Z    ^~~~~~
2021-06-15T20:37:26.6142707Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf-5.0/app_filter.c:1040:3: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
2021-06-15T20:37:26.6143914Z    void *udata = umsg + sizeof(struct af_msg_hdr);
2021-06-15T20:37:26.6144385Z    ^~~~
2021-06-15T20:37:26.6291021Z At top level:
2021-06-15T20:37:26.6292675Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf-5.0/app_filter.c:803:13: warning: 'bytes1' defined but not used [-Wunused-variable]
2021-06-15T20:37:26.6294011Z   static int bytes1 = 0;
2021-06-15T20:37:26.6294431Z              ^~~~~~
2021-06-15T20:37:26.6333318Z cc1: some warnings being treated as errors
2021-06-15T20:37:26.6390444Z make[5]: *** [scripts/Makefile.build:279: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf-5.0/app_filter.o] Error 1
2021-06-15T20:37:26.6392020Z make[4]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.43'
2021-06-15T20:37:26.6393852Z make[3]: Leaving directory '/workdir/openwrt/feeds/luci/applications/luci-app-oaf/oaf'
2021-06-15T20:37:26.6395354Z make[4]: *** [Makefile:1821: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf-5.0] Error 2
2021-06-15T20:37:26.6396797Z make[3]: *** [Makefile:44: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf-5.0/.built] Error 2
2021-06-15T20:37:26.6404077Z time: package/feeds/luci/oaf/compile#1.31#0.41#1.90
2021-06-15T20:37:26.6409138Z     ERROR: package/feeds/luci/oaf failed to build.
2021-06-15T20:37:26.6412201Z make[2]: *** [package/Makefile:116: package/feeds/luci/oaf/compile] Error 1
2021-06-15T20:37:26.6420500Z make[2]: Leaving directory '/workdir/openwrt'
2021-06-15T20:37:26.6422945Z make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2021-06-15T20:37:26.6444031Z make[1]: Leaving directory '/workdir/openwrt'
2021-06-15T20:37:26.6458187Z make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
2021-06-15T20:37:26.6507980Z ##[error]Process completed with exit code 2.
2021-06-15T20:37:26.6849851Z Post job cleanup.
2021-06-15T20:37:26.9364969Z [command]/usr/bin/git version
2021-06-15T20:37:26.9413992Z git version 2.31.1
2021-06-15T20:37:26.9476214Z [command]/usr/bin/git config --local --name-only --get-regexp core\.sshCommand
2021-06-15T20:37:26.9628949Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'core\.sshCommand' && git config --local --unset-all 'core.sshCommand' || :
2021-06-15T20:37:26.9941306Z [command]/usr/bin/git config --local --name-only --get-regexp http\.https\:\/\/github\.com\/\.extraheader
2021-06-15T20:37:26.9970151Z http.https://github.com/.extraheader
2021-06-15T20:37:26.9985430Z [command]/usr/bin/git config --local --unset-all http.https://github.com/.extraheader
2021-06-15T20:37:27.0021921Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'http\.https\:\/\/github\.com\/\.extraheader' && git config --local --unset-all 'http.https://github.com/.extraheader' || :
2021-06-15T20:37:27.0414153Z Cleaning up orphan processes

liqwnt, 2021-06-16:
> 反馈bug/问题模板，提建议请删除
> 
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
> 
> * [ ]  没有类似的issue
> 
> ## 2. 详细叙述
> ### (1) 具体问题
> A：
> 
> ### (2) 路由器型号和固件版本
> A：
> 
> ### (3) 详细日志
> A：
> mkdir -p /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
> SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/libmaxminddb-1.6.0/.pkgdir/libmaxminddb/. /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/'
> touch /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.libmaxminddb_installed
> echo "libmaxminddb" >> /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/libmaxminddb.default.install
> make[3]: Leaving directory '/workdir/openwrt/feeds/packages/libs/libmaxminddb'
> time: package/feeds/packages/libmaxminddb/compile#0.17#0.03#0.19
> make[3]: Entering directory '/workdir/openwrt/feeds/packages/utils/lua-maxminddb'
> mkdir -p /workdir/openwrt/dl
> SHELL= flock /workdir/openwrt/tmp/.lua-maxminddb-0.2.tar.gz.flock -c ' /workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "lua-maxminddb-0.2.tar.gz" "f9b870968f75c8020a311cca2482139ac8b2b55d4f1b5b0ce00f7844b083c866" "" || ( /workdir/openwrt/scripts/dl_github_archive.py --dl-dir="/workdir/openwrt/dl" --url="https://github.com/fabled/lua-maxminddb.git" --version="93da9f4e6c814c3a23044dd2cdd22d4a6b4f665b" --subdir="lua-maxminddb-0.2" --source="lua-maxminddb-0.2.tar.gz" --hash="f9b870968f75c8020a311cca2482139ac8b2b55d4f1b5b0ce00f7844b083c866" || ( echo "Checking out files from the git repository..."; mkdir -p /workdir/openwrt/tmp/dl && cd /workdir/openwrt/tmp/dl && rm -rf lua-maxminddb-0.2 && [ ! -d lua-maxminddb-0.2 ] && git clone https://github.com/fabled/lua-maxminddb.git lua-maxminddb-0.2 && (cd lua-maxminddb-0.2 && git checkout 93da9f4e6c814c3a23044dd2cdd22d4a6b4f665b && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd lua-maxminddb-0.2 && git log -1 --format='\''@%ct'\''` && rm -rf lua-maxminddb-0.2/.git && tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c lua-maxminddb-0.2 | gzip -nc > /workdir/openwrt/tmp/dl/lua-maxminddb-0.2.tar.gz && mv /workdir/openwrt/tmp/dl/lua-maxminddb-0.2.tar.gz /workdir/openwrt/dl/ && rm -rf lua-maxminddb-0.2; ); ) '
> Hash of the local file lua-maxminddb-0.2.tar.gz does not match (file: 8aff0958546e4082d48854fdeb32f4fd1ef6a9f78ef066a1886f4d9569898d0e, requested: f9b870968f75c8020a311cca2482139ac8b2b55d4f1b5b0ce00f7844b083c866) - deleting download.
> gzip: warning: GZIP environment variable is deprecated; use an alias or script
> lua-maxminddb-0.2.tar.gz: Download from https://github.com/fabled/lua-maxminddb.git failed
> lua-maxminddb-0.2.tar.gz: Wrong hash (probably caused by .gitattributes), expecting f9b870968f75c8020a311cca2482139ac8b2b55d4f1b5b0ce00f7844b083c866, got 8aff0958546e4082d48854fdeb32f4fd1ef6a9f78ef066a1886f4d9569898d0e
> Checking out files from the git repository...
> Cloning into 'lua-maxminddb-0.2'...
> Note: switching to '93da9f4e6c814c3a23044dd2cdd22d4a6b4f665b'.
> 
> You are in 'detached HEAD' state. You can look around, make experimental
> changes and commit them, and you can discard any commits you make in this
> state without impacting any branches by switching back to a branch.
> 
> If you want to create a new branch to retain commits you create, you may
> do so (now or later) by using -c with the switch command. Example:
> 
> git switch -c
> 
> Or undo this operation with:
> 
> git switch -
> 
> Turn off this advice by setting config variable advice.detachedHead to false
> 
> HEAD is now at [93da9f4](https://github.com/coolsnowwolf/lede/commit/93da9f4c63003e03238d4408293645e17689f92b) Fixed implicit declaration of gai_strerror() compiler warning, that lead to occational segfualts, i.e. if the address we attempt to look up is invalid
> Packing checkout...
> touch /workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2/.prepared_222fd5e95371af760b15044047372987_6664517399ebbbc92a37c5bb081b5c53_check
> . /workdir/openwrt/include/shell.sh; gzip -dc /workdir/openwrt/dl/lua-maxminddb-0.2.tar.gz | tar -C /workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2/.. -xf -
> [ ! -d ./src/ ] || cp -fpR ./src/. /workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2
> touch /workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2/.prepared_222fd5e95371af760b15044047372987_6664517399ebbbc92a37c5bb081b5c53
> rm -f /workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2/.configured_*
> rm -f /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.lua-maxminddb_installed
> touch /workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2/.configured_68b329da9893e34099c7d8ad5cb9c940
> rm -f /workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2/.built
> touch /workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2/.built_check
> MAKEFLAGS="" /workdir/openwrt/staging_dir/host/bin/ninja -j1 -C /workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2
> ninja: Entering directory `/workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2'
> ninja: error: loading 'build.ninja': No such file or directory
> make[3]: *** [Makefile:57: /workdir/openwrt/build_dir/target-x86_64_musl/lua-maxminddb-0.2/.built] Error 1
> make[3]: Leaving directory '/workdir/openwrt/feeds/packages/utils/lua-maxminddb'
> time: package/feeds/packages/lua-maxminddb/compile#0.33#0.09#0.73
[... cut ...]
> make[2]: Leaving directory '/workdir/openwrt'
> make[2]: *** [package/Makefile:116: package/feeds/packages/lua-maxminddb/compile] Error 1
> make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
> make[1]: Leaving directory '/workdir/openwrt'
> make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
> 
> 请问这个是V/S/S/R的lua-maxminddb依赖有问题还是什么？这里有一句ninja: error: loading 'build.ninja': No such file or directory

我也是和楼主一样的情况，请问怎么解决？？？有高手在不？？？

281677160, 2021-06-16:
> 我也是和楼主一样的情况，请问怎么解决？？？有高手在不？？？

大雕已经更新完毕源码了，我已经在编译了，才2个小时，没那么快啊，起码要3个多小时才知道结果，编译完成了我告诉你结果，或者你现在也可以开始编译的

281677160, 2021-06-16:
> 大雕已经更新完毕源码了，我已经在编译了，才2个小时，没那么快啊，起码要3个多小时才知道结果，编译完成了我告诉你结果，或者你现在也可以开始编译的

还是一样的错误，不知道怎么搞了
[image]

Dimitri2020007, 2021-06-16:
出现相同的情况

HLD21, 2021-06-16:
出现相同的情况+1

bugprogrammer, 2021-06-16:
我用自己的办法解决了。作者已经更新了导致问题的MakeFile，可以在如下链接里找到并替换
https://github.com/coolsnowwolf/lua-maxminddb，把这里的MakeFile替换lede/package/openwrt-packages/lua-maxminddb下的MakeFile即可通过编译。

281677160, 2021-06-16:
> 我用自己的办法解决了。作者已经更新了导致问题的MakeFile，可以在如下链接里找到并替换
> [https://github.com/coolsnowwolf/lua-maxminddb，](https://github.com/coolsnowwolf/lua-maxminddb%EF%BC%8C) 把这里的MakeFile替换lede/package/openwrt-packages/lua-maxminddb下的MakeFile即可通过编译。

哥们，这个链接错误的吧

bugprogrammer, 2021-06-16:
链接里多了个逗号，抱歉

281677160, 2021-06-16:
感谢，在L大的仓库找到了，谢谢你！

bugprogrammer, 2021-06-16:
> 感谢，在L大的仓库找到了，谢谢你！
> […](#)

有空可以实测一下

mzl0408, 2021-06-16:
> > 感谢，在L大的仓库找到了，谢谢你！
> > […](#)
> 
> 有空可以实测一下

按照你的方法替换后正在编译，目前没遇到问题，应该解决了

mzl0408, 2021-06-16:
> > 感谢，在L大的仓库找到了，谢谢你！
> > […](#)
> 
> 有空可以实测一下

额，又报错了

bugprogrammer, 2021-06-16:
没发现作者更新代码啊，报的什么错误？

mzl0408, 2021-06-16:
> 没发现作者更新代码啊，报的什么错误？

还是一样的报错，晚上把lede文件夹删了重新来一遍

281677160, 2021-06-17:
我昨天晚上编译两次都成功了，可能作者已经修复好了

mkmichael, 2021-06-17:
删除lede项目文件夹，重新拉源代码，再编译，还是同样的错误。哎.......有解决问题的大神么？

281677160, 2021-06-17:
> 删除lede项目文件夹，重新拉源代码，再编译，还是同样的错误。哎.......有解决问题的大神么？

呃，这个诡异啊，我编译了3次都成功了

281677160, 2021-06-17:
> > > 删除lede项目文件夹，重新拉源代码，再编译，还是同样的错误。哎.......有解决问题的大神么？
> > 
> > 
> > 呃，这个诡异啊，我编译了3次都成功了
> 
> make menuconfig 这里不选 VSSR就可以了，我刚刚就是把VSSR去掉了，就顺利成功了。

我选VSSR了啊，昨天晚上1点后开始编译，都编译了3次了，都能成功，我发现新版源码里面用了Lienol大的adguardhome会自动下载核心，但是没开关选择，我拉取天灵的在测试呢

MedyMa, 2021-06-17:
lua-maxminddb 这个编译报错lean修复好了，vssr作者那也修复好了，oaf编译报错什么原因

281677160, 2021-06-17:
> lua-maxminddb 这个编译报错lean修复好了，vssr作者那也修复好了，oaf编译报错什么原因

我昨天编译oaf也报错的，后来都取消了，没测试，如果你还报错就去插件源码作者那里反映一下吧，你在这里也可以开一个issue这样大雕比较容易看到你的问题