coolsnowwolf/lede #7002: passwall编译错误

FengNning, 2021-06-15:
[1/24] Building CXX object CMakeFiles/trojan.dir/src/core/config.cpp.o
FAILED: CMakeFiles/trojan.dir/src/core/config.cpp.o 
/home/mufeng/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/bin/mipsel-openwrt-linux-musl-g++ -DBADVPN_BREACTOR_BADVPN=1 -DBADVPN_LINUX=1 -DBADVPN_LITTLE_ENDIAN=1 -DBADVPN_THREADWORK_USE_PTHREAD=1 -DBADVPN_THREAD_SAFE=1 -DBADVPN_USE_EPOLL=1 -DBADVPN_USE_SIGNALFD=1 -DBOOST_BIND_GLOBAL_PLACEHOLDERS=1 -DDEFAULT_CONFIG=\"/etc/trojan/config.json\" -DENABLE_NAT -DENABLE_REUSE_PORT -DENABLE_SSL_KEYLOG -DENABLE_TLS13_CIPHERSUITES -D_GNU_SOURCE=1 -Isrc -INAMES -Iexternal/GSL/include -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -fmacro-prefix-map=/home/mufeng/lede/build_dir/target-mipsel_24kc_musl/trojan-plus-10.0.3=trojan-plus-10.0.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Wall -Wextra -DPIC -fpic -flto -std=c++11 -O3 -ffunction-sections -fdata-sections -DNDEBUG -s   -Wall -Wextra -std=gnu++17 -MD -MT CMakeFiles/trojan.dir/src/core/config.cpp.o -MF CMakeFiles/trojan.dir/src/core/config.cpp.o.d -o CMakeFiles/trojan.dir/src/core/config.cpp.o -c src/core/config.cpp
src/core/config.cpp:26:10: fatal error: gsl/gsl: No such file or directory
 #include <gsl/gsl>
          ^~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
Makefile:69: recipe for target '/home/mufeng/lede/build_dir/target-mipsel_24kc_musl/trojan-plus-10.0.3/.built' failed
make[3]: *** [/home/mufeng/lede/build_dir/target-mipsel_24kc_musl/trojan-plus-10.0.3/.built] Error 1
make[3]: Leaving directory '/home/mufeng/lede/feeds/passwall/trojan-plus'
time: package/feeds/passwall/trojan-plus/compile#0.85#0.77#2.07
    ERROR: package/feeds/passwall/trojan-plus failed to build.
package/Makefile:114: recipe for target 'package/feeds/passwall/trojan-plus/compile' failed
make[2]: *** [package/feeds/passwall/trojan-plus/compile] Error 1
make[2]: Leaving directory '/home/mufeng/lede'
package/Makefile:110: recipe for target '/home/mufeng/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/mufeng/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/mufeng/lede'
/home/mufeng/lede/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2

hxlls, 2021-06-15:
然后呢？这个不应该去passwall 提么？

closed 2021-06-15 by FengNning

zou0804, 2023-05-30:
这个头文件怎么解决的？