coolsnowwolf/lede #7004: 减少编译所用空间

shoucandanghehe, 2021-06-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
如何减少编译所用空间，使用actions编译，空间不足
本地编译完之后lede目录有27.7g

### (2) 路由器型号和固件版本
x86-64

### (3) 详细日志
System.IO.IOException: No space left on device

yinzhefuyun, 2021-06-16:
actions不是有84G吗，你不够吗？

shoucandanghehe, 2021-06-16:
> actions不是有84G吗，你不够吗？

抛去系统和自带的环境
Ubuntu18.04剩余空间22g
[image]

shoucandanghehe, 2021-06-16:
再整理完其他的东西
编译前只剩
[image]

EnnawYang, 2021-06-16:
[image]
可能我用的是假的actions吧

yinzhefuyun, 2021-06-16:
https://github.com/P3TERX/Actions-OpenWrt  你换新脚本试一试，新脚本都是编译完成检测空间 ，不是编译钱检测空间，以前的脚本挂在空间是有问题，后面改了就多了40G

shoucandanghehe, 2021-06-16:
@EnnawYang 你用的啥系统

EnnawYang, 2021-06-16:
> https://github.com/P3TERX/Actions-OpenWrt 你换新脚本试一试，新脚本都是编译完成检测空间 ，不是编译钱检测空间，以前的脚本挂在空间是有问题，后面改了就多了40G

clone阶段也会检测空间的

shoucandanghehe, 2021-06-16:
@yinzhefuyun 我是自己写的脚本
https://github.com/shoucandanghehe/Auto-Build/tree/OpenWrt

EnnawYang, 2021-06-16:
> @yinzhefuyun 我是自己写的脚本
> https://github.com/shoucandanghehe/Auto-Build/tree/OpenWrt

你参照现有的脚本改改吧

yinzhefuyun, 2021-06-16:
> @yinzhefuyun 我是自己写的脚本
> https://github.com/shoucandanghehe/Auto-Build/tree/OpenWrt

这个作者把其他分区的40G软连接给链接过来了，很早以前就发现还有一个分区有40G

    - name: Clone source code
      working-directory: /workdir
      run: |
        df -hT $PWD
        git clone $REPO_URL -b $REPO_BRANCH openwrt
        ln -sf /workdir/openwrt $GITHUB_WORKSPACE/openwrt

281677160, 2021-06-16:
用P3TERX大神的脚本啊，空间大大的有
[image]

shoucandanghehe, 2021-06-16:
我懂了
他删了亿堆东西hhhh
https://github.com/P3TERX/Actions-OpenWrt/blob/5703b4ea5d856d0e94ae20818f7c0d59d6d51082/.github/workflows/build-openwrt.yml#L48

shoucandanghehe, 2021-06-16:
绝了，问题解决
```runner@fv-az76-888:\~/work/Auto-Build/Auto-Build$ df -hT $PWD
Filesystem     Type  Size  Used Avail Use% Mounted on
/dev/sdb1      ext4   84G   63G   21G  75% /
runner@fv-az76-888:\~/work/Auto-Build/Auto-Build$ sudo rm -rf /etc/apt/sources.list.d/*
runner@fv-az76-888:\~/work/Auto-Build/Auto-Build$ df -hT $PWD
Filesystem     Type  Size  Used Avail Use% Mounted on
/dev/sdb1      ext4   84G   63G   21G  75% /
runner@fv-az76-888:\~/work/Auto-Build/Auto-Build$ sudo rm -rf /usr/share/dotnet
runner@fv-az76-888:\~/work/Auto-Build/Auto-Build$ df -hT $PWD
Filesystem     Type  Size  Used Avail Use% Mounted on
/dev/sdb1      ext4   84G   37G   47G  45% /
runner@fv-az76-888:\~/work/Auto-Build/Auto-Build$ sudo rm -rf /usr/local/lib/android
runner@fv-az76-888:\~/work/Auto-Build/Auto-Build$ df -hT $PWD
Filesystem     Type  Size  Used Avail Use% Mounted on
/dev/sdb1      ext4   84G   27G   57G  33% /
runner@fv-az76-888:\~/work/Auto-Build/Auto-Build$ sudo rm -rf /opt/ghc
runner@fv-az76-888:\~/work/Auto-Build/Auto-Build$ df -hT $PWD
Filesystem     Type  Size  Used Avail Use% Mounted on
/dev/sdb1      ext4   84G   25G   59G  30% /
```

closed 2021-06-16 by shoucandanghehe

Nothend, 2022-08-16:
> 
朋友你好，我看你用的“/work/Auto-Build/Auto-Build”这个目录，是在哪里啊，yml文件中如何加入你上面几条删除命令呢？

shoucandanghehe, 2022-08-30:
> 朋友你好，我看你用的“/work/Auto-Build/Auto-Build”这个目录，是在哪里啊，yml文件中如何加入你上面几条删除命令呢？

参考https://github.com/shoucandanghehe/Auto-OpenWrt/blob/OpenWrt/.github/workflows/Build-x86-64.yml#L23-L29