coolsnowwolf/lede #7005: deepin 20 编译失败，大佬看看什么问题

miansan, 2021-06-16:
谢谢大佬进来指导，bow

make[2]: Leaving directory '/home/miansan/lede'
export MAKEFLAGS= ;make -w -r json_overview_image_info
make[2]: Entering directory '/home/miansan/lede'
make[2]: Nothing to be done for 'json_overview_image_info'.
make[2]: Leaving directory '/home/miansan/lede'
export MAKEFLAGS= ;make -w -r checksum
make[2]: Entering directory '/home/miansan/lede'
make[2]: Leaving directory '/home/miansan/lede'
/home/miansan/lede/staging_dir/host/bin/ccache -s
cache directory                     /home/miansan/lede/.ccache
primary config                      /home/miansan/lede/.ccache/ccache.conf
secondary config (readonly)         /home/miansan/lede/staging_dir/host/etc/ccache.conf
stats updated                       Wed Jun 16 01:33:37 2021
cache hit (direct)                  1840
cache hit (preprocessed)            1832
cache miss                         29514
cache hit rate                     11.06 %
called for link                     1151
called for preprocessing            2623
multiple source files                 25
compiler produced stdout               1
compiler produced empty output       116
compile failed                      1481
ccache internal error                 32
preprocessor error                   623
bad compiler arguments               262
unsupported source language            7
autoconf compile/link               5855
unsupported compiler option          143
unsupported code directive             7
output to stdout                       4
no input file                       1195
cleanups performed                     0
files in cache                     56171
cache size                         721.8 MB
max cache size                       5.0 GB
make[1]: Leaving directory '/home/miansan/lede'

EnnawYang, 2021-06-16:
你告诉我这是失败？

miansan, 2021-06-16:
> 你告诉我这是失败？

没有得到combined.img文件，不知道是不是cache太小导致，这个cache能不能改大？高手请指教，谢谢！

closed 2021-06-16 by coolsnowwolf