coolsnowwolf/lede #7007: 昨天源码的交换机页面无法打开

keke1023, 2021-06-16:
具体问题

A：“交换机”页面无法打开
B：luci-app-frpc一旦打开了其他设置里的”开启网页管理“就不能关闭此功能，关闭后保存会提示”一个或多个必选项值为空！“

路由器型号和固件版本
小米路由器4（mi r4）
OpenWrt R21.6.14 / LuCI Master (git-21.166.38875-46d873c) 是2021-6-15下午的源码

详细日志
交换机页面错误：
/usr/lib/lua/luci/dispatcher.lua:469: Failed to execute cbi dispatcher target for entry '/admin/network/vlan'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/admin_network/vlan.lua:190: bad argument #2 to 'tonumber' (integer expected, got string)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>

keke1023, 2021-06-16:
且通过ssh直接修改/etc/config/frp中的对应项为0也不行，结果一样

asushugo, 2021-06-16:
./scripts/feeds update -a && ./scripts/feeds install -a
交换机那个，更新一下即可

closed 2021-06-16 by coolsnowwolf