coolsnowwolf/lede #7010: libpython3--3.9 版本号多了一个 - 导致编译失败

codesnas, 2021-06-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：build: ensure that dash isn't prepended twice to abi version suffix

### (2) 路由器型号和固件版本

A：R24，R4S, RPi-4, RPi-3, Armvirt 上问题相同

### (3) 详细日志

A：Configuring fdisk.
 * pkg_hash_check_unresolved: cannot find dependency libpython3--3.9 for python3-base
 * pkg_hash_fetch_best_installation_candidate: Packages for python3-base found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-samba4:
 * 	libpython3--3.9
 * opkg_install_cmd: Cannot install package luci-app-samba4.
 * pkg_hash_check_unresolved: cannot find dependency libbz2--1.0 for bzip2
 * pkg_hash_fetch_best_installation_candidate: Packages for bzip2 found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package bzip2.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for tar:
 * 	libbz2--1.0
 * opkg_install_cmd: Cannot install package tar.
Configuring luci-i18n-transmission-en.

解决办法：
https://github.com/openwrt/openwrt/commit/fbb9b1f8ed0d8a76dd989cc6c16a4e0fda2b6e74

liwenjie119, 2021-06-16:
我这里还有两个类似的报错，不知道是不是同样的问题

Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libsqlite3--0 for sqlite3-cli
 * pkg_hash_fetch_best_installation_candidate: Packages for sqlite3-cli found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-music-remote-center:
 * 	libsqlite3--0
 * opkg_install_cmd: Cannot install package luci-app-music-remote-center.
 * pkg_hash_check_unresolved: cannot find dependency libpython3--3.9 for python3-base
 * pkg_hash_fetch_best_installation_candidate: Packages for python3-base found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-qbittorrent:
 * 	libpython3--3.9
 * opkg_install_cmd: Cannot install package luci-app-qbittorrent.
 * check_data_file_clashes: Package forked-daapd wants to install file /workdir/openwrt/build_dir/target-x86_64_musl/root-x86/etc/init.d/forked-daapd
	But that file is already provided by package  * luci-app-music-remote-center
 * opkg_install_cmd: Cannot install package forked-daapd.

ybist, 2021-06-16:
我也有
Configuring luci-i18n-firewall-zh-cn.
Configuring luci-i18n-docker-zh-cn.
Configuring libffmpeg-full.
Configuring ppp-mod-pppoe.
Configuring luci-i18n-nlbwmon-zh-cn.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libsqlite3--0 for sqlite3-cli
 * pkg_hash_fetch_best_installation_candidate: Packages for sqlite3-cli found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-music-remote-center:
 *      libsqlite3--0
 * opkg_install_cmd: Cannot install package luci-app-music-remote-center.
 * pkg_hash_check_unresolved: cannot find dependency libpython3--3.9 for python3-base
 * pkg_hash_fetch_best_installation_candidate: Packages for python3-base found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-qbittorrent:
 *      libpython3--3.9
 * opkg_install_cmd: Cannot install package luci-app-qbittorrent.
 * check_data_file_clashes: Package forked-daapd wants to install file /home/ubuntu/lede1/build_dir/target-x86_64_musl/root-x86/etc/init.d/forked-daapd
        But that file is already provided by package  * luci-app-music-remote-center
 * opkg_install_cmd: Cannot install package forked-daapd.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/ubuntu/lede1'
make[1]: *** [package/Makefile:111: /home/ubuntu/lede1/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/ubuntu/lede1'
make: *** [/home/ubuntu/lede1/include/toplevel.mk:230：world] 错误 2
ubuntu@ubuntu-virtual-machine:~/lede1$

closed 2021-06-17 by coolsnowwolf