coolsnowwolf/lede #7015: 编译出错，求救

softtip, 2021-06-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

全新编译，用的普通用户账号，ubuntu14.04，
在终端下login
登录普通用户账号cs
然后普通用户账号git clone然后
./scripts/feeds update -a
./scripts/feeds install -a
make menuconfig
make

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

touch /mnt/x86/lede0616/build_dir/hostpkg/opkg-2021-06-13-1bf042dd/.built
install -m0755 /mnt/x86/lede0616/build_dir/hostpkg/opkg-2021-06-13-1bf042dd/src/opkg-cl /mnt/x86/lede0616/staging_dir/host/bin/opkg
mkdir -p /mnt/x86/lede0616/staging_dir/hostpkg/stamp
touch /mnt/x86/lede0616/build_dir/hostpkg/opkg-2021-06-13-1bf042dd/.built
touch /mnt/x86/lede0616/staging_dir/hostpkg/stamp/.opkg_installed
make[3]: Leaving directory '/mnt/x86/lede0616/package/system/opkg'
time: package/system/opkg/host-compile#6.23#0.81#7.77
make[3]: Entering directory '/mnt/x86/lede0616/package/libs/toolchain'
mkdir -p /mnt/x86/lede0616/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /mnt/x86/lede0616/tmp/.root-copy.flock -c 'cp -fpR /mnt/x86/lede0616/build_dir/target-x86_64_musl/toolchain/.pkgdir/libc/. /mnt/x86/lede0616/staging_dir/target-x86_64_musl/root-x86/'
touch /mnt/x86/lede0616/staging_dir/target-x86_64_musl/root-x86/stamp/.libc_installed
mkdir -p /mnt/x86/lede0616/bin/targets/x86/64/packages /mnt/x86/lede0616/build_dir/target-x86_64_musl/toolchain/ipkg-x86_64/libgcc/CONTROL /mnt/x86/lede0616/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /mnt/x86/lede0616/build_dir/target-x86_64_musl/toolchain/ipkg-x86_64/libgcc/lib
cp -fpR /mnt/x86/lede0616/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/libgcc_s.so.* /mnt/x86/lede0616/build_dir/target-x86_64_musl/toolchain/ipkg-x86_64/libgcc/lib/
find /mnt/x86/lede0616/build_dir/target-x86_64_musl/toolchain/ipkg-x86_64/libgcc -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="x86_64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="x86_64-openwrt-linux-musl-nm" STRIP="/mnt/x86/lede0616/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/mnt/x86/lede0616/scripts/strip-kmod.sh" PATCHELF="/mnt/x86/lede0616/staging_dir/host/bin/patchelf" /mnt/x86/lede0616/scripts/rstrip.sh /mnt/x86/lede0616/build_dir/target-x86_64_musl/toolchain/ipkg-x86_64/libgcc
rstrip.sh: /mnt/x86/lede0616/build_dir/target-x86_64_musl/toolchain/ipkg-x86_64/libgcc/lib/libgcc_s.so.1: shared object
(cd /mnt/x86/lede0616/build_dir/target-x86_64_musl/toolchain/ipkg-x86_64/libgcc/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /mnt/x86/lede0616/bin/targets/x86/64/packages
/mnt/x86/lede0616/staging_dir/host/bin/fakeroot /mnt/x86/lede0616/scripts/ipkg-build -m "" /mnt/x86/lede0616/build_dir/target-x86_64_musl/toolchain/ipkg-x86_64/libgcc /mnt/x86/lede0616/bin/targets/x86/64/packages
/mnt/x86/lede0616/scripts/ipkg-build: 127: /mnt/x86/lede0616/scripts/ipkg-build: realpath: not found
make[3]: *** [Makefile:739: /mnt/x86/lede0616/bin/targets/x86/64/packages/libgcc1_8.4.0-3_x86_64.ipk] Error 127
make[3]: Leaving directory '/mnt/x86/lede0616/package/libs/toolchain'
time: package/libs/toolchain/compile#0.05#0.04#0.25
    ERROR: package/libs/toolchain failed to build.
make[2]: *** [package/Makefile:116: package/libs/toolchain/compile] Error 1
make[2]: Leaving directory '/mnt/x86/lede0616'
make[1]: *** [package/Makefile:110: /mnt/x86/lede0616/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/mnt/x86/lede0616'
make: *** [/mnt/x86/lede0616/include/toplevel.mk:230: world] Error 2
cs@cs-virtual-machine:/mnt/x86/lede0616$ whoami
cs
cs@cs-virtual-machine:/mnt/x86/lede0616$

Dimitri2020007, 2021-06-16:
在编译之前安装了readme中提示的那些软件吗？

softtip, 2021-06-16:
搞定了

softtip, 2021-06-16:
apt-get install realpath
就解决了

closed 2021-06-20 by coolsnowwolf