coolsnowwolf/lede #7019: 因boost版本不匹配导致trojan编译失败

lilunxm12, 2021-06-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译失败,看日志是因为编译trojan时找不到boost1.73,但make download的时候并未报错,回朔发现下载的是1.76版本,和trojan依赖的并不一致

### (2) 路由器型号和固件版本

A：ARCH17

### (3) 详细日志

A：
[image]
[image]

closed 2021-06-17 by coolsnowwolf

lilunxm12, 2021-06-17:
@coolsnowwolf 请问为何关闭了呢?我刚刚试了下还是同样的编译错误啊?还是说这个应该在helloworld里提?可是trojan本身确实是在lean里面而非helloworld里啊

lilunxm12, 2021-06-17:
@coolsnowwolf 推倒重来,重新克隆并安装有新的发现,包括trojan在内有很多makefile报依赖缺失,看样子是改名了?
`
Collecting target info: done
WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libpng', which does not exist
WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libupnp', which does not exist
WARNING: Makefile 'package/lean/amule/Makefile' has a build dependency on 'libgd', which does not exist
WARNING: Makefile 'package/lean/antileech/Makefile' has a build dependency on 'libgd', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'lm-sensors', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'bc', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'lm-sensors', which does not exist
WARNING: Makefile 'package/lean/autosamba/Makefile' has a dependency on 'luci-app-samba', which does not exist
WARNING: Makefile 'package/lean/autosamba/Makefile' has a dependency on 'wsdd2', which does not exist
WARNING: Makefile 'package/lean/baidupcs-web/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_aliyun/Makefile' has a dependency on 'ddns-scripts', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_dnspod/Makefile' has a dependency on 'ddns-scripts', which does not exist
WARNING: Makefile 'package/lean/default-settings/Makefile' has a dependency on 'luci-base', which does not exist
WARNING: Makefile 'package/lean/default-settings/Makefile' has a dependency on 'luci', which does not exist
WARNING: Makefile 'package/lean/default-settings/Makefile' has a dependency on 'luci-compat', which does not exist
WARNING: Makefile 'package/lean/dnsforwarder/Makefile' has a dependency on 'coreutils-base64', which does not exist
WARNING: Makefile 'package/network/utils/dpdk/Makefile' has a build dependency on 'meson/host', which does not exist
WARNING: Makefile 'package/network/utils/dpdk/Makefile' has a build dependency on 'ninja/host', which does not exist
WARNING: Makefile 'package/network/utils/dpdk/Makefile' has a build dependency on 'ninja/host', which does not exist
WARNING: Makefile 'package/network/utils/dpdk/Makefile' has a build dependency on 'ninja/host', which does not exist
WARNING: Makefile 'package/network/utils/dpdk/Makefile' has a build dependency on 'ninja/host', which does not exist
WARNING: Makefile 'package/network/utils/dpdk/Makefile' has a build dependency on 'ninja/host', which does not exist
WARNING: Makefile 'package/lean/frp/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/gmediarender/Makefile' has a dependency on 'gstreamer1', which does not exist
WARNING: Makefile 'package/lean/gmediarender/Makefile' has a dependency on 'libgstreamer1', which does not exist
WARNING: Makefile 'package/lean/gmediarender/Makefile' has a dependency on 'gstreamer1-libs', which does not exist
WARNING: Makefile 'package/lean/gmediarender/Makefile' has a dependency on 'glib2', which does not exist
WARNING: Makefile 'package/lean/gmediarender/Makefile' has a dependency on 'libupnp', which does not exist
WARNING: Makefile 'package/lean/ipv6-helper/Makefile' has a dependency on 'luci-proto-ipv6', which does not exist
WARNING: Makefile 'package/boot/kexec-tools/Makefile' has a dependency on 'liblzma', which does not exist
WARNING: Makefile 'package/network/services/lldpd/Makefile' has a dependency on 'libnetsnmp', which does not exist
WARNING: Makefile 'package/lean/luci-app-accesscontrol/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-adbyby-plus/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-airplay2/Makefile' has a dependency on 'alsa-utils', which does not exist
WARNING: Makefile 'package/lean/luci-app-airplay2/Makefile' has a dependency on 'shairport-sync-openssl', which does not exist
WARNING: Makefile 'package/lean/luci-app-airplay2/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-amule/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-arpbind/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-autoreboot/Makefile' has a dependency on 'luci', which does not exist
WARNING: Makefile 'package/lean/luci-app-autoreboot/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-baidupcs-web/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-cifs-mount/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-cifsd/Makefile' has a dependency on 'ksmbd-server', which does not exist
WARNING: Makefile 'package/lean/luci-app-cifsd/Makefile' has a build dependency on 'luci-base/host', which does not exist
[... cut ...]
WARNING: Makefile 'package/lean/trojan/Makefile' has a dependency on 'boost-program_options', which does not exist
WARNING: Makefile 'package/lean/trojan/Makefile' has a dependency on 'boost-date_time', which does not exist
WARNING: Makefile 'package/lean/UnblockNeteaseMusic/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/lean/UnblockNeteaseMusicGo/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/wxbase/Makefile' has a dependency on 'libexpat', which does not exist
`

EnnawYang, 2021-06-17:
...就你最离谱

lilunxm12, 2021-06-17:
> 
> 
> > 重新克隆并安装有新的发现,包括trojan在内有很多makefile报依赖缺失
> 
> ...就你最离谱

请问怎么离谱了呢?我也不希望浪费开发者时间,已经尽我力所能及的debug了,所有下载都经过了代理,make download日志我仔细检查了确保没有404等导致下载缺失,编译日志我也看了,确认了是boost版本不匹配,最后也换了新容器重新clone然后编译,复现出来的仍然是boost不匹配的错误,就是有新的发现了而已.如果有哪里做的不好让你觉得我离谱了,请指出