coolsnowwolf/lede #7023: 最新的5.10内核版本的广告大师好像更新有问题

hxlls, 2021-06-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：更新失败，并且是秒返回那种，用的是不保存配置升级的方式，应该排除了之前配置的影响。
[image]

[image]

### (2) 路由器型号和固件版本

A：
[image]

### (3) 详细日志

A：

weilai-all, 2021-06-16:
有遇到，输出是没找到 wget-ssl。

hxlls, 2021-06-16:
> 有遇到，输出是没找到 wget-ssl。

查了下配置，wget-ssl 好像已经勾选，应该不是这个原因
[image]

stztao, 2021-06-16:
我刚解决了，ssh下运行ln /usr/bin/wget /usr/bin/wget-ssl 再 /usr/share/adbyby/adupdate.sh 重启下

weilai-all, 2021-06-16:
> > 有遇到，输出是没找到 wget-ssl。
> 
> 查了下配置，wget-ssl 好像已经勾选，应该不是这个原因
> [image]

嗯，才疏学浅，也比对过文件确实存在，按照@stztao的方法创建软链接应该可以解决，之前百度说什么改变了。

closed 2021-06-16 by coolsnowwolf