coolsnowwolf/lede #7033: 各位同学前来帮忙一下，还差一点马上就要成功了。。。

flyhong666, 2021-06-17:
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libpython3--3.9 for python3-base
 * pkg_hash_fetch_best_installation_candidate: Packages for python3-base found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for python3-bcrypt:
 *      libpython3--3.9
 * opkg_install_cmd: Cannot install package python3-bcrypt.
 * pkg_hash_check_unresolved: cannot find dependency libsqlite3--0 for sqlite3-cli
 * pkg_hash_fetch_best_installation_candidate: Packages for sqlite3-cli found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for forked-daapd:
 *      libsqlite3--0
 * opkg_install_cmd: Cannot install package forked-daapd.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/fly/lede'
make[1]: *** [package/Makefile:111: /home/fly/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/fly/lede'
make: *** [/home/fly/lede/include/toplevel.mk:230：world] 错误 2

ldy8182, 2021-06-17:
make clean

flyhong666, 2021-06-17:
> make clean

我dirclean 都执行过了，又来？

closed 2021-06-17 by coolsnowwolf

flyhong666, 2021-06-17:
去掉qbt和redsocks2以及相关的一些依赖， 然后clean，编译成功，太难了。。。