coolsnowwolf/lede #7035: 编译出现错误

13810169, 2021-06-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译错误

### (2) 路由器型号和固件版本

A：x86-64

### (3) 详细日志

A：
echo "xray-geodata" >> /home/qi/lede/staging_dir/target-x86_64_musl/pkginfo/xray-core.default.install
make[3]: Leaving directory '/home/qi/lede/feeds/small/xray-core'
time: package/feeds/small/xray-core/compile#29.82#8.52#42.45
make[2]: Leaving directory '/home/qi/lede'
make[1]: *** [package/Makefile:110: /home/qi/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/qi/lede'
make: *** [/home/qi/lede/include/toplevel.mk:230：world] 错误 2

EnnawYang, 2021-06-17:
1. 日志不全
2. xray不是本源码库项目

Beginner-Go, 2021-06-17:
> 1. 日志不全
> 2. xray不是本源码库项目

别管他，又是一个杂交Package,他上游出问题 赖我们源码

13810169, 2021-06-17:
唉，，我没说源码怎么样，，我是好奇以前多次编译都没有出现问题，这次升级之后失败了，而且用大雕的源码不添加其他插件编译是可以成功的。这里不是人多么，想看看大家有没有什么解决办法，这个翻墙的插件总不能没有啊。行吧，我关闭了。

closed 2021-06-17 by 13810169