coolsnowwolf/lede #7036: Build dependency: Please install the GNU C Compiler (gcc) 6 or later

AndyLin2020, 2021-06-17:
### (1) 具体问题

A：更新后一直无法正常编译，重新删除后pull，在./scripts/feeds update -a && ./scripts/feeds install -a时就一直提示编译器问题
 
### (3) 详细日志

A：
Checking 'which'... ok.
Checking 'ldconfig-stub'... ok.

Build dependency: Please install the GNU C Compiler (gcc) 6 or later
Build dependency: Please install the GNU C++ Compiler (g++) 6 or later
Build dependency: Please install Python >= 3.6
Build dependency: Please install Python >= 3.6

/home/andy/Desktop/lede/include/prereq.mk:9: recipe for target 'prereq' failed
Prerequisite check failed. Use FORCE=1 to override.
/home/andy/Desktop/lede/include/toplevel.mk:179: recipe for target 'staging_dir/host/.prereq-build' failed
make: *** [staging_dir/host/.prereq-build] Error 1

AndyLin2020, 2021-06-17:
重新升级了gcc6和g++6，以及Python 3.8.9

coolsnowwolf, 2021-06-17:
ubuntu 版本太低，需要 18LTS + 系统

closed 2021-06-17 by coolsnowwolf